mpromonet/webrtc-streamer #1: rtspvideocapturer

mattj103, 2015-04-02:
How do you use the rtspvideocapturer? I noticed you have an h264 codec in there, are you using a modified webrtc source to get the h264 codec to work? It does not appear to work with the latest webrtc source.

Thanks for the great work also, this is a great up to date example, very helpful.

mpromonet, 2015-04-03:
I tried to use h264, but Native WebRTC seems really broken, and it seems only firefox support a little H264 (only Baseline profile).

mattj103, 2015-04-03:
How far did you get with it? Were you ever able to get a stream? I'd like to see if I can get it running.

mpromonet, 2015-04-09:
Using an RTSP server to send H264 stream produce following output : 

      NOTIFY size:5905
      [027:228] Error(webrtcvideoframe.cc:267): Error parsing format: H264 return code : -1
      [027:228] Error(webrtcvideoframefactory.cc:38): Failed to create WebRtcVideoFrame in      CreateAliasedFrame.
      [027:228] Error(videocapturer.cc:562): Couldn't convert to I420! From H264 720x576x0 To 720 x 576
      NOTIFY size:5974
      [027:269] Error(webrtcvideoframe.cc:267): Error parsing format: H264 return code : -1
      [027:269] Error(webrtcvideoframefactory.cc:38): Failed to create WebRtcVideoFrame in CreateAliasedFrame.
      [027:270] Error(videocapturer.cc:562): Couldn't convert to I420! From H264 720x576x0 To 720 x 576
      NOTIFY size:5916
      [027:310] Error(webrtcvideoframe.cc:267): Error parsing format: H264 return code : -1
      [027:310] Error(webrtcvideoframefactory.cc:38): Failed to create WebRtcVideoFrame in CreateAliasedFrame.
      [027:310] Error(videocapturer.cc:562): Couldn't convert to I420! From H264 720x576x0 To 720 x 576
      NOTIFY size:5696
      [027:351] Error(webrtcvideoframe.cc:267): Error parsing format: H264 return code : -1
      [027:351] Error(webrtcvideoframefactory.cc:38): Failed to create WebRtcVideoFrame in CreateAliasedFrame.
      [027:351] Error(videocapturer.cc:562): Couldn't convert to I420! From H264 720x576x0 To 720 x 576

The H264 code of webrtc seems completely broken.

DanielArnett, 2015-04-27:
I get an error on `make`

`rtspvideocapturer.h:24:24: fatal error: liveMedia.hh: No such file or directory
 #include "liveMedia.hh"`

mpromonet, 2015-04-28:
This is a really different concern. Please  use a new issue for that. 
The code actually need live555 installed, I will manage differently this dependency (or remove it)

DanielArnett, 2015-04-28:
I apologize, but appreciate the quick response!

YFJiang, 2016-06-12:
I find this: https://bugs.chromium.org/p/webrtc/issues/detail?id=5687
Does it mean that google webrtc library does not support encoded frames such as h264 yet?

mpromonet, 2016-06-12:
WebRTC developpment is a little hard to follow, it seems that H264 support is progressing.
From [M51 Release Notes](https://groups.google.com/forum/#!topic/discuss-webrtc/GdsmhrVaxdU/discussion):

    **Improved H.264 compatibility**

    The experimental support (still behind a flag) for H.264 in WebRTC now works with more H.264 endpoints, including Firefox.

mpromonet, 2017-01-05:
Hi,

I recently modified the rtspcapturer to follow webrtc evolution and add a step to decode H264, then it is possible to make a proxy between an H264 streamed over RTSP  to WebRTC. 
But this decode H264 and re-encode to H264 or VP8, VP9 depending on webrtc client side.
In order to try this you should install live555 and uncomment in the Makefile
`#	CFLAGS += -DHAVE_LIVE555`
The decode/encode step have the advantage to allow encoding adaptation to the bandwidth and to serve client that doesnot support H264, and obviously the drawback is the computing cost.
If you are interested to make a lightweight gateway between RTSP and webrtc, you should look to [janus-gateway](https://github.com/meetecho/janus-gateway) project.

Best Regards,
Michel.

saket424, 2017-01-06:
Michel,
Thanks for the update and the special mention of the benefit of re-encoding to cater to bandwidth adaptation which I had not considered. Will definitely give it a whirl.

saket424, 2017-01-16:
Michel,
I successfully compiled the webrtc-streamer head with live555 to attempt rtsp capture
I unsuccessfully tried to use a public h264rtsp url rtsp://85.255.175.244:554/h264
Do you have a different public rtsp url that streams h264 which works with your webrtc-streamer server?

mpromonet, 2017-01-16:
Hi saket,
I just try to connect to rtsp://85.255.175.244:554/h264 and it is working well. Maybe the RTSP url is not reachable from where `webrtc-streamer` is running. 
Did you tried with VLC, or better openRTSP on the same server ? Maybe the log could also helps to localize the problem.
You should also take care that WebRTC should be build with H264 decoder, to build webRTC, I am using : 

> gn gen out/Release --args='is_debug=false rtc_use_h264=true ffmpeg_branding="Chrome" rtc_include_tests=false'

Regards,
Michel.

saket424, 2017-01-18:
Michel,
Thanks. I confirm the rtsp url listed above is working quite well. One question I had was if two peerconnections try to access the same rtsp h264 URL, will that be two re-encode pipelines or can the encoded frames be re-used for both peer connections?

saket424, 2017-01-19:
Michel,
Disconnect(hangup) currently does not seem to stop the rtsp stream. frames continue to arrive at the server.

saket424, 2017-01-19:
Michel,
I am unable to play out any rtsp video feeds from an axis camera(h264 baseline profile). Can you try this rtsp url? rtsp://217.17.220.110:554/axis-media/media.amp

ffplay and vlc seem to play it ok

    Stream #0:0: Video: h264 (Baseline), yuvj420p(pc, bt709), 1280x800 [SAR 1:1 DAR 8:5], 15.08 tbr, 90k tbn, 180k tbc

Appreciate any insights into why it does not play with webrtc-streamer

Thanks!

mpromonet, 2017-01-19:
Hi saket,
It fails because SPS/PPS are not in the stream and SPS/PPS in the RTSP DESCRIBE was not used.
I fixed this in this commit 884b0573e516d9190e05dd1d611c42b09a03f340.
Now it should works.
Regards,
Michel.

saket424, 2017-01-20:
Thanks for the quick turnaround. The axis camera feed now works flawlessly with commit 884b057

closed 2017-01-28 by mpromonet