mpromonet/webrtc-streamer #2: liveMedia.hh

DanielArnett, 2015-04-28:
Properly posting an earlier issue:

I get an error on `make`

`rtspvideocapturer.h:24:24: fatal error: liveMedia.hh: No such file or directory #include "liveMedia.hh"`

mpromonet responded

>"... The code actually need live555 installed, I will manage differently this dependency (or remove it)"

I'll download and install [live555](http://www.live555.com/) and give that a try.

DanielArnett, 2015-04-28:
I've [downloaded](http://www.live555.com/mediaServer/#downloading) the LiveMediaServer binary, but don't see a way to use it in webrtc-streamer. Now I'm attempting building from source using [these instructions](http://www.live555.com/liveMedia/#config-unix)

1. [Download the live555-latest.tar.gz](http://www.live555.com/liveMedia/public/)
2. Extract to a working directory.
3. Enter the extracted `live` directory.
4. `./genMakefiles linux`
5. `make`
6. `sudo make install`

I now have the missing liveMedia.hh file in the `live/liveMedia/include` directory. 

How do I reference a header file from another directory while building? I have tried adding the include directory to my $PATH variable (e.g. `export PATH=$PATH:/home/dan/live/liveMedia/include`), but I am still receiving the above error.

DanielArnett, 2015-04-28:
cd into the `live` directory and run this command:

`sudo mkdir /usr/include/liveMedia /usr/include/groupsock /usr/include/UsageEnvironment /usr/include/BasicUsageEnvironment;sudo cp -r liveMedia/include/* /usr/include/liveMedia;sudo cp -r groupsock/include/* /usr/include/groupsock;sudo cp -r UsageEnvironment/include/* /usr/include/UsageEnvironment;sudo cp -r BasicUsageEnvironment/include/* /usr/include/BasicUsageEnvironment;`

This moves all of the include files into the locations specified by the webrtc-streamer Makefile on line 10:
`# CFLAGS
live555 += -I /usr/include/liveMedia  -I /usr/include/groupsock -I /usr/include/UsageEnvironment -I /usr/include/BasicUsageEnvironment/`

mpromonet, 2015-04-28:
Usually distribution include live555.
You could install :  
- on debian like distribution using `apt-get install liblivemedia-dev`
- on rehat like distribution using `yum install live555-devel`
- on arch like distribution `pacman -Sv live-media`

If you build it from source you can avoid the manual copy using :

         sudo make install PREFIX=/usr

DanielArnett, 2015-04-28:
Good to know how it's properly done. Now I'm getting issues with the JSON Writer packaged with webrtc. 

Now I'm having issues with the JSON StyledWriter

`/home/dan/andy/webrtc-streamer/main.cpp:68: undefined reference to `Json::StyledWriter::StyledWriter()'`

I'll post a new issue about it...

closed 2015-05-01 by mpromonet