mpromonet/webrtc-streamer #3: undefined reference to `Json::StyledWriter::StyledWriter()'

DanielArnett, 2015-04-28:
Error:
`/home/dan/andy/webrtc-streamer/main.cpp:68: undefined reference to `Json::StyledWriter::StyledWriter()'`

When I look into the `webrtc-checkout/src/chromium/src/third_party/jsoncpp/source/include/` directory there is nothing there but a `json/` directory.

Returning to the Makefile  on line 19 I append
`CFLAGS += -I $(WEBRTCROOT)/src -I $(WEBRTCROOT)/src/chromium/src/third_party/jsoncpp/source/include`
to be 
`CFLAGS += -I $(WEBRTCROOT)/src -I $(WEBRTCROOT)/src/chromium/src/third_party/jsoncpp/source/include/json`

But this only returns deeper errors with the #include "string.h"

`In file included from main.cpp:10:0:
/usr/include/string.h:39:69: error: missing binary operator before token "("
 #if defined __cplusplus && (__cplusplus >= 199711L || __GNUC_PREREQ (4, 4))`...

DanielArnett, 2015-04-28:
The best explanation of what might be causing the error is from [this post](http://stackoverflow.com/a/5413181) on stackoverflow

>The reason for this peculiar behavior is probably a file name collision: string.h includes <features.h> which should usually resolve to /usr/include/features.h or similar. But the json/ dir also contains a features.h, so if it comes first in the include path, that will be included instead, although it probably serves a different purpose than string.h needs.

mpromonet, 2015-04-28:
From webrtc/base/json.h : 

     #if !defined(WEBRTC_EXTERNAL_JSON)
     #include "json/json.h"
     #else
     #include "third_party/jsoncpp/json.h"
     #endif
Then you should not add the "/json" to the  `-I $(WEBRTCROOT)/src/chromium/src/third_party/jsoncpp/source/include`.

Which options did you used to compile webrtc ? did you define WEBRTC_EXTERNAL_JSON setting build_json=0 ?

DanielArnett, 2015-04-28:
Only used `ninja -C src/out/Debug` to build. I assume I should build the release as well. How do I define the WEBRTC_EXTERNAL_JSON?

mpromonet, 2015-04-28:
Building webrtc is still hard to understand for me... step by step I read the json makefile, but...
The options are set by GYP_DEFINES..

If you are using Debug, you should build with make `WEBRTCROOT=<path to WebRTC> WEBRTCBUILD=Debug`, but I guess you find this in the makefile.

DanielArnett, 2015-04-28:
Very good to know, thank you for your help! I'll try it out and let you know how it goes.

DanielArnett, 2015-04-29:
I tried `export GYP_DEFINES=build_json=0`

and in my `src/webrtc/base/BUILD.gn` I have

    if (rtc_build_json) {
        deps += [ "//third_party/jsoncpp" ]
    } else {
          include_dirs += [ rtc_jsoncpp_root ]
        
          # When defined changes the include path for json.h to where it is
          # expected to be when building json outside of the standalone build.
          defines += [ "WEBRTC_EXTERNAL_JSON" ]
    }

so it looks to me like `WEBRTC_EXTERNAL_JSON` is defined.

But the `StyledWriter` error persists.

mpromonet, 2015-05-01:
I never build in Debug, I will try to see if I get this issue.

bshouse, 2015-09-20:
I was getting the same error after struggling to build WebRTC. Once it got it to build properly, I had to do a "make clean" on this project and then was able to compile without errors.

mpromonet, 2015-09-24:
The makefile create a librarie with all the librairies found in WebRTC, then if some are missing when this librarie is build, it is needed to remove it (with makce clean as you dif) because there is no dependency in the makefile.
I will see if I can avoid this, perhaps adding `jsoncpp` in the link dependency ?

commit e73d8114 (2015-10-10): fix #3 : adding the list of webrtc libraries to the dependency of libwebrtc.a
closed 2015-10-10 by mpromonet