mpromonet/webrtc-streamer #5: Unable to compile code.

takacsv, 2016-01-25:
Hey Michel! I'm trying to build this project and it fails at the make command, here is the output:
```
g++  -o webrtc-server__Release -W -pthread -g -std=gnu++0x -I mongoose -DWEBRTC_POSIX -fno-rtti -I ./src -I ./src/chromium/src/third_party/jsoncpp/source/include main.cpp PeerConnectionManager.cpp mongoose/mongoose.c libWebRTC__Release.a -lX11 -lXext -lexpat -ldl -lnss3 -lnssutil3 -lplc4 -lnspr4 -lrt
PeerConnectionManager.cpp: In member function ‘std::pair<rtc::scoped_refptr<webrtc::PeerConnectionInterface>, PeerConnectionManager::PeerConnectionObserver*> PeerConnectionManager::CreatePeerConnection(const string&)’:
PeerConnectionManager.cpp:93:15: error: no matching function for call to ‘webrtc::PeerConnectionFactoryInterface::CreatePeerConnection(webrtc::PeerConnectionInterface::IceServers&, NULL, NULL, NULL, PeerConnectionManager::PeerConnectionObserver*&)’
            obs);
               ^
In file included from PeerConnectionManager.h:16:0,
                 from PeerConnectionManager.cpp:20:
./src/talk/app/webrtc/peerconnectioninterface.h:545:55: note: candidate: virtual rtc::scoped_refptr<webrtc::PeerConnectionInterface> webrtc::PeerConnectionFactoryInterface::CreatePeerConnection(const webrtc::PeerConnectionInterface::RTCConfiguration&, const webrtc::MediaConstraintsInterface*, rtc::scoped_ptr<cricket::PortAllocator>, rtc::scoped_ptr<webrtc::DtlsIdentityStoreInterface>, webrtc::PeerConnectionObserver*)
   virtual rtc::scoped_refptr<PeerConnectionInterface> CreatePeerConnection(
                                                       ^
./src/talk/app/webrtc/peerconnectioninterface.h:545:55: note:   no known conversion for argument 1 from ‘webrtc::PeerConnectionInterface::IceServers {aka std::vector<webrtc::PeerConnectionInterface::IceServer>}’ to ‘const webrtc::PeerConnectionInterface::RTCConfiguration&’
Makefile:36: recipe for target 'webrtc-server__Release' failed
make: *** [webrtc-server__Release] Error 1
```
Is this some kind of incompatibility between your code and the webrtc master? I'm on HEAD master of the webrtc repo as per the instructions and have a successful ninja run .

Thanks!

mpromonet, 2016-01-25:
Hi Takacs,
I fix this short time ago (in this commit 9a1e7ba0101b82cc96969299f10d237273c0d591).
Which code version are you using ?
Best Regards,
Michel.

takacsv, 2016-01-26:
I had to update to get rid of those errors, unfortunately I've got new ones, the make output is here: https://gist.github.com/takacsv/b9db00d626b4b60421d4

mpromonet, 2016-01-26:
This is due to mixing c++11 and std compilation. I update the Makefile in order to build using c++11 (commit 01780a8754069bb8e98fac8c363e8892adc7712d)

takacsv, 2016-01-26:
Awesome, I was able to build the project just fine. Thanks for the prompt reply, man! Also, should I close this issue now that it is solved?

closed 2016-01-26 by mpromonet

mpromonet, 2016-01-26:
Nice it solve your issue, for my build environment it doesnot change anything.
I will try to keep only with webrtc upgrade.
Best Regards
Michel.