mpromonet/webrtc-streamer #8: RTSP source can get worked

zackxue, 2016-08-17:
We use a Axis IP Camera for the testing, input the URL to input box in the webui, it seems that the rtsp work well, but the web can get any video, both  firefox and chrome

Sending request: PLAY rtsp://159.99.251.76/onvif-media/media.amp?profile=quality_h264&sessiontimeout=60&streamtype=unicast RTSP/1.0
CSeq: 5
Authorization: Digest username="admin", realm="AXIS_WS_00408CA21FFF", nonce="00006227Y732189e0a47aeea6dc1e057ada6997c111101", uri="rtsp://159.99.251.76/onvif-media/media.amp/", response="08f0a0b54cd1294b057f60cb553a9c26"
User-Agent: LIVE555 Streaming Media v2016.08.07
Session: B3926245
Range: npt=0.000-

1804289383:{
   "sdp" : "v=0\r\no=- 2665439333175596351 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE video\r\na=msid-semantic: WMS stream_label\r\nm=video 9 UDP/TLS/RTP/SAVPF 100 101 116 117 96 97 98\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:TiCG/rgYvBn4YaNH\r\na=ice-pwd:qaxHNknBu6JXN5/Q16kIa/1e\r\na=fingerprint:sha-256 BD:01:A9:C7:8A:B2:24:43:F8:5B:FA:17:0C:57:95:32:48:DC:46:CD:B6:3F:77:D3:BB:3E:B9:9D:D6:66:9B:E1\r\na=setup:actpass\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=sendrecv\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:100 VP8/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=rtpmap:101 VP9/90000\r\na=rtcp-fb:101 ccm fir\r\na=rtcp-fb:101 nack\r\na=rtcp-fb:101 nack pli\r\na=rtcp-fb:101 goog-remb\r\na=rtcp-fb:101 transport-cc\r\na=rtpmap:116 red/90000\r\na=rtpmap:117 ulpfec/90000\r\na=rtpmap:96 rtx/90000\r\na=fmtp:96 apt=100\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=101\r\na=rtpmap:98 rtx/90000\r\na=fmtp:98 apt=116\r\na=ssrc-group:FID 1931720387 2078885298\r\na=ssrc:1931720387 cname:cg6pjT6riTx0Qk81\r\na=ssrc:1931720387 msid:stream_label video_label\r\na=ssrc:1931720387 mslabel:stream_label\r\na=ssrc:1931720387 label:video_label\r\na=ssrc:2078885298 cname:cg6pjT6riTx0Qk81\r\na=ssrc:2078885298 msid:stream_label video_label\r\na=ssrc:2078885298 mslabel:stream_label\r\na=ssrc:2078885298 label:video_label\r\n",
   "type" : "offer"
}

[015:879] [1854] (PeerConnectionManager.h:32): virtual void PeerConnectionManager::SetSessionDescriptionObserver::OnSuccess() type:offer v=0
o=- 2665439333175596351 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE video
a=msid-semantic: WMS stream_label
m=video 9 UDP/TLS/RTP/SAVPF 100 101 116 117 96 97 98
c=IN IP4 0.0.0.0
a=rtcp:9 IN IP4 0.0.0.0
a=ice-ufrag:TiCG/rgYvBn4YaNH
a=ice-pwd:qaxHNknBu6JXN5/Q16kIa/1e
a=fingerprint:sha-256 BD:01:A9:C7:8A:B2:24:43:F8:5B:FA:17:0C:57:95:32:48:DC:46:CD:B6:3F:77:D3:BB:3E:B9:9D:D6:66:9B:E1
a=setup:actpass
a=mid:video
a=extmap:2 urn:ietf:params:rtp-hdrext:toffset
a=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
a=extmap:4 urn:3gpp:video-orientation
a=sendrecv
a=rtcp-mux
a=rtcp-rsize
a=rtpmap:100 VP8/90000
a=rtcp-fb:100 ccm fir
a=rtcp-fb:100 nack
a=rtcp-fb:100 nack pli
a=rtcp-fb:100 goog-remb
a=rtcp-fb:100 transport-cc
a=rtpmap:101 VP9/90000
a=rtcp-fb:101 ccm fir
a=rtcp-fb:101 nack
a=rtcp-fb:101 nack pli
a=rtcp-fb:101 goog-remb
a=rtcp-fb:101 transport-cc
a=rtpmap:116 red/90000
a=rtpmap:117 ulpfec/90000
a=rtpmap:96 rtx/90000
a=fmtp:96 apt=100
a=rtpmap:97 rtx/90000
a=fmtp:97 apt=101
a=rtpmap:98 rtx/90000
a=fmtp:98 apt=116
a=ssrc-group:FID 1931720387 2078885298
a=ssrc:1931720387 cname:cg6pjT6riTx0Qk81
a=ssrc:1931720387 msid:stream_label video_label
a=ssrc:1931720387 mslabel:stream_label
a=ssrc:1931720387 label:video_label
a=ssrc:2078885298 cname:cg6pjT6riTx0Qk81
a=ssrc:2078885298 msid:stream_label video_label
a=ssrc:2078885298 mslabel:stream_label
a=ssrc:2078885298 label:video_label

Received 254 new bytes of response data.
Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 5
Session: B3926245
Range: npt=0-
RTP-Info: url=rtsp://159.99.251.76/onvif-media/media.amp/trackID=1?profile=quality_h264&sessiontimeout=60&streamtype=unicast;seq=14334;rtptime=2856969365
Date: Wed, 17 Aug 2016 16:21:43 GMT

PLAY OK 
===> HTTP request /getIceCandidate
res:1
body:
===> HTTP request /answer
res:1
body:{"type":"answer","sdp":"v=0\r\no=mozilla...THIS_IS_SDPARTA-48.0 5570441187170516131 0 IN IP4 0.0.0.0\r\ns=-\r\nt=0 0\r\na=fingerprint:sha-256 94:6F:90:49:AF:A9:64:9F:EA:0E:42:EA:D6:C1:E8:02:DE:63:CF:77:08:57:75:A5:0A:EC:DF:09:2E:78:8F:26\r\na=group:BUNDLE video\r\na=ice-options:trickle\r\na=msid-semantic:WMS *\r\nm=video 9 UDP/TLS/RTP/SAVPF 100\r\nc=IN IP4 0.0.0.0\r\na=recvonly\r\na=fmtp:100 max-fs=12288;max-fr=60\r\na=ice-pwd:a3e34811420eb8ce74dc79c7ee0fda92\r\na=ice-ufrag:b0865c1b\r\na=mid:video\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-mux\r\na=rtpmap:100 VP8/90000\r\na=setup:active\r\na=ssrc:2174317038 cname:{19b200c1-3f71-4641-a5e8-0a84d5c9fcc8}\r\n"}
[017:757] [1854] (PeerConnectionManager.cpp:141): From peerid:1804289383 received session description :answer
[018:194] [1854] (PeerConnectionManager.h:32): virtual void PeerConnectionManager::SetSessionDescriptionObserver::OnSuccess() type:answer v=0
o=- 2665439333175596351 2 IN IP4 127.0.0.1
s=-
t=0 0
a=group:BUNDLE video
a=msid-semantic: WMS stream_label
m=video 40505 UDP/TLS/RTP/SAVPF 100 101 116 117 96 97 98
c=IN IP4 192.168.56.101
a=rtcp:58120 IN IP4 192.168.56.101
a=candidate:1749055319 1 udp 2122260223 192.168.56.101 40505 typ host generation 0 network-id 2 network-cost 50
a=candidate:3530337216 1 udp 2122194687 159.99.251.183 54696 typ host generation 0 network-id 1 network-cost 50
a=candidate:1749055319 2 udp 2122260222 192.168.56.101 58120 typ host generation 0 network-id 2 network-cost 50
a=candidate:3530337216 2 udp 2122194686 159.99.251.183 36920 typ host generation 0 network-id 1 network-cost 50
a=candidate:650349479 1 tcp 1518280447 192.168.56.101 43174 typ host tcptype passive generation 0 network-id 2 network-cost 50
a=candidate:2632931120 1 tcp 1518214911 159.99.251.183 60014 typ host tcptype passive generation 0 network-id 1 network-cost 50
a=candidate:650349479 2 tcp 1518280446 192.168.56.101 36604 typ host tcptype passive generation 0 network-id 2 network-cost 50
a=candidate:2632931120 2 tcp 1518214910 159.99.251.183 35967 typ host tcptype passive generation 0 network-id 1 network-cost 50
a=ice-ufrag:TiCG/rgYvBn4YaNH
a=ice-pwd:qaxHNknBu6JXN5/Q16kIa/1e
a=fingerprint:sha-256 BD:01:A9:C7:8A:B2:24:43:F8:5B:FA:17:0C:57:95:32:48:DC:46:CD:B6:3F:77:D3:BB:3E:B9:9D:D6:66:9B:E1
a=setup:actpass
a=mid:video
a=extmap:2 urn:ietf:params:rtp-hdrext:toffset
a=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time
a=extmap:4 urn:3gpp:video-orientation
a=sendrecv
a=rtcp-mux
a=rtcp-rsize
a=rtpmap:100 VP8/90000
a=rtcp-fb:100 ccm fir
a=rtcp-fb:100 nack
a=rtcp-fb:100 nack pli
a=rtcp-fb:100 goog-remb
a=rtcp-fb:100 transport-cc
a=rtpmap:101 VP9/90000
a=rtcp-fb:101 ccm fir
a=rtcp-fb:101 nack
a=rtcp-fb:101 nack pli
a=rtcp-fb:101 goog-remb
a=rtcp-fb:101 transport-cc
a=rtpmap:116 red/90000
a=rtpmap:117 ulpfec/90000
a=rtpmap:96 rtx/90000
a=fmtp:96 apt=100
a=rtpmap:97 rtx/90000
a=fmtp:97 apt=101
a=rtpmap:98 rtx/90000
a=fmtp:98 apt=116
a=ssrc-group:FID 1931720387 2078885298
a=ssrc:1931720387 cname:cg6pjT6riTx0Qk81
a=ssrc:1931720387 msid:stream_label video_label
a=ssrc:1931720387 mslabel:stream_label
a=ssrc:1931720387 label:video_label
a=ssrc:2078885298 cname:cg6pjT6riTx0Qk81
a=ssrc:2078885298 msid:stream_label video_label
a=ssrc:2078885298 mslabel:stream_label
a=ssrc:2078885298 label:video_label

===> HTTP request /addicecandidate
res:1
body:{"candidate":"candidate:1 1 UDP 2122252543 192.168.56.1 64890 typ host","sdpMid":"video","sdpMLineIndex":0}
===> HTTP request /addicecandidate
res:1
body:{"candidate":"candidate:0 1 UDP 2122187007 159.99.251.130 64889 typ host","sdpMid":"video","sdpMLineIndex":0}

[025:519] [1855] (stunport.cc:89): Binding request timed out from 192.168.56.101:40505 (eth1)
[025:528] [1855] (stunport.cc:89): Binding request timed out from 159.99.251.183:54696 (eth0)

mpromonet, 2016-08-17:
Hi zackxue,
Last time I tried the videoCapturer doesnot accept H264 as input, it should be an uncompressed format (and the RTSPCapturer needs some improvement). I hope this will change in the future.
You can enable more log to see if packets are received.
If you want a gateway RTSP->WebRTC, you could consider [janus-gateway](https://github.com/meetecho/janus-gateway) that relay the RTP paquet without decoding/encoding.
Regards,
Michel.

zackxue, 2016-08-18:
Thanks !
what this line stand for 
[025:528] 1855: Binding request timed out from 159.99.251.183:54696 (eth0)?