mpromonet/webrtc-streamer #9: Undefined references when linking

dpenkler, 2016-08-30:
webrtc builds ok.
webrtc-streamer compiles ok but fails in link step.
Could be caused by some libraries in webrtc being generated as shared libraries with .so suffix.
ar -rcT libWebRTC__Default.a ../webrtc/src//out/Default/obj/webrtc/librtc_event_log_proto.a ../webrtc/src//out/Default/obj/webrtc/modules/audio_coding/libneteq_unittest_proto.a ../webrtc/src//out/Default/obj/webrtc/modules/audio_processing/libaudioproc_debug_proto.a ../webrtc/src//out/Default/obj/webrtc/modules/audio_processing/libaudioproc_unittest_proto.a ../webrtc/src//out/Default/obj/webrtc/tools/libgraph_proto.a ../webrtc/src//out/Default/obj/webrtc/base/librtc_base_approved.a ../webrtc/src//out/Default/obj/webrtc/base/librtc_task_queue.a ../webrtc/src//out/Default/obj/webrtc/base/librtc_base.a ../webrtc/src//out/Default/obj/testing/gmock/libgmock_main.a ../webrtc/src//out/Default/obj/testing/gmock/libgmock.a ../webrtc/src//out/Default/obj/testing/gtest/libgtest.a ../webrtc/src//out/Default/obj/third_party/openmax_dl/dl/libdl.a ../webrtc/src//out/Default/obj/third_party/usrsctp/libusrsctp.a ../webrtc/src//out/Default/obj/third_party/libsrtp/libsrtp.a ../webrtc/src//out/Default/obj/third_party/libyuv/libyuv.a ../webrtc/src//out/Default/obj/third_party/libjpeg_turbo/libjpeg.a ../webrtc/src//out/Default/obj/third_party/libjpeg_turbo/libsimd_asm.a ../webrtc/src//out/Default/obj/third_party/libjpeg_turbo/libsimd.a ../webrtc/src//out/Default/obj/third_party/libvpx/libvpx.a ../webrtc/src//out/Default/obj/third_party/libvpx/libvpx_yasm.a ../webrtc/src//out/Default/obj/third_party/protobuf/libprotobuf_full.a ../webrtc/src//out/Default/obj/third_party/protobuf/libprotoc_lib.a ../webrtc/src//out/Default/obj/third_party/opus/libopus.a ../webrtc/src//out/Default/obj/third_party/yasm/libyasm_utils.a ../webrtc/src//out/Default/obj/base/third_party/libevent/libevent.a ../webrtc/src//out/Default/obj/build/config/sanitizers/liboptions_sources.a
g++  -o webrtc-server__Default src/PeerConnectionManager.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Default.a -pthread  -lX11 -ldl -lrt 
src/PeerConnectionManager.o: In function `PeerConnectionManager::PeerConnectionManager(std::string const&)':
/home/dave/webrtc-streamer/src/PeerConnectionManager.cpp:42: undefined reference to `Json::Value::Value(Json::ValueType)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::~PeerConnectionManager()':
/home/dave/webrtc-streamer/src/PeerConnectionManager.cpp:46: undefined reference to `Json::Value::~Value()'

``

mpromonet, 2016-08-30:
Hi,
Surely it miss a lots of librairies... 
Did you tried to run `make clean` and `make` ?
If you are right and now webrtc use shared librairy, it needs some modification in the Makefile.
Best Regards,
Michel.

dpenkler, 2016-09-01:
Did `make clean; make` multiple times. It may be caused by them migrating to gn from gyp.
There are only 2 .so files the rest are just .o files in the build directories.
Can you send the complete list of .a files please.

mpromonet, 2016-09-01:
Last time I built webrtc it was based on gyp, I am building with `gn` to see what is missing.

dpenkler, 2016-09-02:
It is now almost building.

g++  -o webrtc-server__Release src/PeerConnectionManager.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a -pthread  -lX11 -ldl -lrt 
src/PeerConnectionManager.o: In function `cricket::FakeVideoCapturer::ResetSupportedFormats(std::vector<cricket::VideoFormat, std::allocator<cricket::VideoFormat> > const&)':
/home/dave/webrtc-streamer/../webrtc/src/webrtc/media/base/fakevideocapturer.h:62: undefined reference to `cricket::VideoCapturer::SetSupportedFormats(std::vector<cricket::VideoFormat, std::allocator<cricket::VideoFormat> > const&)'
src/PeerConnectionManager.o:(.rodata._ZTVN7cricket17FakeVideoCapturerE[_ZTVN7cricket17FakeVideoCapturerE]+0xc8): undefined reference to `non-virtual thunk to cricket::VideoCapturer::AddOrUpdateSink(rtc::VideoSinkInterface<cricket::VideoFrame>*, rtc::VideoSinkWants const&)'
src/PeerConnectionManager.o:(.rodata._ZTVN7cricket17FakeVideoCapturerE[_ZTVN7cricket17FakeVideoCapturerE]+0xd0): undefined reference to `non-virtual thunk to cricket::VideoCapturer::RemoveSink(rtc::VideoSinkInterface<cricket::VideoFrame>*)'
collect2: error: ld returned 1 exit status
Makefile:162: recipe for target 'webrtc-server__Release' failed
make: *** [webrtc-server__Release] Error 1
dave@edgelinegw:~/webrtc-streamer$ 
``
Here is the makefile.
`CC = $(CROSS)g++ $(foreach sysroot,$(SYSROOT),--sysroot=$(sysroot))
AR = $(CROSS)ar
CFLAGS = -W -pthread -g -std=c++11 -Iinc
LDFLAGS = -pthread 

# live555
ifneq ($(wildcard $(SYSROOT)/usr/include/liveMedia/liveMedia.hh),)
	CFLAGS += -DHAVE_LIVE555
	CFLAGS += -I $(SYSROOT)/usr/include/liveMedia  -I $(SYSROOT)/usr/include/groupsock -I $(SYSROOT)/usr/include/UsageEnvironment -I $(SYSROOT)/usr/include/BasicUsageEnvironment/
	LDFLAGS += -lliveMedia -lgroupsock -lUsageEnvironment -lBasicUsageEnvironment
endif

# webrtc
WEBRTCROOT?=../webrtc
WEBRTCBUILD?=Release
WEBRTCLIBPATH=$(WEBRTCROOT)/src/$(GYP_GENERATOR_OUTPUT)/out/$(WEBRTCBUILD)
ifneq ($(wildcard $(WEBRTCROOT)/src/webrtc/media/base/yuvframegenerator.h),)
	CFLAGS += -DHAVE_YUVFRAMEGENERATOR
endif

CFLAGS += -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0
CFLAGS += -I $(WEBRTCROOT)/src -I $(WEBRTCROOT)/src/chromium/src/third_party/jsoncpp/source/include
ifeq ($(WEBRTCBUILD),Debug)
	CFLAGS += -D_GLIBCXX_DEBUG=1
endif
LDFLAGS += -lX11 -ldl -lrt

TARGET = webrtc-server_$(GYP_GENERATOR_OUTPUT)_$(WEBRTCBUILD)
all: $(TARGET)

WEBRTC_LIB = $(shell find $(WEBRTCLIBPATH) -name '*.a')
WEBRTC_OBJ = $(shell find \
	$(WEBRTCLIBPATH)/obj/webrtc/stats/rtc_stats					       \
	$(WEBRTCLIBPATH)/obj/webrtc/video/video					       \
	$(WEBRTCLIBPATH)/obj/webrtc/rtc_event_log_proto				       \
	$(WEBRTCLIBPATH)/obj/webrtc/rtc_event_log_parser				       \
	$(WEBRTCLIBPATH)/obj/webrtc/system_wrappers/field_trial_default		       \
	$(WEBRTCLIBPATH)/obj/webrtc/system_wrappers/system_wrappers			       \
	$(WEBRTCLIBPATH)/obj/webrtc/system_wrappers/metrics_default			       \
	$(WEBRTCLIBPATH)/obj/webrtc/system_wrappers/cpu_features_linux			       \
	$(WEBRTCLIBPATH)/obj/webrtc/pc/rtc_pc						       \
	$(WEBRTCLIBPATH)/obj/webrtc/api/libjingle_peerconnection			       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_capture/video_capture_internal_impl	       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_capture/video_capture		       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_capture/video_capture_module		       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_coding/webrtc_h264			       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_coding/video_coding			       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_coding/webrtc_i420			       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_coding/webrtc_vp8			       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_coding/webrtc_vp9			       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_coding/video_coding_utility		       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_processing/video_processing_sse2	       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/video_processing/video_processing		       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/audio_device/audio_device			       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/remote_bitrate_estimator/bwe_simulator	       \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/remote_bitrate_estimator/remote_bitrate_estimator \
	$(WEBRTCLIBPATH)/obj/webrtc/modules/pacing/pacing				       \
[... 84 lines cut ...]
	$(WEBRTCLIBPATH)/obj/third_party/opus/opus_compare				       \
	$(WEBRTCLIBPATH)/obj/third_party/opus/opus					       \
	$(WEBRTCLIBPATH)/obj/third_party/yasm/genmacro					       \
	$(WEBRTCLIBPATH)/obj/third_party/yasm/re2c					       \
	$(WEBRTCLIBPATH)/obj/third_party/yasm/genperf					       \
	$(WEBRTCLIBPATH)/obj/third_party/yasm/genstring				       \
	$(WEBRTCLIBPATH)/obj/third_party/yasm/genmodule				       \
	$(WEBRTCLIBPATH)/obj/third_party/yasm/genversion				       \
	$(WEBRTCLIBPATH)/obj/third_party/yasm/yasm_utils				       \
	$(WEBRTCLIBPATH)/obj/third_party/yasm/yasm					       \
	$(WEBRTCLIBPATH)/obj/base/third_party/libevent/libevent					\
		-name '*.o')
libWebRTC_$(GYP_GENERATOR_OUTPUT)_$(WEBRTCBUILD).a: $(WEBRTC_OBJ)
	$(AR) -rcT $@ $^

src/%.o: src/%.cpp
	$(CC) -o $@ -c $^ $(CFLAGS) 

FILES = $(wildcard src/*.cpp)
$(TARGET): $(subst .cpp,.o,$(FILES)) libWebRTC_$(GYP_GENERATOR_OUTPUT)_$(WEBRTCBUILD).a
	$(CC) -o $@ $^ $(LDFLAGS) 

clean:
	rm -f src/*.o libWebRTC_$(GYP_GENERATOR_OUTPUT)_$(WEBRTCBUILD).a $(TARGET)
`

commit d2a18cd9 (2016-09-02): fix #9: update makefile to build with webrtc
closed 2016-09-02 by mpromonet