mpromonet/webrtc-streamer #10: Build Error

wyounglee, 2016-09-06:
When I compile the latest one with latest webrtc build, I got the following error.
What's problem?

g++  -o src/PeerConnectionManager.o -c src/PeerConnectionManager.cpp -W -pthread -g -std=c++11 -Iinc -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -I ../webrtc_linux/src -I ../webrtc_linux/src/chromium/src/third_party/jsoncpp/source/include -D_GLIBCXX_DEBUG=1 
g++  -o src/HttpServerRequestHandler.o -c src/HttpServerRequestHandler.cpp -W -pthread -g -std=c++11 -Iinc -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -I ../webrtc_linux/src -I ../webrtc_linux/src/chromium/src/third_party/jsoncpp/source/include -D_GLIBCXX_DEBUG=1 
g++  -o src/main.o -c src/main.cpp -W -pthread -g -std=c++11 -Iinc -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -I ../webrtc_linux/src -I ../webrtc_linux/src/chromium/src/third_party/jsoncpp/source/include -D_GLIBCXX_DEBUG=1 
g++  -o webrtc-server__Default src/PeerConnectionManager.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Default.a -pthread  -lX11 -ldl -lrt  
../webrtc_linux/src/out/Default/obj/testing/libfuzzer/libfuzzer_main/unittest_main.o: In function `main':
./out/Default/../../testing/libfuzzer/unittest_main.cc:36: multiple definition of `main'
src/main.o:/home/astre/work/webrtc/webrtc-streamer/src/main.cpp:23: first defined here
../webrtc_linux/src/out/Default/obj/third_party/expat/expat_xml_parse_fuzzer/expat_xml_parse_fuzzer.o: In function `LLVMFuzzerTestOneInput':
./out/Default/../../third_party/expat/fuzz/expat_xml_parse_fuzzer.cc:23: undefined reference to `XML_ParserCreateNS'
./out/Default/../../third_party/expat/fuzz/expat_xml_parse_fuzzer.cc:23: undefined reference to `XML_ParserCreate'
./out/Default/../../third_party/expat/fuzz/expat_xml_parse_fuzzer.cc:24: undefined reference to `XML_Parse'
./out/Default/../../third_party/expat/fuzz/expat_xml_parse_fuzzer.cc:25: undefined reference to `XML_ParserFree'
collect2: error: ld returned 1 exit status
Makefile:44: recipe for target 'webrtc-server__Default' failed
make: *** [webrtc-server__Default] Error 1

wyounglee, 2016-09-06:
It will be okay with webrtc Release build. It happened only with debug build.

mpromonet, 2016-09-07:
Hi,
I just built with webrtc in debug https://chromium.googlesource.com/external/webrtc/+/37062ed16ff82de73cc6b4f9e71a60afe67d3166 without the problem you described.
Did you run `make clean` between Release build and Debug build ?
Best Regards,
Michel.

wyounglee, 2016-09-08:
My build is 46224dddfc39d6ebaa9c3660fdd2808eb3ce861e.
I could not build it even if I download new one (git clone).

mpromonet, 2016-09-20:
Hi,
I tried to get same release as you, but when I run on a clone of https://chromium.googlesource.com/external/webrtc.git: 
      `git checkout 46224dddfc39d6ebaa9c3660fdd2808eb3ce861e`

I get 

> fatal: reference is not a tree: 46224dddfc39d6ebaa9c3660fdd2808eb3ce861e

What is the git repository of this SHA1 ?
Regards,
Michel.

closed 2016-09-26 by wyounglee