mpromonet/webrtc-streamer #12: Web CAM is not working.

wyounglee, 2016-09-26:
In my previous Ubuntu, there is no issue for running webrtc-streamer.
However, there is no WebCAM selection in new Ubuntu installation.
I used the same webrtc and webrtc-streamer source and successful compilation.
Also, two log files were exactly same.
Do you have any comment or suggestion?

mpromonet, 2016-09-28:
Hi,
You should try to run webrtc-streamer with argument `-vvv`, you should have more details why the video device are not enumerated.
Best Regards,
Michel.

wyounglee, 2016-09-28:
Thanks for updating.
As I mentioned above, two logs (as you described) are exactly same. ^^

mpromonet, 2016-09-28:
Hi,
If you see the device enumeration with your webcam in the verbose log, the combobox should show it.
You could call /getDeviceList to  get the list of detected device.
Regards,
Michel.

franck-talbart, 2016-12-22:
Hi,

I can't see my webcam either. /getDeviceList is returning "null".
My webcam is available here:
ls /dev/video0 
/dev/video0

Here is the log. Am I missing something?
[log.txt](https://github.com/mpromonet/webrtc-streamer/files/669155/log.txt)

closed 2017-01-28 by mpromonet