mpromonet/webrtc-streamer #13: webrtc-streamer core dump

saket424, 2016-11-26:
Is there a specific stable commended version/branch of webrtc libs that is recommended to compile webrtc-streamer? The latest git HEAD is dumping core. Any ideas or anyone else having this issue ?
Does webrtc-streamer have to have any other associated v4l2loopback/compress software dependency?

I was using git HEAD 
commit 71b9b58a3a3ac651e5a5c5d095411620bb425659
Author: solenberg <solenberg@webrtc.org>
Date:   Fri Nov 25 11:45:05 2016 -0800

Successfully compiled webrtc for linux using instructions available here
https://github.com/ipop-project/ipop-project.github.io/wiki/Building-the-WebRTC-lib-for-Linux

gn gen out/Release
ninja -C out/Release

~/rtsp-stuff/webrtc-streamer$ ./webrtc-server__Release -vvv
Logger level:1
[000:000] [1522] (webrtcvoiceengine.cc:554): WebRtcVoiceEngine::WebRtcVoiceEngine
[000:001] [1522] (webrtcvoiceengine.cc:561): Supported send codecs in order of preference:
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: opus/48000/2 (111)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: ISAC/16000/1 (103)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: ISAC/32000/1 (104)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: G722/8000/1 (9)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: ILBC/8000/1 (102)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: PCMU/8000/1 (0)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: PCMA/8000/1 (8)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: CN/32000/1 (106)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: CN/16000/1 (105)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: CN/8000/1 (13)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: telephone-event/48000/1 (110)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: telephone-event/32000/1 (112)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: telephone-event/16000/1 (113)
[000:001] [1522] (webrtcvoiceengine.cc:314): Adding supported codec: telephone-event/8000/1 (126)
[000:001] [1522] (webrtcvoiceengine.cc:564): opus/48000/2 (111)
[000:001] [1522] (webrtcvoiceengine.cc:564): ISAC/16000/1 (103)
[000:001] [1522] (webrtcvoiceengine.cc:564): ISAC/32000/1 (104)
[000:001] [1522] (webrtcvoiceengine.cc:564): G722/8000/1 (9)
[000:001] [1522] (webrtcvoiceengine.cc:564): ILBC/8000/1 (102)
[000:001] [1522] (webrtcvoiceengine.cc:564): PCMU/8000/1 (0)
[000:001] [1522] (webrtcvoiceengine.cc:564): PCMA/8000/1 (8)
[000:001] [1522] (webrtcvoiceengine.cc:564): CN/32000/1 (106)
[000:001] [1522] (webrtcvoiceengine.cc:564): CN/16000/1 (105)
[000:001] [1522] (webrtcvoiceengine.cc:564): CN/8000/1 (13)
[000:001] [1522] (webrtcvoiceengine.cc:564): telephone-event/48000/1 (110)
[000:001] [1522] (webrtcvoiceengine.cc:564): telephone-event/32000/1 (112)
[000:001] [1522] (webrtcvoiceengine.cc:564): telephone-event/16000/1 (113)
[000:001] [1522] (webrtcvoiceengine.cc:564): telephone-event/8000/1 (126)
[000:001] [1522] (webrtcvoiceengine.cc:567): Supported recv codecs in order of preference:
[000:001] [1522] (webrtcvoiceengine.cc:570): opus/48000/2 (111)
[000:001] [1522] (webrtcvoiceengine.cc:570): isac/16000/1 (103)
[000:002] [1522] (webrtcvoiceengine.cc:570): isac/32000/1 (104)
[000:002] [1522] (webrtcvoiceengine.cc:570): G722/8000/1 (9)
[000:002] [1522] (webrtcvoiceengine.cc:570): iLBC/8000/1 (102)
[000:002] [1522] (webrtcvoiceengine.cc:570): PCMU/8000/1 (0)
[000:002] [1522] (webrtcvoiceengine.cc:570): PCMA/8000/1 (8)
[000:002] [1522] (webrtcvoiceengine.cc:570): cn/32000/1 (106)
[000:002] [1522] (webrtcvoiceengine.cc:570): cn/16000/1 (105)
[000:002] [1522] (webrtcvoiceengine.cc:570): cn/8000/1 (13)
[000:002] [1522] (webrtcvoiceengine.cc:570): telephone-event/48000/1 (110)
[000:002] [1522] (webrtcvoiceengine.cc:570): telephone-event/32000/1 (112)
[000:002] [1522] (webrtcvoiceengine.cc:570): telephone-event/16000/1 (113)
[000:002] [1522] (webrtcvoiceengine.cc:570): telephone-event/8000/1 (126)
[000:002] [1522] (webrtcvoiceengine.cc:578): VoiceEngine 4.1.0
[000:002] [1522] (audio_device_impl.cc:86): Create
[000:002] [1522] (audio_device_buffer.cc:73): AudioDeviceBuffer::ctor
[000:002] [1522] (audio_device_impl.cc:133): AudioDeviceModuleImpl
[000:002] [1522] (audio_device_impl.cc:141): CheckPlatform
[000:002] [1522] (audio_device_impl.cc:155): current platform is Linux
[000:002] [1522] (audio_device_impl.cc:182): CreatePlatformSpecificObjects
[000:002] [1522] (audio_device_impl.cc:1876): PlatformAudioLayer
[000:002] [1522] (audio_device_impl.cc:285): attempting to use the Linux PulseAudio APIs...
[... 69 lines cut ...]
[000:013] [1522] (webrtcvoiceengine.cc:1024): webrtc: SetRecordingChannel() unable to set the recording channel (error=10028)
[000:013] [1522] (audio_device_impl.cc:1323): SetRecordingDevice(0)
[000:014] [1522] (audio_device_impl.cc:558): InitMicrophone
[000:014] [1522] (audio_device_impl.cc:923): StereoRecordingIsAvailable
[000:014] [1522] (audio_device_impl.cc:933): output: 1
[000:014] [1522] (audio_device_impl.cc:942): SetStereoRecording(1)
[000:014] [1522] (audio_device_buffer.cc:213): SetRecordingChannels(2)
[000:015] [1522] (audio_device_impl.cc:1431): Playing
[000:015] [1522] (audio_device_impl.cc:1228): SetPlayoutDevice(0)
[000:015] [1522] (audio_device_impl.cc:548): InitSpeaker
[000:015] [1522] (audio_device_impl.cc:1037): StereoPlayoutIsAvailable
[000:016] [1522] (audio_device_impl.cc:1047): output: 1
[000:016] [1522] (audio_device_impl.cc:1056): SetStereoPlayout(1)
[000:016] [1522] (audio_device_buffer.cc:220): SetPlayoutChannels(2)
[000:016] [1522] (webrtcvoiceengine.cc:961): Set microphone to (id=0) and speaker to (id=0)
[000:017] [1522] (audio_device_impl.cc:1497): RegisterAudioCallback
[000:017] [1522] (audio_device_buffer.cc:88): RegisterAudioCallback
[000:017] [1522] (audio_device_impl.cc:1497): RegisterAudioCallback
[000:017] [1522] (audio_device_buffer.cc:88): RegisterAudioCallback
[000:017] [1522] (webrtcvideoengine2.cc:499): WebRtcVideoEngine2::WebRtcVideoEngine2()
[000:017] [1522] (webrtcvideoengine2.cc:641): Internally supported codecs: {VideoCodec[0:VP8], VideoCodec[0:VP9], VideoCodec[0:red], VideoCodec[0:ulpfec]}
[000:018] [1522] (webrtcvideoengine2.cc:508): WebRtcVideoEngine2::Init
Segmentation fault (core dumped)

[305412.035096] webrtc-server__[1462]: segfault at ff00000001 ip 00007f4a9623958f sp 00007ffd221e44c8 error 6 in libstdc++.so.6.0.21[7f4a9619b000+172000]

saket424, 2016-11-26:
I stuck an strace to get more info on what preceded the core dump. Any insights on why this coredump will be much appreciated

strace ./webrtc-server__Release -vvv
...
[000:010] [13908] (webrtcvoiceengine.cc:960): Set microphone to (id=0) and speaker to (id=0)
[000:010] [13908] (webrtcvideoengine2.cc:561): WebRtcVideoEngine2::WebRtcVideoEngine2()
[000:010] [13908] (webrtcvideoengine2.cc:642): Supported codecs: {VideoCodec[100:VP8], VideoCodec[96:rtx], VideoCodec[101:VP9], VideoCodec[97:rtx], VideoCodec[116:red], VideoCodec[98:rtx], VideoCodec[117:ulpfec]}
)          = 1 (in [3])
getsockopt(3, SOL_SOCKET, SO_ERROR, 0x7ffe8f9972ac, 0x7ffe8f9972a8) = -1 ENOTSOCK (Socket operation on non-socket)
read(3, "\0", 4)                        = 1
write(4, "\0", 1)                       = 1
write(8, "\0", 1)                       = 1
futex(0x7ffe8f997944, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 3, {1480182186, 194846000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997918, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x7ffe8f997944, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 5, {1480182186, 194947000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997918, FUTEX_WAKE_PRIVATE, 1) = 0
close(7)                                = 0
close(8)                                = 0
pipe([7, 8])                            = 0
futex(0x7ffe8f997894, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, {1480182186, 195293000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997868, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x1cc5ddc, FUTEX_CMP_REQUEUE_PRIVATE, 1, 2147483647, 0x1cc5db0, 4) = 1
futex(0x1cc5db0, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x1cc5bc8, FUTEX_WAIT_PRIVATE, 2, NULL) = -1 EAGAIN (Resource temporarily unavailable)
futex(0x1cc5bc8, FUTEX_WAKE_PRIVATE, 1) = 0
write(8, "\0", 1)                       = 1
futex(0x7ffe8f997894, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 3, {1480182186, 195560000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997868, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x7ffe8f997894, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 5, {1480182186, 195656000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997868, FUTEX_WAKE_PRIVATE, 1) = 0
close(7)                                = 0
close(8)                                = 0
pipe([7, 8])                            = 0
futex(0x7ffe8f997564, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, {1480182186, 196027000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997538, FUTEX_WAKE_PRIVATE, 1) = 0
write(6, "\0", 1)                       = 1
futex(0x1cc5a10, FUTEX_WAKE_PRIVATE, 1) = 1
write(8, "\0", 1)                       = 1
futex(0x7ffe8f997564, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 3, {1480182186, 196237000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997538, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x7ffe8f997564, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 5, {1480182186, 196333000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997538, FUTEX_WAKE_PRIVATE, 1) = 0
close(7)                                = 0
close(8)                                = 0
pipe([7, 8])                            = 0
futex(0x7ffe8f997574, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 1, {1480182186, 196587000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997548, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x1cc5ddc, FUTEX_CMP_REQUEUE_PRIVATE, 1, 2147483647, 0x1cc5db0, 6) = 1
futex(0x1cc5db0, FUTEX_WAKE_PRIVATE, 1) = 1
select(4, [3], [], NULL, NULL[000:013] [13908] (webrtcvideoengine2.cc:570): WebRtcVideoEngine2::Init
)          = 1 (in [3])
getsockopt(3, SOL_SOCKET, SO_ERROR, 0x7ffe8f996edc, 0x7ffe8f996ed8) = -1 ENOTSOCK (Socket operation on non-socket)
read(3, "\0", 4)                        = 1
write(4, "\0", 1)                       = 1
write(8, "\0", 1)                       = 1
futex(0x7ffe8f997574, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 3, {1480182186, 196923000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997548, FUTEX_WAKE_PRIVATE, 1) = 0
futex(0x7ffe8f997574, FUTEX_WAIT_BITSET_PRIVATE|FUTEX_CLOCK_REALTIME, 5, {1480182186, 197018000}, ffffffff) = -1 ETIMEDOUT (Connection timed out)
futex(0x7ffe8f997548, FUTEX_WAKE_PRIVATE, 1) = 0
close(7)                                = 0
close(8)                                = 0
socket(PF_INET, SOCK_STREAM, IPPROTO_IP) = 7
fcntl(7, F_GETFL)                       = 0x2 (flags O_RDWR)
fcntl(7, F_SETFL, O_RDWR|O_NONBLOCK)    = 0
bind(7, {sa_family=AF_INET, sin_port=htons(8000), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
getsockname(7, {sa_family=AF_INET, sin_port=htons(8000), sin_addr=inet_addr("0.0.0.0")}, [16]) = 0
listen(7, 5)                            = 0
getsockname(7, {sa_family=AF_INET, sin_port=htons(8000), sin_addr=inet_addr("0.0.0.0")}, [16]) = 0
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0xff00000001} ---
+++ killed by SIGSEGV (core dumped) +++
Segmentation fault (core dumped)

saket424, 2016-11-29:
Anyone running webrtc-streamer successfully, can you comment on what version of webrtc code and webrtc-streamer code you are using ? Thanks in advance

mpromonet, 2016-11-29:
Hi,
Last timem I tried, it crashed after a few seconds, webrtc code is changing so quickly, it is difficult to follow its changes.
Did you get a backtrace withh a debug build ?
Depending on webrtc building options, it may be  needs to remove -D_GLIBCXX_USE_CXX11_ABI=0 
Regards,
Michel.

saket424, 2016-11-29:
Michel,
I think the crash is not within the webrtc code but in the HTTP server socket code itself. I'll try to get a stack trace using the method you suggested.

I assume no options are needed to launch webrtc-server__Release tell me thse. Can you tell me which version of webrtc you are using so i can match your version and replicate your success

Regards,
Anand

saket424, 2016-11-29:
when i try to compile without -D_GLIBCXX_USE_CXX11_ABI=0, I get several linker errors

g++  -o webrtc-server__Release src/PeerConnectionManager.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a -pthread  -lliveMedia -lgroupsock -lUsageEnvironment -lBasicUsageEnvironment -lX11 -ldl -lrt  
src/PeerConnectionManager.o: In function `PeerConnectionManager::setAnswer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:111: undefined reference to `Json::Reader::parse(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Json::Value&, bool)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:119: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:120: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:126: undefined reference to `webrtc::CreateSessionDescription(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, webrtc::SdpParseError*)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::addIceCandidate(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:152: undefined reference to `Json::Reader::parse(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Json::Value&, bool)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:161: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:162: undefined reference to `rtc::GetIntFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int*)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:163: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:169: undefined reference to `webrtc::CreateIceCandidate(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, webrtc::SdpParseError*)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::getOffer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:326: undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:327: undefined reference to `Json::Value::Value(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:330: undefined reference to `Json::StyledWriter::write[abi:cxx11](Json::Value const&)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::call(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:365: undefined reference to `Json::Reader::parse(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Json::Value&, bool)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:373: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
/media/anand/webrtc-streamer/src/PeerConnectionManager.cpp:374: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
[... cut ...]
src/main.o: In function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > cricket::VectorToString<cricket::DataCodec>(std::__debug::vector<cricket::DataCodec, std::allocator<cricket::DataCodec> > const&)':
/media/anand/webrtc-streamer/../webrtc/src/webrtc/media/base/mediachannel.h:79: undefined reference to `cricket::DataCodec::ToString[abi:cxx11]() const'
collect2: error: ld returned 1 exit status
Makefile:44: recipe for target 'webrtc-server__Release' failed
make: *** [webrtc-server__Release] Error 1

mpromonet, 2016-11-29:
Hi,
Using webrtc commit b7510d3f49a0a7a91221cb070dd637aa5ae62e91, the crash occurs in `void SendSideBandwidthEstimation::UpdateEstimate(int64_t now_ms)`.
Commenting the call of eventlog seems to solve the problem.
```
    event_log_->LogBwePacketLossEvent(capped_bitrate, last_fraction_loss_,
                                      expected_packets_since_last_loss_update_);

```
Maybe some initialization of event_log is missing somewhere.
Regards,
Michel.

saket424, 2016-11-30:
My stack trace shows a segfault at src/main.cpp:81
                        // connect httpserver to a request handler
                        HttpServerRequestHandler http(&httpServer, &webRtcServer);

(gdb) run
Starting program: /home/anand/rtsp-stuff/webrtc-streamer/webrtc-server__Release 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Logger level:4
[New Thread 0x7ffff61dd700 (LWP 1323)]
[New Thread 0x7ffff59dc700 (LWP 1324)]
[New Thread 0x7ffff7fc7700 (LWP 1325)]
[New Thread 0x7ffff51db700 (LWP 1326)]
[New Thread 0x7fffee6cf700 (LWP 1327)]
[New Thread 0x7ffff413f700 (LWP 1328)]
[New Thread 0x7fffe9ecd700 (LWP 1329)]
[000:000] [1324] (audio_device_generic.cc:51): virtual bool webrtc::AudioDeviceGeneric::BuiltInAECIsAvailable() const: Not supported on this platform
[000:000] [1324] (audio_device_generic.cc:61): virtual bool webrtc::AudioDeviceGeneric::BuiltInAGCIsAvailable() const: Not supported on this platform
[000:000] [1324] (audio_device_generic.cc:71): virtual bool webrtc::AudioDeviceGeneric::BuiltInNSIsAvailable() const: Not supported on this platform

Thread 1 "webrtc-server__" received signal SIGSEGV, Segmentation fault.
0x00007ffff73ad58f in std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff73ad58f in std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00000000004279df in std::__cxx1998::list<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*, std::allocator<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*> >::_M_insert<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*>(std::__cxx1998::_List_iterator<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*>, sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*&&) (
Python Exception <class 'ValueError'> Cannot find type std::__cxx1998::list<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*, std::allocator<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*> >::iterator::_Node: 
    this=0x7fffffffdbf0, __position=)
    at /usr/include/c++/5/bits/stl_list.h:1764
#2  0x0000000000426d59 in std::__cxx1998::list<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*, std::allocator<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*> >::push_back(sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*&&) (
    this=0x7fffffffdbf0, 
    __x=<unknown type in /home/anand/rtsp-stuff/webrtc-streamer/webrtc-server__Release, CU 0x129bc1, DIE 0x1b0a39>) at /usr/include/c++/5/bits/stl_list.h:1094
#3  0x0000000000426837 in sigslot::signal2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>::connect<HttpServerRequestHandler> (
    this=0x7fffffffdbc8, pclass=0x7fffffffda40, pmemfun=
---Type <return> to continue, or q <return> to quit---
    Handler::*)(HttpServerRequestHandler * const, rtc::HttpServer *, rtc::HttpServerTransaction *)) 0x421800 <HttpServerRequestHandler::OnRequest(rtc::HttpServer*, rtc::HttpServerTransaction*)>) at ../webrtc/src/webrtc/base/sigslot.h:2333
#4  0x0000000000426570 in HttpServerRequestHandler::HttpServerRequestHandler (this=0x7fffffffda40, server=0x7fffffffdb50, webRtcServer=0x7fffffffdab0) at inc/HttpServerRequestHandler.h:22
#5  0x000000000042475e in main (argc=1, argv=0x7fffffffde98) at src/main.cpp:81

saket424, 2016-11-30:
when I use webrtc branch 51 (commit 457b1b0264399be73c285e40f2e965e624d8d1b5
Author: Henrik Lundin <henrik.lundin@webrtc.org>
Date:   Fri May 20 10:55:31 2016 +0200)

and

webrtc-streamer (commit c4e52a03ad4d11a4fb55ab4df57aab31c97525cd
Author: mpromonet <michel.promonet@free.fr>
Date:   Fri Sep 2 22:42:01 2016 +0200) 

it does not crash and works. The newer HEAD gives me the stack trace above

saket424, 2016-12-02:
Michel,
Any insight into why I am getting a crash in a different spot in the HttpServerRequestHandler even when I use exactly the same webrtc commit  b7510d3f49a0a7a91221cb070dd637aa5ae62e91 as you are? Any others have success with this revisiion or HEAD? If so what steps are you using? I am just using what is documented in https://webrtc.org/native-code/development/ to build the webrtc Default library

(gdb) run -H0.0.0.0:1111
Starting program: /home/anand/rtsp-stuff/webrtc-streamer/webrtc-server__Default -H0.0.0.0:1111
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Logger level:4
[New Thread 0x7ffff61dd700 (LWP 18174)]
[New Thread 0x7ffff59dc700 (LWP 18175)]
[New Thread 0x7ffff7fc7700 (LWP 18176)]
[New Thread 0x7ffff51db700 (LWP 18177)]
[New Thread 0x7fffee6cf700 (LWP 18178)]
[New Thread 0x7ffff413f700 (LWP 18179)]
[New Thread 0x7fffedece700 (LWP 18180)]
[000:000] [18175] (audio_device_generic.cc:51): virtual bool webrtc::AudioDeviceGeneric::BuiltInAECIsAvailable() const: Not supported on this platform
[000:001] [18175] (audio_device_generic.cc:61): virtual bool webrtc::AudioDeviceGeneric::BuiltInAGCIsAvailable() const: Not supported on this platform
[000:001] [18175] (audio_device_generic.cc:71): virtual bool webrtc::AudioDeviceGeneric::BuiltInNSIsAvailable() const: Not supported on this platform

Thread 1 "webrtc-server__" received signal SIGSEGV, Segmentation fault.
0x00007ffff73ad58f in std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff73ad58f in std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
Python Exception <class 'ValueError'> Cannot find type std::__cxx1998::list<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*, std::allocator<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*> >::iterator::_Node: 
#1  0x00000000004279df in std::__cxx1998::list<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*, std::allocator<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*> >::_M_insert<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*>(std::__cxx1998::_List_iterator<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*>, sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*&&) (this=0x7fffffffdbc0, __position=) at /usr/include/c++/5/bits/stl_list.h:1764
#2  0x0000000000426d59 in std::__cxx1998::list<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*, std::allocator<sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*> >::push_back(sigslot::_connection_base2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>*&&) (this=0x7fffffffdbc0, 
    __x=<unknown type in /home/anand/rtsp-stuff/webrtc-streamer/webrtc-server__Default, CU 0x129bc1, DIE 0x1b0a39>) at /usr/include/c++/5/bits/stl_list.h:1094
#3  0x0000000000426837 in sigslot::signal2<rtc::HttpServer*, rtc::HttpServerTransaction*, sigslot::single_threaded>::connect<HttpServerRequestHandler> (this=0x7fffffffdb98, pclass=0x7fffffffda10, 
    pmemfun=
    (void (HttpServerRequestHandler::*)(HttpServerRequestHandler * const, rtc::HttpServer *, rtc::HttpServerTransaction *)) 0x421800 <HttpServerRequestHandler::OnRequest(rtc::HttpServer*, rtc::HttpServerTransaction*)>) at ../webrtc/src/webrtc/base/sigslot.h:2333
#4  0x0000000000426570 in HttpServerRequestHandler::HttpServerRequestHandler (this=0x7fffffffda10, server=0x7fffffffdb20, webRtcServer=0x7fffffffda80) at inc/HttpServerRequestHandler.h:22
#5  0x000000000042475e in main (argc=2, argv=0x7fffffffde68) at src/main.cpp:81

mpromonet, 2016-12-04:
This is probably cause mixing different STL, I usually build webrtc with `use_sysroot=false`, sometime also with `is_clang=false`

saket424, 2016-12-06:
Hi Michel,
I tried using use_sysroot=false and is_debug=false
gn gen out/Release "--args=is_debug=false use_sysroot=false target_cpu=\"x64\""

I get linker errors when I build for Release (I am hoping this serves as a clue to you, to tell me what i am doing wrong. No linker error when compile Debug but core dump

g++  -o webrtc-server__Release src/PeerConnectionManager.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a -pthread  -lliveMedia -lgroupsock -lUsageEnvironment -lBasicUsageEnvironment -lX11 -ldl -lrt  
src/PeerConnectionManager.o: In function `PeerConnectionManager::addIceCandidate(std::string const&, std::string const&)':
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:111: undefined reference to `Json::Reader::parse(std::string const&, Json::Value&, bool)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:120: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:121: undefined reference to `rtc::GetIntFromJsonObject(Json::Value const&, std::string const&, int*)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:122: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:128: undefined reference to `webrtc::CreateIceCandidate(std::string const&, int, std::string const&, webrtc::SdpParseError*)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::call(std::string&, std::string const&, std::string const&)':
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:268: undefined reference to `Json::Reader::parse(std::string const&, Json::Value&, bool)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:276: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:277: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:283: undefined reference to `webrtc::CreateSessionDescription(std::string const&, std::string const&, webrtc::SdpParseError*)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:329: undefined reference to `Json::Value::Value(std::string const&)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:330: undefined reference to `Json::Value::Value(std::string const&)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:333: undefined reference to `Json::StyledWriter::write(Json::Value const&)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::PeerConnectionObserver::OnIceCandidate(webrtc::IceCandidateInterface const*)':
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:390: undefined reference to `Json::Value::Value(std::string const&)'
/home/anand/rtsp-stuff/webrtc-streamer/src/PeerConnectionManager.cpp:392: undefined reference to `Json::Value::Value(std::string const&)'
src/PeerConnectionManager.o: In function `std::string cricket::VectorToString<cricket::AudioCodec>(std::vector<cricket::AudioCodec, std::allocator<cricket::AudioCodec> > const&)':
/home/anand/rtsp-stuff/webrtc-streamer/../webrtc/src/webrtc/media/base/mediachannel.h:78: undefined reference to `cricket::AudioCodec::ToString() const'
src/PeerConnectionManager.o: In function `std::string cricket::VectorToString<webrtc::RtpExtension>(std::vector<webrtc::RtpExtension, std::allocator<webrtc::RtpExtension> > const&)':
/home/anand/rtsp-stuff/webrtc-streamer/../webrtc/src/webrtc/media/base/mediachannel.h:78: undefined reference to `webrtc::RtpExtension::ToString() const'
src/PeerConnectionManager.o: In function `std::string cricket::VectorToString<cricket::DataCodec>(std::vector<cricket::DataCodec, std::allocator<cricket::DataCodec> > const&)':
/home/anand/rtsp-stuff/webrtc-streamer/../webrtc/src/webrtc/media/base/mediachannel.h:78: undefined reference to `cricket::DataCodec::ToString() const'
src/PeerConnectionManager.o: In function `rtc::ArrayView<int>::CheckInvariant() const':
/home/anand/rtsp-stuff/webrtc-streamer/../webrtc/src/webrtc/base/array_view.h:141: undefined reference to `rtc::FatalMessage::FatalMessage(char const*, int, std::string*)'
src/PeerConnectionManager.o: In function `rtc::ArrayView<bool>::CheckInvariant() const':
/home/anand/rtsp-stuff/webrtc-streamer/../webrtc/src/webrtc/base/array_view.h:141: undefined reference to `rtc::FatalMessage::FatalMessage(char const*, int, std::string*)'
src/HttpServerRequestHandler.o: In function `HttpServerRequestHandler::OnRequest(rtc::HttpServer*, rtc::HttpServerTransaction*)':
/home/anand/rtsp-stuff/webrtc-streamer/src/HttpServerRequestHandler.cpp:23: undefined reference to `rtc::HttpRequestData::getRelativeUri(std::string*, std::string*) const'
/home/anand/rtsp-stuff/webrtc-streamer/src/HttpServerRequestHandler.cpp:37: undefined reference to `rtc::HttpData::hasHeader(std::string const&, std::string*) const'
/home/anand/rtsp-stuff/webrtc-streamer/src/HttpServerRequestHandler.cpp:41: undefined reference to `Json::StyledWriter::write(Json::Value const&)'
[... cut ...]
/home/anand/rtsp-stuff/webrtc-streamer/../webrtc/src/webrtc/media/base/mediachannel.h:78: undefined reference to `webrtc::RtpExtension::ToString() const'
src/main.o: In function `std::string cricket::VectorToString<cricket::DataCodec>(std::vector<cricket::DataCodec, std::allocator<cricket::DataCodec> > const&)':
/home/anand/rtsp-stuff/webrtc-streamer/../webrtc/src/webrtc/media/base/mediachannel.h:78: undefined reference to `cricket::DataCodec::ToString() const'
collect2: error: ld returned 1 exit status
Makefile:43: recipe for target 'webrtc-server__Release' failed
make: *** [webrtc-server__Release] Error 1

saket424, 2016-12-19:
Michel,
I continue to get core dumps with gn generated ninja files and no core dump with the the now deprecated gyp generated ninja files from some months ago (chrome54)

Can you please tell me the exact gn commands that you use to generate the build. Also I am using ubuntu16.04 64-bit Intel -- may i ask which platform you use for the compile on linux ?

Thanks

saket424, 2016-12-19:
I am using g++ 5.4.0 

anand@alien15r2:~/webrtc-stuff/webrtc-streamer$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/5/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 5.4.0-6ubuntu1~16.04.4' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.4)

saket424, 2016-12-20:
When I used g++-4.9, no core dump but getDeviceList returns nothing. Can you please let me know which version of g++ you are using?

anand@alien15r2:~/webrtc-stuff/webrtc-streamer$ make
g++-4.9  -o webrtc-server__Release src/PeerConnectionManager.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a -pthread  -lX11 -ldl -lrt  -lexpat
anand@alien15r2:~/webrtc-stuff/webrtc-streamer$ ./webrtc-server__Release
Logger level:4
[000:000] [19327] (audio_device_generic.cc:51): virtual bool webrtc::AudioDeviceGeneric::BuiltInAECIsAvailable() const: Not supported on this platform
[000:000] [19327] (audio_device_generic.cc:61): virtual bool webrtc::AudioDeviceGeneric::BuiltInAGCIsAvailable() const: Not supported on this platform
[000:000] [19327] (audio_device_generic.cc:71): virtual bool webrtc::AudioDeviceGeneric::BuiltInNSIsAvailable() const: Not supported on this platform
HTTP Listening at 0.0.0.0:8000
STUN Listening at 127.0.0.1:3478
===> HTTP request /
res:1
body:
===> HTTP request /getDeviceList
res:1
body:
===> HTTP request /favicon.ico
res:1
body:

mpromonet, 2016-12-20:
Hi,
I am using ubuntu 16.04 (gcc 5.4.0), with webrtc ff84afd. 
Building webrtc with : 

    gn gen out/Release --args='is_debug=false rtc_use_h264=true rtc_include_tests=false' 
    ninja -C out/Release

I made some test yesterday, and it seems to work well.
Best Regards,
Michel.

saket424, 2016-12-21:
Michel,
Here is another puzzling datapoint

webrtc git revision HEAD as of today a4215a378469d5c88a6e9f3eeaed38b87d895e64
webrtc-streamer git revision HEAD as of today af9345f11f3d7f0abbdb5d14d15551f82c7b5695
ubuntu14.04 gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3)

I get no video rendering in Chrome Version 55.0.2883.87 m (64-bit)
Works fine in Firefox version 50.1.0 (64-bit)

Any ideas of why Chrome isn't working and where I can begin with debugging the issue?

saket424, 2016-12-21:
Michel,
I found the commit 03ccf59ad89fe8786d163146258762db77f49f51 causes the video to not render in chrome but works in firefox

I am closing this issue now
Thanks for your help

closed 2016-12-21 by saket424

mpromonet, 2017-01-02:
Hi, 
This should be fix by the merge of pullrequest #14 , the specification changes from onaddatream to ontrack.
Best Regards,
Michel.