mpromonet/webrtc-streamer #16: webrtc-streamer interworking with meetecho-janus-gateway

saket424, 2017-01-27:
Michel,
I opened a thread on the meetcho-janus google group thread
Would appreciate your comments on the modifications needed to make this work as video publisher into a janus room instead of the current usage which is limited to remote access via a browser

https://groups.google.com/forum/#!topic/meetecho-janus/WQB97s4eEDg

Thanks

mpromonet, 2017-01-30:
HI,
I think this just need to ask to the webrtc-streamer to join a janus room.
Maybe this could be achieve using javascript calling the webrtc-streamer and use SDP to join the jnaus room ?
Best Regards,
Michel.

saket424, 2017-01-30:
Yes. Currently both webrtc-streamer and janus act as servers. One of them has to be triggered to initiate an offer to the other for the communication to get started. I was planning to begin modifying webrtc-streamer to act as the client and janus the server in this client-server relationship. Was just wondering if you or anyone on this thread has experience with janus video rooms before I embarked on it myself since I am still a relative newbie.

saket424, 2017-02-04:
I am going to close this issue. I think I have an idea of how to negotiate creation of a publisher to a janus room and to obtain the janus sdp. Once remote sdp is obtained via Janus api by parsing the json sdp response,  this can be treated as the incoming offer and setremotedescription can be called . Subsequently an answer sdp is created and setlocaldescription is set and this triggers establishment of the peerconnection. In addition to negotiation of room using meetecho-janus JavaScript API,
I found two other related and relevant examples in Csharp and android respectively that maybe of interest that browses to this issue

https://github.com/Computician/JanusSharp/blob/master/README.md
https://github.com/Computician/janus-gateway-android/blob/master/README.md

mpromonet, 2017-05-08:
Hi,
I implement in an ugly way the negociation between Janus/WebRTC-streamer in [janusvideoroom.js](https://github.com/mpromonet/webrtc-streamer/blob/master/html/janusvideoroom.js).
Then you can ask WebRTC-streamer to join and publish streams to Janus-Gateway using : 

		var janus = new JanusVideoRoom("https://janus.conf.meetecho.com/janus", 1234);
		janus.connect("rtsp://217.17.220.110/axis-media/media.amp", "norway");

I made this try in Javascript inside the HTTP server of WebRTC-streamer but same stuff could be achieve in any script or langage that can call HTTP requests.
Best Regards,
Michel.

saket424, 2017-05-09:
Michel,
I am already using this for the past few days and it seems to work well . Thanks for implementing this powerful new use-case

closed 2017-05-09 by saket424