mpromonet/webrtc-streamer #17: No audio playback from rtsp url

saket424, 2017-02-02:
This URL rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov contains both video and audio. Only the video stream is included in the PeerConnection. Would be nice to have the associated audio also.

9a4gl, 2017-04-09:
I have one MS-C3577-PNA camera that sends also audio, maybe I can give a try.

9a4gl, 2017-04-14:
My camera is providing PCMU data, so is easy. But I got stuck creating audio track and audio source where I would feed those samples. mpromonet, do you have idea what rtspvideocapturer should extend/implement so we can pass data to audio track (cricket::VideoCapturer equivalent for audio) ?

saket424, 2017-07-17:
I tried the latest webrtc-streamer docker container using a chinese camera that has an rtsp stream with both audio and video tracks without success. Here are the logs but it could be because the camera is using a higher H264 profile(profile-level-id=644029) than the webrtc-browser/code can decode

[132:569] [56] (PeerConnectionManager.cpp:440): Close PeerConnection no more used rtsp://192.168.155.172/unicast
Close session: video/H264
answer:true

answer:{
   "iceServers" : [
      {
         "url" : "stun:stun.l.google.com:19302"
      }
   ]
}

body:{"type":"offer","sdp":"v=0\r\no=- 5802244000341421251 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:6/co\r\na=ice-pwd:5PvIf69/usR6gyZVoOJ6Wir8\r\na=fingerprint:sha-256 0F:EA:FB:D6:14:93:07:84:C8:13:41:20:CE:E8:BA:9C:7F:E2:86:82:0C:04:30:5B:A5:B4:82:3E:BD:9A:CE:FE\r\na=setup:actpass\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 isac/16000\r\na=rtpmap:104 isac/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 cn/32000\r\na=rtpmap:105 cn/16000\r\na=rtpmap:13 cn/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 98 100 102 127 97 99 101 125\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:6/co\r\na=ice-pwd:5PvIf69/usR6gyZVoOJ6Wir8\r\na=fingerprint:sha-256 0F:EA:FB:D6:14:93:07:84:C8:13:41:20:CE:E8:BA:9C:7F:E2:86:82:0C:04:30:5B:A5:B4:82:3E:BD:9A:CE:FE\r\na=setup:actpass\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=recvonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtpmap:100 H264/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=fmtp:100 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:102 red/90000\r\na=rtpmap:127 ulpfec/90000\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:125 rtx/90000\r\na=fmtp:125 apt=102\r\n"}
body:{"candidate":"candidate:3885250869 1 udp 2122260223 172.17.0.1 51844 typ host generation 0 ufrag 6/co network-id 1 network-cost 50","sdpMid":"audio","sdpMLineIndex":0}
answer:1

body:{"candidate":"candidate:4096743608 1 udp 2122194687 10.8.0.69 54642 typ host generation 0 ufrag 6/co network-id 3","sdpMid":"audio","sdpMLineIndex":0}
answer:1

Created new TCP socket 33 for connection
Connecting to 192.168.155.172, port 554 on socket 33...
[157:041] [58] (PeerConnectionManager.cpp:673): audiourl: options: idx_audioDevice:-1
...remote connection opened
Sending request: DESCRIBE rtsp://192.168.155.172/unicast RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2017.06.04
Accept: application/sdp

body:{"candidate":"candidate:1653295173 1 udp 2122129151 192.168.155.244 47262 typ host generation 0 ufrag 6/co network-id 2","sdpMid":"audio","sdpMLineIndex":0}
answer:1

answer:{
   "sdp" : "v=0\r\no=- 3857279469916545654 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE video\r\na=msid-semantic: WMS rtsp://192.168.155.172/unicast\r\nm=audio 0 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=inactive\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 isac/16000\r\na=rtpmap:104 isac/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 cn/32000\r\na=rtpmap:105 cn/16000\r\na=rtpmap:13 cn/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 57565 UDP/TLS/RTP/SAVPF 96 98 100 102 127 97 99 101 125\r\nc=IN IP4 172.17.0.2\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=candidate:4112619227 1 udp 2122260223 172.17.0.2 57565 typ host generation 0 network-id 1 network-cost 50\r\na=ice-ufrag:Hukc\r\na=ice-pwd:rDR36UMiFEhi5sYTaO+3O97D\r\na=ice-options:trickle\r\na=fingerprint:sha-256 4F:00:D2:FC:0C:17:D0:DA:E1:46:82:19:21:1A:CF:E5:77:27:9A:FC:F2:C9:85:AF:5E:34:4A:9E:3E:AA:19:97\r\na=setup:active\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:6 [... cut ...]
[h264 @ 0x7efbc0007740] no frame!
[157:068] [84] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[h264 @ 0x7efbc0007740] non-existing PPS 0 referenced
[h264 @ 0x7efbc0007740] decode_slice_header error
[h264 @ 0x7efbc0007740] no frame!
[157:111] [84] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
...
rolling NOTIFYs of avcodec_decode_video2 errors
...
 no more used rtsp://192.168.155.172/unicast
Close session: video/H264
answer:true

mpromonet, 2017-07-17:
Hi,
This is not working, I made some experimentation, but there is works to do :)
However it should not impact the video stream, this is strange. 
Regards,
Michel.

saket424, 2017-11-15:
Michel,
Sorry it took so long, When I increased the loglevel, I discovered that according to the original DESCRIBE, track1 is the video track and track2 is the audio track.

For some reason, webrtc-streamer is crossing the streams and trying to interpret the audio-track as video and video-track as audio

Is there some assumption in the code that the first track is always audio ? what is causing the streams to cross the onNewSession ?

[010:415] [4395] (rtspvideocapturer.h:99): RTSPAudioSource::onNewSession video/H264 m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:956707
a=rtpmap:96 H264/90000
a=fmtp:96 profile-level-id=644029;sprop-parameter-sets=Z2RAKawsqAeAIn5U,aO44gA==
a=control:track1

[010:419] [4394] (rtspvideocapturer.cpp:51): RTSPVideoCapturer::onNewSession audio/PCMA m=audio 0 RTP/AVP 8
c=IN IP4 0.0.0.0
b=AS:956707
a=control:track2

mpromonet, 2017-11-18:
@saket424 
Hi,
I think you misinterpret the output, all the session callback onNewSession that then choose the RTP session it needs.
Best Regards,
Michel.

saket424, 2017-11-18:
But notice  rtspvideocapturer::onnewsession has audio sdp and rtspaudiosource::onnewsession has video sdp. Something is certainly odd with at don't you agree?

mpromonet, 2017-11-19:
@saket424
Hi,
I upgrade the log to avoid the confusion you did in this commit 7bbdb75f100936375b3d4f6c1f7e9a409d541fd0
Best Regards,
Michel.

saket424, 2017-11-19:
The logs look better now with RTSPVideoCapturer::onNewSession rightly listing video sdp

[018:865] [6163] (rtspvideocapturer.cpp:53): RTSPVideoCapturer::onNewSession video/H264 m=video 0 RTP/AVP 96
c=IN IP4 0.0.0.0
b=AS:956707
a=rtpmap:96 H264/90000
a=fmtp:96 profile-level-id=644029;sprop-parameter-sets=Z2RAKawsqAeAIn5U,aO44gA==
a=control:track1

[018:865] [6163] (h264_sprop_parameter_sets.cc:32): Parsing sprop "Z2RAKawsqAeAIn5U,aO44gA=="
[018:865] [6163] (rtspvideocapturer.cpp:104): RTSPVideoCapturer:onData size:16 ts:0
[018:865] [6163] (rtspvideocapturer.cpp:113): RTSPVideoCapturer:onData SPS
[018:865] [6163] (rtspvideocapturer.cpp:120): RTSPVideoCapturer:onData SPS set timing_info_present_flag:0 0 0
[018:865] [6163] (rtspvideocapturer.cpp:129): RTSPVideoCapturer:onData SPS set format 1920x1080 fps:25
[018:865] [6163] (h264.cc:95): Creating H264DecoderImpl.
[018:866] [6163] (rtspvideocapturer.cpp:104): RTSPVideoCapturer:onData size:8 ts:0
[018:866] [6163] (rtspvideocapturer.cpp:141): RTSPVideoCapturer:onData PPS
Start playing sink for "video/H264" subsession

I continue to get rolling errors of 

[h264 @ 0x7f899c011160] non-existing PPS 0 referenced
[h264 @ 0x7f899c011160] decode_slice_header error
[h264 @ 0x7f899c011160] no frame!
[032:649] [5687] (h264_decoder_impl.cc:327): avcodec_send_packet error: -1094995529
NOTIFY failed

Any reason/ideas why the decode may still be failing within webrtc-streamer? It appears to have decoded the 1920x1080 @ 25 fps from the sprop correctly. vlc/ffplay are able to play the rtsp stream (audio+video) without issue

ffplay rtsp://192.168.155.172/unicast
ffplay version 2.8.11-0ubuntu0.16.04.1 Copyright (c) 2003-2017 the FFmpeg developers
Input #0, rtsp, from 'rtsp://192.168.155.172/unicast':   0B f=0/0   
  Metadata:
    title           : LIVE555 Streaming Media v2014.07.04
    comment         : LIVE555 Streaming Media v2014.07.04
  Duration: N/A, start: 0.000875, bitrate: N/A
    Stream #0:0: Video: h264 (High), yuv420p, 1920x1080, 25 fps, 25 tbr, 90k tbn, 180k tbc
    Stream #0:1: Audio: pcm_alaw, 8000 Hz, 1 channels, s16, 64 kb/s
   8.49 A-V: -0.011 fd=   0 aq=   13KB vq=  185KB sq=    0B f=0/0

iashwinprasada, 2018-01-09:
The video is working well, but Audio doesn't work for me either. I'm using the last binary release

The RTSP source codec information from VLC is here 
[image]

i keep getting this error:
[011:753] [9556] (rtspvideocapturer.h:172): RTSPAudioSource::onData error:No Audio decoder
NOTIFY failed

the detailed log for this is here: https://pastebin.com/RFH4tK2H

revolunet, 2018-10-26:
Hi,

trying with some AMCREST webcams and the audio doesnt seem to work while it works in VLC

example rtsp url : `rtsp://user:pass@192.168.0.42/cam/realmonitor?channel=1&subtype=1`

[image]

rossbcan, 2018-11-12:
What is the current status of audio?

I ported webrtc/webrtc-streamer to OpenWrt x86_64 and am seeing:

BigBuckBunny: no Audio
rtsp IP cam (with audio): no Audio

firefox: about:webrtc
RTP Stats
inbound_rtp_audio_0
Local: 07:03:48 GMT-0500 (Eastern Standard Time) inbound-rtp SSRC: 0

inbound_rtp_video_1
Decoder: Avg. bitrate: 0.19 Mbps (0.25 SD) Avg. framerate: 18.31 fps (8.75 SD)
Local: 07:03:48 GMT-0500 (Eastern Standard Time) inbound-rtp SSRC: 2246900416 Received: 2272 packets (2151.56 Kb) Lost: 0 Jitter: 0.004
Remote: 09:27:32 GMT-0500 (Eastern Standard Time) outbound-rtp SSRC: 2246900416 Sent: 2256 packets (2074.19 Kb)

Also: (is a problem?, what am I missing?):
[000:000] [20726] (audio_device_generic.cc:17): BuiltInAECIsAvailable: Not supported on this platform
[000:000] [20726] (audio_device_generic.cc:27): BuiltInAGCIsAvailable: Not supported on this platform
[000:000] [20726] (audio_device_generic.cc:37): BuiltInNSIsAvailable: Not supported on this platform

Regards;
Bill

mpromonet, 2018-11-13:
Hi Bill,

I did not spend lot of time on the subject, the audio format managed are those supported by WebRTC, just look at https://github.com/mpromonet/webrtc-streamer/blob/master/src/rtspaudiocapturer.cpp
Bunny have AAC audio, that is not supported,, managed (maybe not working) RTP format are PCMU, OPUS and L16.

The error about audio_device at startup depends on the audiolayer you are using, using -a0 it should disappear.

Best Regards,
Michel.

cl84, 2019-05-21:
Goodmorning everyone,
I have an rtsp stream with audio opus, but I still can't hear it.
Why?

xinghuolei, 2019-07-25:
I have a rtsp stream with h264 video and opus audio.
I can see video on bowser ,but can't hear the audio.
Server log like blow
```
#onNewSession
[037:568] [46495] (rtspaudiocapturer.cpp:44): RTSPAudioSource::onNewSession audio/OPUS m=audio 0 RTP/AVP 97
b=AS:96
a=rtpmap:97 opus/48000/2
a=fmtp:97 sprop-stereo=1
a=control:streamid=1

[037:568] [46495] (rtspaudiocapturer.cpp:66): RTSPAudioSource::onNewSession freq:8000 channel:1
Start playing sink for "audio/OPUS" subsession

#onData
[037:653] [46495] (rtspaudiocapturer.cpp:118): RTSPAudioSource::onData error:No Audio decoder
NOTIFY failed
[037:653] [46495] (rtspaudiocapturer.cpp:118): RTSPAudioSource::onData error:No Audio decoder
NOTIFY failed
```
Any advice?

xinghuolei, 2019-07-27:
I found SDP parse is wrong,I will try to fix it

mpromonet, 2019-07-27:
Hi xinghuolei,
You are right, the SDP says `opus/48000/2` and the capturer is initialized with `freq:8000 channel:1`.
The SDP parsing is quite rough, I thing the problem is due to codec that is "OPUS" and the sdp line with "opus". 
Best Regards,
Michel.

bkanuka, 2019-08-18:
Any status update on this bug?  I am also getting no audio on rtsp streams (video works). I'm using gstreamer to encode the audio and have tried opus and pcmu.  Audio and video work well in VLC.

http://termbin.com/773a

mpromonet, 2019-08-21:
Hi @bkanuka 
There was a problem initializing audio, this commit https://github.com/mpromonet/webrtc-streamer/commit/051ce9ee79b06283ca4427bf9bf15765584ea678 should help.
Best Regards,
Michel.

bkanuka, 2019-08-22:
I still cannot get sound to work in the browser (Chrome).  When the video plays, I right-click, show controls, and then unmute, but no sound.  I am using the latest docker image, which looks like it was built after commit  051ce9e

How can I help? Which logs would you like to see? 

Btw Chrome is still giving me the error described in issue #264  - just in case these are related.

mpromonet, 2019-08-22:
Hi @bkanuka,

Not sure that the latest docker contains the modification, if you are using latest no, if you are using latest-amd64 yes. I am fighting with windows docker :)

To know if you receive audio you should look to RTSPAudioSource::onNewSession logs and RTSPAudioSource::onData enabling the most verbose log.

Best Regards,

Michel.

P.S I did not look but I guess the error message describe in issue #264 just indicate the end of candidate events publishing an empty candidate.

mpromonet, 2019-08-28:
Hi @hanochg 

The decoding of SDP seems to have still some problem, with `a=rtpmap:96 OPUS/48000/2` it should get  freq:48000 wiht 2 channel but not :

> [012:114] [29] (rtspaudiocapturer.cpp:72): RTSPAudioSource::onNewSession code:OPUS freq:8000 channel:1

Best Regards,
Michel.

hanochg, 2019-08-28:
Hi @mpromonet, 
First, you did a great job with this project. It helped me a lot!
Second, I can combine video and audio when using G711 audio codec (over RTSP), but when using Opus codec the video appears without sound.
I'm using the latest 'latest-amd64' docker as mentioned above.

**Sound streaming**
I'm streaming from a WAV file to RTSP with Opus codec with the following FFmpeg command:
`ffmpeg -re  -i sound.wav -ab 32000 -ar 48000 -acodec libopus -f rtsp -rtsp_transport tcp rtsp://192.168.1.2:554/live/audio_in`
**Webrtc_streamer command (what the entrypoint in the docker runs):**
`/app/webrtc-streamer -a -T10 -C/app/config.json -H8000 -sstun.l.google.com:19302 -vvv
`
**config.json**
``` 
    {"urls":{
        "my_stream": {"video": "rtsp://192.168.1.2:554/live/video_in", "audio": "rtsp://192.168.1.2:554/live/audio_in"}
    }}
```
**webrtc-streamer relevant log (grep "onNewSession\|onData")**
```
(rtspaudiocapturer.cpp:47): RTSPAudioSource::onNewSession audio/OPUS m=audio 0 RTP/AVP 96#015
a=rtpmap:96 OPUS/48000/2#015
a=control:trackID=1#015
[012:114] [10] (audio_send_stream.cc:143): AudioSendStream: 4264067936
[012:114] [29] (rtspaudiocapturer.cpp:72): RTSPAudioSource::onNewSession code:OPUS freq:8000 channel:1
[012:114] [29] (rtspaudiocapturer.cpp:79): RTSPAudioSource::onNewSession not support codecm=audio 0 RTP/AVP 96#015
a=rtpmap:96 OPUS/48000/2#015
a=control:trackID=1#015
```

**Full docker's log file in max log verbosity**
https://justpaste.it/3uuzl

hanochg, 2019-08-28:
Thanks for the help, the SDP looks great and it looks that the parsing issue was resolved.
Another issue that raised, now it fails to decode the audio:
```
(rtspaudiocapturer.cpp:93): RTSPAudioSource::onData size:76 decoded:-1
(rtspaudiocapturer.cpp:99): RTSPAudioSource::onData error:Decode Audio failed
(rtspaudiocapturer.cpp:93): RTSPAudioSource::onData size:75 decoded:-1
(rtspaudiocapturer.cpp:99): RTSPAudioSource::onData error:Decode Audio failed
(rtspaudiocapturer.cpp:93): RTSPAudioSource::onData size:81 decoded:-1
(rtspaudiocapturer.cpp:99): RTSPAudioSource::onData error:Decode Audio failed
(rtspaudiocapturer.cpp:93): RTSPAudioSource::onData size:84 decoded:-1
(rtspaudiocapturer.cpp:99): RTSPAudioSource::onData error:Decode Audio failed
```
The setup is the same as I posted before.
Any ideas why?

bkanuka, 2019-08-30:
@mpromonet  "latest" docker image is working with pcmu/mulaw audio now. Thank you!

hanochg, 2019-09-01:
Hi again @mpromonet ,
I changed the log level but couldn't get additional log messages that describe the decoding phase.
I started debugging the code, I found that the codec is sent in upper case letters.
I fixed it locally but it didn't change much.
The fix:
From:
`webrtc::SdpAudioFormat format = webrtc::SdpAudioFormat(codec, m_freq, m_channel);`
To:
`webrtc::SdpAudioFormat format = webrtc::SdpAudioFormat(codecstr, m_freq, m_channel);`

I've found a Webrtc [unit test](https://github.com/pristineio/webrtc-mirror/blob/master/webrtc/modules/audio_coding/codecs/builtin_audio_decoder_factory_unittest.cc#L114) that initializes the Opus codec, the flow looks the same except for the 'params' variable.
The 'params' variable is optional but I tested it and added it with multiple values ("0", "1", "2"). 
Sadly, it didn't fix anything and I got the same error message.

I can see that the buffer size that's sent to the Opus decoding is not fixed (the size moves between 70-90), is that a wanted behavior?

mpromonet, 2019-09-01:
Hi @hanochg,

From the unit test you point out, it seems opus codec need to be initialized with add a parameter stereo with a value 0 or 1.

Best Regards,
Michel.

hanochg, 2019-09-04:
Hi, I managed to fix the issue with Opus, it's working great with mono and stereo.

I fixed the following issues locally and will create a pull request soon. 
For now, I'll just explain the issues:
- As mentioned above, the 'stereo' parameter is needed and needs to be set to "1".
- The function `m_decoder->Decode` returns the size of the decoded buffer size. The size includes the channels of the stereo so there's no need to multiply the returned size with the number of channels. The code changed to:
```
if (res > 0) {
	for (int i = 0 ; i < res; ++i) {
		m_buffer.push(decoded[i]);
	}
} else {
	RTC_LOG(LS_ERROR) << "RTSPAudioSource::onData error:Decode Audio failed";
}	
```
- The `decoded` buffer was too small for 48k audio, the decoder failed in the Webrtc function `webrtc::AudioDecoder::Decode` under the condition (the duration was bigger than zero):
``` 
if (duration >= 0 &&
      duration * Channels() * sizeof(int16_t) > max_decoded_bytes) {
    return -1;
  }
``` 
As the `max_decoded_bytes` was too small.
In the webrtc-streamer, the call for Decode is:
`int res = m_decoder->Decode(buffer, size, m_freq, size*sizeof(int16_t), decoded, &speech_type);`
Where **size*sizeof(int16_t)** is too small.

For now, I increased the size by a const value (x20) and I'm trying to find the right calculation of the decoded buffer size and the max_decoded_bytes.
The changes are in:
int16_t* decoded = new int16_t[**size*20**];
and:
int res = m_decoder->Decode(buffer, size, m_freq, **size * sizeof(int16_t) * 20** , decoded, &speech_type);

closed 2020-01-13 by mpromonet