mpromonet/webrtc-streamer #18: [Question] - What Commit of webrtc?

juvasquezg, 2017-03-07:
Hi,

Thank you for your time and code.

I tried with last webrtc, but commit a1991c517598fda9c9c0cf8876886e3968436ff9
removes http code

[webrtc commit](https://chromium.googlesource.com/external/webrtc/+/a1991c517598fda9c9c0cf8876886e3968436ff9)

```bash
Removing HTTPS and SOCKS proxy server code.
```

And not found webrtc/base/httpcommon-inl.h

And then I get 

```bash
conversion Error cricket::VideoCapturer*’ to std::unique_ptr<cricket::VideoCapturer>
```

Thank you!!

mpromonet, 2017-03-09:
Did you tried to add .release() to the std::unique_ptr ?
I will probably do a better fix later.

juvasquezg, 2017-03-09:
Hi,

Not really, I could not compile the code.

The last error was src 

```bash
The last error was, rtc :: HttpListenServer is not recognized
```

Thank you.

juvasquezg, 2017-03-09:
```
main.cpp line 88
```

mpromonet, 2017-03-09:
You need to relink the .a library because I changed the Makefile.
Using "make clean && make" should works if you buildld webrtc with httpserver support

juvasquezg, 2017-03-09:
Hi,

I built with

```bash
https://chromium.googlesource.com/external/webrtc/+log/branch-heads/56
```

I will test it.

Tks

juvasquezg, 2017-03-11:
Hello,

Nothing :(

```bash
g++  -o src/HttpServerRequestHandler.o -c src/HttpServerRequestHandler.cpp -Wall -pthread -g -std=c++11 -Iinc -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -I ../webrtc/src -I ../webrtc/src/chromium/src/third_party/jsoncpp/source/include
g++  -o src/main.o -c src/main.cpp -Wall -pthread -g -std=c++11 -Iinc -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -I ../webrtc/src -I ../webrtc/src/chromium/src/third_party/jsoncpp/source/include
g++  -o src/PeerConnectionManager.o -c src/PeerConnectionManager.cpp -Wall -pthread -g -std=c++11 -Iinc -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -I ../webrtc/src -I ../webrtc/src/chromium/src/third_party/jsoncpp/source/include
src/PeerConnectionManager.cpp: In member function ‘const Json::Value PeerConnectionManager::getDeviceList()’:
src/PeerConnectionManager.cpp:60:109: error: no matching function for call to ‘webrtc::VideoCaptureFactory::CreateDeviceInfo()’
  std::unique_ptr<webrtc::VideoCaptureModule::DeviceInfo> info(webrtc::VideoCaptureFactory::CreateDeviceInfo());
                                                                                                             ^
src/PeerConnectionManager.cpp:60:109: note: candidate is:
In file included from src/PeerConnectionManager.cpp:13:0:
../webrtc/src/webrtc/modules/video_capture/video_capture_factory.h:38:42: note: static webrtc::VideoCaptureModule::DeviceInfo* webrtc::VideoCaptureFactory::CreateDeviceInfo(int32_t)
   static VideoCaptureModule::DeviceInfo* CreateDeviceInfo(
                                          ^
../webrtc/src/webrtc/modules/video_capture/video_capture_factory.h:38:42: note:   candidate expects 1 argument, 0 provided
src/PeerConnectionManager.cpp: In member function ‘cricket::VideoCapturer* PeerConnectionManager::OpenVideoCaptureDevice(const string&)’:
src/PeerConnectionManager.cpp:496:110: error: no matching function for call to ‘webrtc::VideoCaptureFactory::CreateDeviceInfo()’
   std::unique_ptr<webrtc::VideoCaptureModule::DeviceInfo> info(webrtc::VideoCaptureFactory::CreateDeviceInfo());
                                                                                                              ^
src/PeerConnectionManager.cpp:496:110: note: candidate is:
In file included from src/PeerConnectionManager.cpp:13:0:
../webrtc/src/webrtc/modules/video_capture/video_capture_factory.h:38:42: note: static webrtc::VideoCaptureModule::DeviceInfo* webrtc::VideoCaptureFactory::CreateDeviceInfo(int32_t)
   static VideoCaptureModule::DeviceInfo* CreateDeviceInfo(
                                          ^
../webrtc/src/webrtc/modules/video_capture/video_capture_factory.h:38:42: note:   candidate expects 1 argument, 0 provided
src/PeerConnectionManager.cpp:510:59: error: request for member ‘release’ in ‘factory.cricket::WebRtcVideoDeviceCapturerFactory::Create((*(const cricket::Device*)(& cricket::Device((* & std::basic_string<char>(((const char*)(& name)), (*(const std::allocator<char>*)(& std::allocator<char>())))), 0))))’, which is of pointer type ‘cricket::VideoCapturer*’ (maybe you meant to use ‘->’ ?)
       capturer = factory.Create(cricket::Device(name, 0)).release();
                                                           ^
make: *** [src/PeerConnectionManager.o] Error 1
````

with [branch-heads/51](https://chromium.googlesource.com/external/webrtc/+log/branch-heads/51)

and

with [branch-heads/56](https://chromium.googlesource.com/external/webrtc/+log/branch-heads/56)

Thank you!!

juvasquezg, 2017-03-11:
If you run

```bash
cd webrtc/src
git branch
git log
````

What is your result?

Excuse me!!
Thank you so much.

mpromonet, 2017-03-11:
The last webrtc release I built is from mester branch commit [1e1c1ff7b917b6c6403d078cfcf7b7b04a7836df](https://chromium.googlesource.com/external/webrtc/+/1e1c1ff7b917b6c6403d078cfcf7b7b04a7836df)

juvasquezg, 2017-03-11:
Hello,

Great, with master branch commit  [1e1c1ff7b917b6c6403d078cfcf7b7b04a7836df](https://chromium.googlesource.com/external/webrtc/+/1e1c1ff7b917b6c6403d078cfcf7b7b04a7836df)

But I use Ubuntu Server 14.04 LTS:

1. I use [Vagrant](https://www.vagrantup.com/)
2. Provider is [VirtualBox](https://www.virtualbox.org/)

when I run, Out:

```bash
Logger level:3
[000:000] [26425] (webrtcvoiceengine.cc:1053): webrtc:   failed to connect context, error=-1
[000:001] [26425] (audio_device_pulse_linux.cc:173): failed to initialize PulseAudio
[000:002] [26425] (audio_device_impl.cc:279): Linux PulseAudio is *not* supported => ALSA APIs will be utilized instead
[000:003] [26425] (audio_device_alsa_linux.cc:176): failed to open X display, typing detection will not work
[000:010] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL
[000:011] [26425] (webrtcvoiceengine.cc:1053): webrtc:      snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:012] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL
[000:014] [26425] (webrtcvoiceengine.cc:1053): webrtc:      snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:015] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL
[000:017] [26425] (webrtcvoiceengine.cc:1053): webrtc:      snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:017] [26425] (webrtcvoiceengine.cc:1053): webrtc:   InitSpeaker() failed
[000:017] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM
[000:018] [26425] (webrtcvoiceengine.cc:1053): webrtc:      unable to open playback device: No such file or directory (-2)
[000:019] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL
[000:020] [26425] (webrtcvoiceengine.cc:1053): webrtc:      snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:020] [26425] (webrtcvoiceengine.cc:1053): webrtc:   InitMicrophone() failed
[000:020] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM
[000:021] [26425] (webrtcvoiceengine.cc:1053): webrtc:     unable to open record device: No such file or directory
[000:022] [26425] (audio_device_generic.cc:51): BuiltInAECIsAvailable: Not supported on this platform
[000:026] [26425] (audio_device_generic.cc:61): BuiltInAGCIsAvailable: Not supported on this platform
[000:029] [26425] (audio_device_generic.cc:71): BuiltInNSIsAvailable: Not supported on this platform
[000:029] [26425] (audio_device_buffer.cc:217): Not implemented
[000:029] [26425] (webrtcvoiceengine.cc:1053): webrtc: SetRecordingChannel() unable to set the recording channel (error=10028)
[000:030] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL
[000:030] [26425] (webrtcvoiceengine.cc:1053): webrtc:      snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:030] [26425] (webrtcvoiceengine.cc:1053): webrtc: SetRecordingDevice() cannot access microphone (error=9004)
[000:030] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL
[000:030] [26425] (webrtcvoiceengine.cc:1053): webrtc:      snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:030] [26425] (webrtcvoiceengine.cc:1053): webrtc:   InitMicrophone() failed
[000:030] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM
[000:030] [26425] (webrtcvoiceengine.cc:1053): webrtc:     unable to open record device: No such file or directory
[000:031] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL
[000:031] [26425] (webrtcvoiceengine.cc:1053): webrtc:      snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:031] [26425] (webrtcvoiceengine.cc:1053): webrtc: SetPlayoutDevice() cannot access speaker (error=9005)
[000:031] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:953:(snd_ctl_open_noupdate) Invalid CTL
[000:031] [26425] (webrtcvoiceengine.cc:1053): webrtc:      snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:031] [26425] (webrtcvoiceengine.cc:1053): webrtc:   InitSpeaker() failed
[000:031] [26425] (webrtcvoiceengine.cc:1053): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM
[000:031] [26425] (webrtcvoiceengine.cc:1053): webrtc:      unable to open playback device: No such file or directory (-2)
HTTP Listen at 0.0.0.0:8000
HTTP Listening at 0.0.0.0:8000
STUN Listening at 127.0.0.1:3478
```

Thank you!!

juvasquezg, 2017-03-12:
Hi,

Excuse me,

I started a new project and built it. 

When run example, I get:

```bash
[011:946] [1876] (PeerConnectionManager.cpp:549): Cannot create capturer rtsp://217.17.220.110/axis-media/media.amp
[011:946] [1876] (PeerConnectionManager.cpp:597): Cannot find stream
```

complete stdout: 

```bash
===> HTTP request path:/
filename:./html/index.html
===> HTTP request path:/favicon.ico
filename:./html/favicon.ico
===> HTTP request path:/ajax.js
filename:./html/ajax.js
===> HTTP request path:/webrtcstreamer.js
filename:./html/webrtcstreamer.js
===> HTTP request path:/getIceServers
body:
answer:{
   "iceServers" : [
      {
         "url" : "stun:stun.l.google.com:19302"
      }
   ]
}

===> HTTP request path:/getDeviceList
body:
answer:[
   "rtsp://217.17.220.110/axis-media/media.amp",
   "rtsp://85.255.175.241/h264",
   "rtsp://85.255.175.244/h264",
   "rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov"
]

===> HTTP request path:/favicon.ico
filename:./html/favicon.ico
===> HTTP request path:/call
body:{"type":"offer","sdp":"v=0\r\no=- 8420262131234635730 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:32Xj\r\na=ice-pwd:G6NA1uvmCYbTouKq/oLwallB\r\na=fingerprint:sha-256 D3:75:46:08:C1:5F:0E:8D:E0:33:12:4C:35:61:4B:EC:F8:4E:9A:DE:AD:43:E1:74:55:CF:77:74:D5:5C:27:F3\r\na=setup:actpass\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 isac/16000\r\na=rtpmap:104 isac/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 cn/32000\r\na=rtpmap:105 cn/16000\r\na=rtpmap:13 cn/8000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 100 101 107 116 117 96 97 99 98\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:32Xj\r\na=ice-pwd:G6NA1uvmCYbTouKq/oLwallB\r\na=fingerprint:sha-256 D3:75:46:08:C1:5F:0E:8D:E0:33:12:4C:35:61:4B:EC:F8:4E:9A:DE:AD:43:E1:74:55:CF:77:74:D5:5C:27:F3\r\na=setup:actpass\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=recvonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:100 VP8/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=rtpmap:101 VP9/90000\r\na=rtcp-fb:101 ccm fir\r\na=rtcp-fb:101 nack\r\na=rtcp-fb:101 nack pli\r\na=rtcp-fb:101 goog-remb\r\na=rtcp-fb:101 transport-cc\r\na=rtpmap:107 H264/90000\r\na=rtcp-fb:107 ccm fir\r\na=rtcp-fb:107 nack\r\na=rtcp-fb:107 nack pli\r\na=rtcp-fb:107 goog-remb\r\na=rtcp-fb:107 transport-cc\r\na=fmtp:107 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:116 red/90000\r\na=rtpmap:117 ulpfec/90000\r\na=rtpmap:96 rtx/90000\r\na=fmtp:96 apt=100\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=101\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=107\r\na=rtpmap:98 rtx/90000\r\na=fmtp:98 apt=116\r\n"}
[011:946] [1876] (PeerConnectionManager.cpp:549): **Cannot create capturer rtsp://217.17.220.110/axis-media/media.amp**
[011:946] [1876] (PeerConnectionManager.cpp:597): **Cannot find stream**
===> HTTP request path:/addIceCandidate
body:{"candidate":"candidate:339720134 1 udp 2113937151 192.168.1.53 52552 typ host generation 0 ufrag 32Xj network-cost 50","sdpMid":"audio","sdpMLineIndex":0}
answer:1

===> HTTP request path:/addIceCandidate
body:{"candidate":"candidate:339720134 2 udp 2113937150 192.168.1.53 52554 typ host generation 0 ufrag 32Xj network-cost 50","sdpMid":"audio","sdpMLineIndex":0}
answer:1

===> HTTP request path:/addIceCandidate
body:{"candidate":"candidate:339720134 1 udp 2113937151 192.168.1.53 52556 typ host generation 0 ufrag 32Xj network-cost 50","sdpMid":"video","sdpMLineIndex":1}
answer:1

===> HTTP request path:/addIceCandidate
body:{"candidate":"candidate:339720134 2 udp 2113937150 192.168.1.53 52558 typ host generation 0 ufrag 32Xj network-cost 50","sdpMid":"video","sdpMLineIndex":1}
answer:1

===> HTTP request path:/addIceCandidate
body:{"candidate":"candidate:842163049 1 udp 1677729535 190.251.0.222 52552 typ srflx raddr 192.168.1.53 rport 52552 generation 0 ufrag 32Xj network-cost 50","sdpMid":"audio","sdpMLineIndex":0}
answer:1

===> HTTP request path:/addIceCandidate
body:{"candidate":"candidate:842163049 2 udp 1677729534 190.251.0.222 52554 typ srflx raddr 192.168.1.53 rport 52554 generation 0 ufrag 32Xj network-cost 50","sdpMid":"audio","sdpMLineIndex":0}
answer:1

===> HTTP request path:/addIceCandidate
body:{"candidate":"candidate:842163049 1 udp 1677729535 190.251.0.222 52556 typ srflx raddr 192.168.1.53 rport 52556 generation 0 ufrag 32Xj network-cost 50","sdpMid":"video","sdpMLineIndex":1}
answer:1

===> HTTP request path:/addIceCandidate
body:{"candidate":"candidate:842163049 2 udp 1677729534 190.251.0.222 52558 typ srflx raddr 192.168.1.53 rport 52558 generation 0 ufrag 32Xj network-cost 50","sdpMid":"video","sdpMLineIndex":1}
answer:1
```

Thank you so much.

juvasquezg, 2017-03-12:
I have live555

last line to build is

```bash
g++  -o webrtc-server__Release src/HttpServerRequestHandler.o src/main.o src/PeerConnectionManager.o src/rtspvideocapturer.o libWebRTC__Release.a -pthread  -lX11 -ldl -lr
t                                                                                                                                                                         
```

Include rtspvideocapturer

saket424, 2017-03-13:
>>>>The last webrtc release I built is from master branch commit 1e1c1ff7b917b6c6403d078cfcf7b7b04a7836df

Hi Michel,
I get the following error when I use https://chromium.googlesource.com/external/webrtc/+log/branch-heads/58 and the newest webrtc-streamer HEAD

anand@ubu1404:~/webrtc-streamer$ make
g++  -o src/PeerConnectionManager.o -c src/PeerConnectionManager.cpp -Wall -pthread -g -std=c++11 -Iinc -DHAVE_LIVE555 -I live555helper/inc -I /usr/include/liveMedia  -I /usr/include/groupsock -I /usr/include/UsageEnvironment -I /usr/include/BasicUsageEnvironment/ -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -I ../webrtc/src -I ../webrtc/src/third_party/jsoncpp/source/include 
src/PeerConnectionManager.cpp: In member function ‘cricket::VideoCapturer* PeerConnectionManager::OpenVideoCaptureDevice(const string&)’:
src/PeerConnectionManager.cpp:491:39: error: cannot allocate an object of abstract type ‘RTSPVideoCapturer’
   capturer = new RTSPVideoCapturer(url);
                                       ^
In file included from src/PeerConnectionManager.cpp:20:0:
inc/rtspvideocapturer.h:26:7: note:   because the following virtual functions are pure within ‘RTSPVideoCapturer’:
 class RTSPVideoCapturer : public cricket::VideoCapturer, public RTSPConnection::Callback, public rtc::Thread, public webrtc::DecodedImageCallback
       ^
In file included from inc/rtspvideocapturer.h:17:0,
                 from src/PeerConnectionManager.cpp:20:
live555helper/inc/rtspconnectionclient.h:48:21: note: 	virtual bool RTSPConnection::Callback::onData(const char*, unsigned char*, ssize_t, timeval)
     virtual bool    onData(const char* id, unsigned char* buffer, ssize_t size, struct timeval presentationTime) = 0;
                     ^
make: *** [src/PeerConnectionManager.o] Error 1
anand@ubu1404:~/webrtc-streamer$

saket424, 2017-03-13:
I reverted to an earlier commit of webrtc-streamer  b8606e1c9986246f6fbc97de72707ed516392dba that was known to work with webrtc/branch-heads/57 and made the single change to src/PeerConnectionManager.cpp (capturer=factory.Create(cricket::Device(name, 0)).release()) so that I can try it with webrtc/branch-heads/58 without success

Now I get a linker error

g++  -o webrtc-server__Release src/HttpServerRequestHandler.o src/main.o src/PeerConnectionManager.o live555helper/live555helper.a libWebRTC__Release.a -pthread  live555helper/live555helper.a -l:libliveMedia.a -l:libgroupsock.a -l:libUsageEnvironment.a -l:libBasicUsageEnvironment.a -llog4cpp -lX11 -ldl -lrt  
src/HttpServerRequestHandler.o: In function `HttpServerRequestHandler::OnRequest(rtc::HttpServer*, rtc::HttpServerTransaction*)':
/home/anand/webrtc-streamer/src/HttpServerRequestHandler.cpp:152: undefined reference to `rtc::HttpServer::Respond(rtc::HttpServerTransaction*)'
src/main.o: In function `main':
/home/anand/webrtc-streamer/src/main.cpp:79: undefined reference to `rtc::HttpListenServer::HttpListenServer()'
/home/anand/webrtc-streamer/src/main.cpp:82: undefined reference to `rtc::HttpListenServer::Listen(rtc::SocketAddress const&)'
/home/anand/webrtc-streamer/src/main.cpp:109: undefined reference to `rtc::HttpListenServer::~HttpListenServer()'
/home/anand/webrtc-streamer/src/main.cpp:109: undefined reference to `rtc::HttpListenServer::~HttpListenServer()'
collect2: error: ld returned 1 exit status
make: *** [webrtc-server__Release] Error 1

saket424, 2017-03-13:
They moved the gn target for httpserver.cc (HttpListenServer) to rtc_base_tests_utils in 

https://chromium.googlesource.com/external/webrtc.git/+/14584623035b29ce2a7830ea38956a10c584248d%5E!/#F0

So I needed the Makefile changes also. rtsp with webrtc/branch-heads/58  now works with webrtc-streamer  b8606e1 . Need to see if I can make it work with the HEAD revision of webrtc-streamer

anand@ubu1404:~/webrtc-streamer$ git diff Makefile
diff --git a/Makefile b/Makefile
index 667a8b7..a83a213 100644
--- a/Makefile
+++ b/Makefile
@@ -26,7 +26,7 @@ endif
 WEBRTCLIBPATH=$(WEBRTCROOT)/src/$(GYP_GENERATOR_OUTPUT)/out/$(WEBRTCBUILD)
 
 CFLAGS += -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0
-CFLAGS += -I $(WEBRTCROOT)/src -I $(WEBRTCROOT)/src/chromium/src/third_party/js
+CFLAGS += -I $(WEBRTCROOT)/src -I $(WEBRTCROOT)/src/third_party/jsoncpp/source/
 #detect
 TESTDEBUG=$(shell nm $(wildcard $(WEBRTCLIBPATH)/obj/webrtc/media/rtc_media/vid
 ifeq ($(TESTDEBUG),debug)
@@ -35,7 +35,7 @@ endif
 LDFLAGS += -lX11 -ldl -lrt  
 
 WEBRTC_LIB = $(shell find $(WEBRTCLIBPATH)/obj/base -name '*.o')
-WEBRTC_LIB += $(shell find $(WEBRTCLIBPATH)/obj/webrtc -name '*.o' ! -path '*te
+WEBRTC_LIB += $(shell find $(WEBRTCLIBPATH)/obj/webrtc -name '*.o')
 WEBRTC_LIB += $(shell find $(WEBRTCLIBPATH)/obj/third_party -name '*.o')
 LIBS+=libWebRTC_$(GYP_GENERATOR_OUTPUT)_$(WEBRTCBUILD).a
 libWebRTC_$(GYP_GENERATOR_OUTPUT)_$(WEBRTCBUILD).a: $(WEBRTC_LIB)

saket424, 2017-03-13:
Back to webrtc-streamer HEAD as of March 11instead of  b8606e1 and I get the following error when I enable live555 for rtsp with webrtc/branch-heads/58

anand@ubu1404:~/webrtc-streamer$ git diff Makefile
diff --git a/Makefile b/Makefile
index 580b584..180b29d 100644
--- a/Makefile
+++ b/Makefile
@@ -10,7 +10,7 @@ all: $(TARGET)
 
 # live555helper
 ifneq ($(wildcard $(SYSROOT)/usr/include/liveMedia/liveMedia.hh),)
-ifneq ($(wildcard $(SYSROOT)/usr/lib/libliveMedia.a),)
+# ifneq ($(wildcard $(SYSROOT)/usr/lib/libliveMedia.a),)
 LIBS+=live555helper/live555helper.a
 live555helper/live555helper.a:
        make -C live555helper
@@ -21,14 +21,14 @@ CFLAGS += -I $(SYSROOT)/usr/include/liveMedia  -I $(SYSROOT)/usr/include/groupso
 
 LDFLAGS += live555helper/live555helper.a
 LDFLAGS += -l:libliveMedia.a -l:libgroupsock.a -l:libUsageEnvironment.a -l:libBasicUsageEnvironment
-endif
+# endif
 endif
 
 # webrtc
 WEBRTCLIBPATH=$(WEBRTCROOT)/src/$(GYP_GENERATOR_OUTPUT)/out/$(WEBRTCBUILD)
 
 CFLAGS += -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0
-CFLAGS += -I $(WEBRTCROOT)/src -I $(WEBRTCROOT)/src/chromium/src/third_party/jsoncpp/source/include
+CFLAGS += -I $(WEBRTCROOT)/src -I $(WEBRTCROOT)/src/third_party/jsoncpp/source/include
 #detect
 TESTDEBUG=$(shell nm $(wildcard $(WEBRTCLIBPATH)/obj/webrtc/media/rtc_media/videocapturer.o $(WEBRT
 ifeq ($(TESTDEBUG),debug)
anand@ubu1404:~/webrtc-streamer$ make
g++  -o src/PeerConnectionManager.o -c src/PeerConnectionManager.cpp -Wall -pthread -g -std=c++11 -Iinc -DHAVE_LIVE555 -I live555helper/inc -I /usr/include/liveMedia  -I /usr/include/groupsock -I /usr/include/UsageEnvironment -I /usr/include/BasicUsageEnvironment/ -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -I ../webrtc/src -I ../webrtc/src/third_party/jsoncpp/source/include 
src/PeerConnectionManager.cpp: In member function ‘cricket::VideoCapturer* PeerConnectionManager::OpenVideoCaptureDevice(const string&)’:
src/PeerConnectionManager.cpp:491:39: error: cannot allocate an object of abstract type ‘RTSPVideoCapturer’
   capturer = new RTSPVideoCapturer(url);
                                       ^
In file included from src/PeerConnectionManager.cpp:20:0:
inc/rtspvideocapturer.h:26:7: note:   because the following virtual functions are pure within ‘RTSPVideoCapturer’:
 class RTSPVideoCapturer : public cricket::VideoCapturer, public RTSPConnection::Callback, public rtc::Thread, public webrtc::DecodedImageCallback
       ^
In file included from inc/rtspvideocapturer.h:17:0,
                 from src/PeerConnectionManager.cpp:20:
live555helper/inc/rtspconnectionclient.h:48:21: note: 	virtual bool RTSPConnection::Callback::onData(const char*, unsigned char*, ssize_t, timeval)
     virtual bool    onData(const char* id, unsigned char* buffer, ssize_t size, struct timeval presentationTime) = 0;
                     ^
make: *** [src/PeerConnectionManager.o] Error 1
anand@ubu1404:~/webrtc-streamer$

juvasquezg, 2017-03-13:
Hi @saket424 How can I enable live555. I installed but I don't know. I work with master branch 1e1c1ff7b917b6c6403d078cfcf7b7b04a7836df

Thank you.

mpromonet, 2017-03-14:
Hi,
Anyway as webrtc remove its httpserver, I will replace it with another one.
Best Regards,
Michel

mpromonet, 2017-03-20:
Hi,
This commit cf84eae1167c11c03b7c09c518a704fd9d6673a0 replace webrtc removed httpserver with civetweb http server.
Now it should build with the commit you tried before.
Best Regards,
Michel.

closed 2017-03-23 by mpromonet