mpromonet/webrtc-streamer #22: Support for RTSP over TCP

9a4gl, 2017-03-25:
I have few cameras and some of them do not work with vlc or genius vision without switching to TCP transport. Exactly those cameras do not work webrtc-streamer. Can I somehow force it to use TCP for RTSP ? Support for that would be very useful.

BR,
Tihomir

mpromonet, 2017-03-28:
Hi Tihomir,
You just need to modify the way the SETUP command is sent in [rtspconnectionclient.cpp](https://github.com/mpromonet/live555helper/blob/1ca427a7fede0bef868ea85fcb78b1d1886a0ec6/src/rtspconnectionclient.cpp)  from :

    this->sendSetupCommand(*m_subSession, continueAfterSETUP); 
to:

    this->sendSetupCommand(*m_subSession, continueAfterSETUP, false, true);

I will modify the live555Helper to give more options.

Best Regards,
Michel.

9a4gl, 2017-04-02:
Looks you merged changes in lib555helper, but not in main repo. Build is broken now.

mpromonet, 2017-04-02:
I guess you should update the submodule, you could do this removing the live555Helper folder, `make clean` and re-run make.

commit 905b50af (2017-04-09): fix #22: allow to use RTP over TCP (and configure timeout)
closed 2017-04-09 by mpromonet