mpromonet/webrtc-streamer #23: RTSP stream not closed

9a4gl, 2017-03-25:
After I close browser, webrtc-streamer trace connection closed:
```
[006:150] [26080] (PeerConnectionManager.cpp:345): Close PeerConnection
[006:151] [26080] (PeerConnectionManager.cpp:358): Close PeerConnection no more used rtsp://user:pass@some.ip:554/2
answer:true
```

But it keeps tracing:
```
[h264 @ 0x7fcfa8007b60] no frame!
[010:202] [26128] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
```

Which indiacte h264 stream is not closed. Same error is traced also when browser is opened and stream is shown in browser, so the error itself is not a problem.

BR,
Tihomir 9a4gl

9a4gl, 2017-03-30:
I have spent some time debugging expecting fix will be trivial, but...
In PeerConnectionManager::hangUp I added removing of video tracks
'''
					while (it->second->GetVideoTracks().size() > 0)
					{
						it->second->RemoveTrack(it->second->GetVideoTracks().at(0));
					}
'''
But destructor of RTSPVideoCapturer crash in some base webrtc code :(

mpromonet, 2017-03-31:
Hi,
Then it needs more debugging.... I made the same experience but did not takes the time to debug the crash.
Best Regards,
Michel.

9a4gl, 2017-03-31:
Here are some new and not so good findings. If webrtc is debug build, then it does not crash. But if webrtc is release build, then it crash even if I make rtspvideocapturer dummy :( I think I will live with debug for now.

Everything smells like stack corruption, but valgrind in debug mode didn't helped me much, I got things like:

```
==9299== Conditional jump or move depends on uninitialised value(s)
==9299==    at 0x5DA23D4: pthread_mutex_destroy (pthread_mutex_destroy.c:31)
==9299==    by 0x42A1F2: cricket::VideoCapturer::~VideoCapturer() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x428BC4: RTSPVideoCapturer::~RTSPVideoCapturer() (rtspvideocapturer.cpp:29)
==9299==    by 0x428C23: RTSPVideoCapturer::~RTSPVideoCapturer() (rtspvideocapturer.cpp:36)
==9299==    by 0x464BFD: webrtc::VideoCapturerTrackSource::~VideoCapturerTrackSource() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x464CF8: rtc::RefCountedObject<webrtc::VideoCapturerTrackSource>::~RefCountedObject() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x464ADE: rtc::RefCountedObject<webrtc::VideoCapturerTrackSource>::Release() const (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x459742: webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface>::DestroyInternal() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x45965A: webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface>::~VideoTrackSourceProxyWithInternal() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__R
==9299==    by 0x458988: rtc::RefCountedObject<webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface> >::~RefCountedObject() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-se
==9299==    by 0x45897B: rtc::RefCountedObject<webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface> >::Release() const (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server
==9299==    by 0x465F6A: webrtc::VideoTrack::~VideoTrack() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
```

```
==9299== Conditional jump or move depends on uninitialised value(s)
==9299==    at 0x5DA23E0: pthread_mutex_destroy (pthread_mutex_destroy.c:32)
==9299==    by 0x42A1F2: cricket::VideoCapturer::~VideoCapturer() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x428BC4: RTSPVideoCapturer::~RTSPVideoCapturer() (rtspvideocapturer.cpp:29)
==9299==    by 0x428C23: RTSPVideoCapturer::~RTSPVideoCapturer() (rtspvideocapturer.cpp:36)
==9299==    by 0x464BFD: webrtc::VideoCapturerTrackSource::~VideoCapturerTrackSource() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x464CF8: rtc::RefCountedObject<webrtc::VideoCapturerTrackSource>::~RefCountedObject() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x464ADE: rtc::RefCountedObject<webrtc::VideoCapturerTrackSource>::Release() const (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x459742: webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface>::DestroyInternal() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x45965A: webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface>::~VideoTrackSourceProxyWithInternal() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__R
==9299==    by 0x458988: rtc::RefCountedObject<webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface> >::~RefCountedObject() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-se
==9299==    by 0x45897B: rtc::RefCountedObject<webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface> >::Release() const (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server
==9299==    by 0x465F6A: webrtc::VideoTrack::~VideoTrack() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
```

```
==9299==    at 0x5DA23D4: pthread_mutex_destroy (pthread_mutex_destroy.c:31)
==9299==    by 0x42A204: cricket::VideoCapturer::~VideoCapturer() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x428BC4: RTSPVideoCapturer::~RTSPVideoCapturer() (rtspvideocapturer.cpp:29)
==9299==    by 0x428C23: RTSPVideoCapturer::~RTSPVideoCapturer() (rtspvideocapturer.cpp:36)
==9299==    by 0x464BFD: webrtc::VideoCapturerTrackSource::~VideoCapturerTrackSource() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x464CF8: rtc::RefCountedObject<webrtc::VideoCapturerTrackSource>::~RefCountedObject() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x464ADE: rtc::RefCountedObject<webrtc::VideoCapturerTrackSource>::Release() const (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x459742: webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface>::DestroyInternal() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x45965A: webrtc::VideoTrackSourceProxyWithInternal<webrtc::VideoTrackSourceInterface>::~VideoTrackSourceProxyWithInternal() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__R
[... cut ...]
==9299==    by 0x42D92E: rtc::VideoSourceBase::~VideoSourceBase() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x42DA0C: rtc::VideoBroadcaster::~VideoBroadcaster() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)
==9299==    by 0x42A216: cricket::VideoCapturer::~VideoCapturer() (in /usr/local/src/webrtc-amd64/webrtc-streamer/webrtc-server__Release)  
==9299==    by 0x428BC4: RTSPVideoCapturer::~RTSPVideoCapturer() (rtspvideocapturer.cpp:29)
```

9a4gl, 2017-04-04:
Kind a good news, ~VideoCapturer() looks to be problem, patch webrtc with this and you voila !

```
diff --git a/webrtc/media/base/videocapturer.cc b/webrtc/media/base/videocapturer.cc
index efd075d4d..b4ee32c4b 100644
--- a/webrtc/media/base/videocapturer.cc
+++ b/webrtc/media/base/videocapturer.cc
@@ -37,6 +37,10 @@ VideoCapturer::VideoCapturer() : apply_rotation_(false) {
   Construct();
 }
 
+VideoCapturer::~VideoCapturer()
+{
+}
+
 void VideoCapturer::Construct() {
   enable_camera_list_ = false;
   capture_state_ = CS_STOPPED;
diff --git a/webrtc/media/base/videocapturer.h b/webrtc/media/base/videocapturer.h
index 510063bc1..3d23c6e59 100644
--- a/webrtc/media/base/videocapturer.h
+++ b/webrtc/media/base/videocapturer.h
@@ -83,7 +83,7 @@ class VideoCapturer : public sigslot::has_slots<>,
  public:
   VideoCapturer();
 
-  virtual ~VideoCapturer() {}
+  virtual ~VideoCapturer();
 
   // Gets the id of the underlying device, which is available after the capturer
   // is initialized. Can be used to determine if two capturers reference the
```
So, can we solve this on our side or we need to beg webrtc people ?

9a4gl, 2017-04-05:
Exactly, just move destructor from .h to cc. 

I tried to find that destructor with nm, readelf, objdump but unfortunatelly was not able to detect which symbol is destructor :( my plan was to remove destructor with objcopy from rtspvideocapturer.o and to force linking with one in webrtc.

another approach would be to link webrtc first, but that brings other problems, which may be resolved with some flags to tell linker to recursively search for symbols, will try that too

gcc says no-inline-small-functions works with -O2, but we do not have -O2. Anyway will try later today.

mpromonet, 2017-04-05:
I tried but it still crash, even building webrtc-streamer with clang crash.

9a4gl, 2017-04-05:
hm, the only additional change I have (and I do remember) is I built webrtc with use_rtti=true and removed no-rtti from your Makefile, maybe it is combination

9a4gl, 2017-04-05:
I just tried again, rtti is not important. I tried also use_rtti=false (default) and no issues if you move destrucor from header to cc.

9a4gl, 2017-04-05:
Using -fno-inline-small-functions I have crash

9a4gl, 2017-04-05:
Have you tried to build webrtc using gcc, you can do that by adding to gn --args
is_clang=false use_sysroot=false treat_warnings_as_errors=false

I successfully built webrtc, but failed linking webrtc-streaming with
/usr/bin/ld: <somepaath>src/out/Release/obj/webrtc/base/librtc_base_approved.a(checks.o): unrecognized relocation (0x2a) in section `.text._ZN3rtc11VPrintErrorEPKcP13__va_list_tag'

looks like libc issue on my system, ...

9a4gl, 2017-04-08:
I am using my patch for quite a while, looks it works and rtsp channel is closing correctly. I am bit confused why it does not work for you. Today I was developing my web page that uses webrtc-streamer and was switching stream in same page. I had crash in Environment.cpp, too bad I ignored that, but there is also something fishy. 
I will close this issue as I fixed that, at least for me, but will open another one similar.

closed 2017-04-08 by 9a4gl