mpromonet/webrtc-streamer #24: Missing support for TURN relay

9a4gl, 2017-04-01:
I will create a patch and send you as I already have that implemented locally

9a4gl, 2017-04-01:
[0001-Add-support-for-TURN-relay.patch.txt](https://github.com/mpromonet/webrtc-streamer/files/887812/0001-Add-support-for-TURN-relay.patch.txt)

Please review and merge upstream if you agree with change.

mpromonet, 2017-04-01:
Hi,
Thanks, it is more convenient to submit modifications as a pull request.
Regards
Michel.

9a4gl, 2017-04-01:
I would do that, but I cannot push. I am new at github, do you need to allow me to push on your repo or ? Help me and I will do that :)

mpromonet, 2017-04-01:
Hi,
The standard way to process with GitHub  is to fork my repo, push the modification to your forked repo and submit a pull request to my repo.
Then it is easy to see the difference and accept the pull request.
Best Regards,
Michel.

9a4gl, 2017-04-01:
Thanks, forking sounded to me like expensive operation that should be used if I want to split from your project. Ok, thanks.

closed 2017-04-01 by mpromonet