mpromonet/webrtc-streamer #26: Use libwebrtc from https://github.com/aisouard/libwebrtc

9a4gl, 2017-04-02:
Guys at https://github.com/aisouard/libwebrtc simplify usage/build of webrtc. And looks they are also close with arm support. Will give a try and report here if we can gain from their work.

mpromonet, 2017-04-02:
This seems interesting, I will look at that

9a4gl, 2017-04-02:
pkg-config --cflags LibWebRTC
-D_GLIBCXX_DEBUG=1 -D_DEBUG=1 -DWEBRTC_POSIX -D_GLIBCXX_USE_CXX11_ABI=0 -std=gnu++0x -pthread -I/usr/local/include  

pkg-config --libs LibWebRTC
-L/usr/local/lib -lwebrtc -lSM -lICE -lX11 -lXext -ldl -lrt

9a4gl, 2017-04-02:
Note they stick to "latest working release" of webrtc, not to master. I do not know how latest working release is defined, but webrtc-streamer do not compile with that. I am having troubles building master with their system. Somehow I think webrtc-stream should also stick to stable releases as these VideoCapturer deletion and arm issues may come from using latest code.

mpromonet, 2017-04-02:
I don't think problems with VideoCapturer deletion and arm crash comes from master (but I could be wrong). I saw these problems sevrals month ago, I guess the problem is related to inconsistent compilation flags. 
What problems do you have to build webrtc-streamer, I guess you should build enabling H264 support with GN_EXTRA_ARGS='rtc_use_h264=true ffmpeg_branding="Chrome"', maybe with WEBRTC_BRANCH_HEAD=master

9a4gl, 2017-04-02:
Another problem with libwebrtc it does not include thirtparty stuff, so it is missing json.h files.

mpromonet, 2017-04-03:
json.h is in out/webrtc/src/thrird_party/jsoncpp/source/include/json

9a4gl, 2017-04-03:
I know, I am telling you that header is missing when you use https://github.com/aisouard/libwebrtc

mpromonet, 2017-04-04:
me too, in libwebrtc/out/webrtc/src/thrird_party/jsoncpp/source/include/json

9a4gl, 2017-04-08:
I do not find this library helpful for building webrtc streamer and results with arm are not good.

closed 2017-04-08 by 9a4gl