mpromonet/webrtc-streamer #27: Docker build

pims, 2017-04-06:
Hi there,

For those of us having troubles compiling and linking webrtc, would you consider providing a Docker image with the a webrtc revision you know to be _compatible_ ?

An alternative could be to pin the webrtc revision to a guaranteed compatible version.

What do you think?

mpromonet, 2017-04-22:
Hi,

My actual indertanding is that Webrtc is still an experimental technology and WebRTC project is evolving quickly, for now I am using docker to build for Heroku, but I am not, fow now consider to provide an image with the source or binaries...

Maybe you should look to [libwebrtc](https://github.com/aisouard/libwebrtc) that simplify the build.

Regards,
Michel.

mpromonet, 2017-05-14:
Hi,
Finally using docker cloud I was able to automate build (without timeout like in hubdocker) and push  docker image from the github repo.
You can use it : 
     
    docker run -p 8000:8000 -it mpromonet/webrtc-streamer

Regards,
Michel.

pims, 2017-05-15:
Awesome. Thanks Michel!

closed 2017-05-15 by pims

saket424, 2017-07-12:
I am able to build the container and all the associated webrtc source code. I get an error starting the docker container. any idea why?

docker run -p 8000:8000 -it mpromonet/webrtc-streamer
Unable to find image 'mpromonet/webrtc-streamer:latest' locally
latest: Pulling from mpromonet/webrtc-streamer
c02c7df4a131: Already exists 
a3ed95caeb02: Already exists 
865fad5b0dc2: Already exists 
df758ad700b6: Already exists 
cae3bd937a24: Pull complete 
339c73548ae7: Pull complete 
e2e79f47ae79: Pull complete 
73e930cd219a: Pull complete 
Digest: sha256:a7987406a98b488829b92d12e8b4611aacde6f931384e1f470f5a04bc9e44e73
Status: Downloaded newer image for mpromonet/webrtc-streamer:latest
docker: Error response from daemon: containerd: container not started.

mpromonet, 2017-07-13:
There should be fixed now, this is a consequence I forgot changing the makefile

saket424, 2017-07-13:
Confirmed that your changes fixed it. the docker container is functional now. Thanks!