mpromonet/webrtc-streamer #30: add a new mode to simply relay rtp stream instead of transcode ?

saket424, 2017-04-10:
Michel,
Would it be possible to have a mode in webrtc-streamer to simply retrieve the rtsp using live555helper and then proxy the rtp media to the browser and not transcode at all. I understand transcoding gives us all the goodness of webrtc bandwidth adaptation but in some cases we want something very lightweight

I know you initially contributed to the janus streaming plugin but it does not play rtsp streams as well as this project does. This is an enhancemnet request.

Thanks for your consideration

saket424, 2017-04-11:
Possibly control whether or not to transcode behavior using URL query parameter

mpromonet, 2017-04-11:
Hi sacket424,

I think this is not possible using the webrtc framework, still the VideoCapturer doesnot accept compress format. webrtc-streamer decompress H264 without knowing what format negociated by the browser.
Maybe you can ask to webrtc support.

For lightweight, without transcoding, as you point it out, janus-gateway is a good solution.
My experience with it show clearly a lower latency, I remember some problems depending on the browser due to javascript webrtc used (webrtc-streamer use the old interface to be easily supported). It depends what do you called "it does not play rtsp strames as well..."

Best Regards,
Michel.

saket424, 2017-04-12:
For example , rtsp://217.17.220.110/axis-media/media.amp yields a=fmtp:96 packetization-mode=1; profile-level-id=420029; sprop-parameter-sets=Z0IAKeKQCgDLYC3AQEBpB4kRUA==,aM48gA== and chrome setRemoteDescription is not happy with the h264 profile but webrtc-streamer has no problem pulling and rendering the stream. So I guess for the time being transcoding cannot be avoided and I prefer webrtc-streamer project compared to janus-gateway since it is compatible with many more rtsp sources.

Closing this issue .

closed 2017-04-12 by saket424