mpromonet/webrtc-streamer #31: Dynamically add/remove rtsp url without restarting webrtc-streamer binary

saket424, 2017-04-12:
Michel,
Is it possible to add as an enhancement, a simple interface to add/remove new rtsp URLs without restarting the webrtc-streamer app. Right now all the URLs need to be specified at the start of the app.

Thanks

mpromonet, 2017-04-13:
Hi saket424,

The rtsp URLs could be given at start of the apps, but this is just to give a list to the HTML side.
In fact the HTML page sned the rtsp url to the server that connect to the backend.

Using index.html, you can add your own RTSP url in the text filed or adding in the http url using ?rtsp://217.17.220.110/axis-media/media.amp
Using webrtcstream.html, you need to specify the url as an argument like webrtcstream.html?rtsp://217.17.220.110/axis-media/media.amp
You can use the javascript API to connect to the URL with  webRtcServer.connect("rtsp://217.17.220.110/axis-media/media.amp").

In these 3 ways you don't need to restart the webrtc-streamer app.

Best Regards,

Michel.

saket424, 2017-04-13:
Got it. Thanks. Closing issue

closed 2017-04-13 by saket424