mpromonet/webrtc-streamer #32: rtmp support?

saket424, 2017-04-12:
Michel,
Any plans to support rtmp URLs in webrtc-streamer? DJI drones and others have a provision to stream live video to a custiom nginx-rtmp server and having a flash player for playback is not a long term option and webrtc-streamer would be the perfect solution to transform the rtmp to a browser-friendly format with all the video bitrate adaptation etc as a bonus.
Thanks

saket424, 2017-04-13:
I need to investigate the use of crtmpserver. It appears that you can publish an rtmp stream to it and retrieve the same stream name as an rtsp stream from it or vice versa. Any experience with crtmpserver ?

mpromonet, 2017-04-13:
Hi,
I never tried rtmp, in my mind it was a sort of proprietary protocol develop by Adobe.
live555 doesnot support RMTP, then if you can convert from RMTP to RTSP using crmtpserver maybe it is a way to proceed.
Best Regards,
Michel.

saket424, 2017-04-13:
Closing issue. Will attempt with crtmpserver and update progress

closed 2017-04-13 by saket424