mpromonet/webrtc-streamer #33: Blank page with none of the local capture devices enumerated

saket424, 2017-04-13:
I just built HEAD of webrtc-server and when I run it without any rtsp URLs, it does not enumerate any of the local capture devices to select from. I just get a blank page. Has the usage changed?

./webrtc-server__Release 
Logger level:4
[000:000] [26594] (audio_device_generic.cc:51): BuiltInAECIsAvailable: Not supported on this platform
[000:000] [26594] (audio_device_generic.cc:61): BuiltInAGCIsAvailable: Not supported on this platform
[000:000] [26594] (audio_device_generic.cc:71): BuiltInNSIsAvailable: Not supported on this platform
[000:000] [26594] (adm_helpers.cc:47): Unable to set recording channel to kChannelBoth.
HTTP Listen at 0.0.0.0:8000
STUN Listening at 127.0.0.1:3478

mpromonet, 2017-04-13:
Hi,
You should look to the log enabling more verbosity.
If the /getDeviceList is empty probably the problems comes from webrtc or the usage webrtc-streamer use webrtc.

However I know I introduce a regression that makes local capture device could not be displayed if there is a space caracter in its name... and this occurs often.

Best Regards,
Michel.

saket424, 2017-04-13:
My device had underscores in it Integrated_Camera_HD and was not enumerated. I compiled with webrtc HEAD of today. Probably a new regression issue since webrtc lib is constantly changing

closed 2017-05-08 by mpromonet