mpromonet/webrtc-streamer #34: Add ability to specify max_bitrate and codec in URL query parameters

saket424, 2017-04-13:
Enhancement request. Similar to the ability to specify new rtsp URLs as part of the query parameters string, it would be nice to specify max bitrate and codec used . These query parameters can then be used to perform the appropriate sdp munging to instruct webrtc to control/override the defaults which is as high as 2Mbit VP8

mpromonet, 2017-05-16:
Hi,

The selection may depends on the browser you are using.

Here is some links that seems to explains you should modify the SDP received : 
* [is-there-a-way-to-choose-codecs-in-webrtc-peerconnection](http://stackoverflow.com/questions/21402990/is-there-a-way-to-choose-codecs-in-webrtc-peerconnection)
* [how-can-i-change-the-default-codec-used-in-webrtc](http://stackoverflow.com/questions/26924430/how-can-i-change-the-default-codec-used-in-webrtc)

Maybe setting some mediaConstraints creating the answer ?

Best Regards,
Michel.

saket424, 2017-05-16:
The codec and bitrate can be set in the janus videoroom cfg file for the janus use case
For the non-janus uses, the SDP offer/answer needs to change(sdp munging)

closed 2017-07-17 by saket424