mpromonet/webrtc-streamer #39: Too many open files (sockets)

9a4gl, 2017-04-20:
After a while when using webrtc streamer with a lot of rtsp different sources webrtc-streamer crash and prints to stderr:

Failed to DESCRIBE: unable to create stream socket: Too many open files
[err] evsignal_init: socketpair: Too many open files

mpromonet, 2017-04-22:
Hi,
Clear the RTP subsession are not closed :)
I modified this is this commit https://github.com/mpromonet/webrtc-streamer/commit/19aca9fb6e5bfde42b3d3f8c375680aa09f8c8f8
It should leaks less.
Best Regards,
Michel.

mpromonet, 2017-05-23:
Hi,
Do you still see filedescritors leaks ?
Regards
Michel.

9a4gl, 2017-05-23:
Hm, I got that while I was developing my webapp, during testing I got that after like 50-100 camera switching.. will test hopefully tomorrow and will report whether number of open files increase

9a4gl, 2017-05-24:
sudo lsof -p 10579 | wc -l
increases by 21-25 after switching camera
Today I found it with about 1000 files opened, when I start it has 63.
So I would say we still leak opened files.

mpromonet, 2017-05-25:
Hi,
Thanks for the update.
Are you using WebRTC in debug ? 
If it reach filedescriptor above 1024, live555 select will failed.
Regards,
Michel.

9a4gl, 2017-05-25:
It looks when it reach 1024 it crash and then get restarted by systemd.

commit 00a2b692 (2017-07-16): Free PeerConnectionObserver in hangUp()

PeerConnectionObserver instances are created and owned by the
PeerConnectionManager. They are tracked internally using
`peer_connectionobs_map_`, from which they are removed in `hangUp()`.

While the pointer was removed from the map, the instance itself was not
deallocated so far. This meant that memory leaked, and
~PeerConnectionObserver was never called. This also meant that the
PeerConnection was not closed, keeping the associated fds in use. This
patch explicitly names the pointer for clarity, and deletes the
PeerConnectionObserver at the end of `ha [... cut]
closed 2017-07-16 by mpromonet