mpromonet/webrtc-streamer #40: Stream Dont Start

daltonjoi, 2017-05-04:
./webrtc-server__Release "rtsp://192.168.0.31/axis-media/media.amp"
Logger level:4
[000:000] [3833] (webrtcvoiceengine.cc:640): webrtc:   failed to connect context, error=-1
[000:000] [3833] (audio_device_pulse_linux.cc:172): failed to initialize PulseAudio
[000:001] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:002] [3833] (webrtcvoiceengine.cc:640): webrtc:      snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:002] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:002] [3833] (webrtcvoiceengine.cc:640): webrtc:      snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:002] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:002] [3833] (webrtcvoiceengine.cc:640): webrtc:      snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:003] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:003] [3833] (webrtcvoiceengine.cc:640): webrtc:      unable to open playback device: No such file or directory (-2)
[000:003] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:003] [3833] (webrtcvoiceengine.cc:640): webrtc:      snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:003] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:003] [3833] (webrtcvoiceengine.cc:640): webrtc:     unable to open record device: No such file or directory
[000:004] [3833] (audio_device_generic.cc:51): BuiltInAECIsAvailable: Not supported on this platform
[000:004] [3833] (audio_device_generic.cc:61): BuiltInAGCIsAvailable: Not supported on this platform
[000:004] [3833] (audio_device_generic.cc:71): BuiltInNSIsAvailable: Not supported on this platform
[000:004] [3833] (adm_helpers.cc:47): Unable to set recording channel to kChannelBoth.
[000:004] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:004] [3833] (webrtcvoiceengine.cc:640): webrtc:      snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:004] [3833] (adm_helpers.cc:56): Unable to access microphone.
[000:005] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:005] [3833] (webrtcvoiceengine.cc:640): webrtc:      snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:005] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:005] [3833] (webrtcvoiceengine.cc:640): webrtc:     unable to open record device: No such file or directory
[000:005] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:005] [3833] (webrtcvoiceengine.cc:640): webrtc:      snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:005] [3833] (adm_helpers.cc:100): Unable to access speaker.
[000:005] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:006] [3833] (webrtcvoiceengine.cc:640): webrtc:      snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:006] [3833] (webrtcvoiceengine.cc:640): webrtc: GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:006] [3833] (webrtcvoiceengine.cc:640): webrtc:      unable to open playback device: No such file or directory (-2)
HTTP Listen at 0.0.0.0:8000
STUN Listening at 127.0.0.1:3478
answer:false

answer:[ "rtsp://192.168.0.31/axis-media/media.amp" ]

answer:{
   "iceServers" : [
      {
         "url" : "stun:stun.l.google.com:19302"
      }
   ]
}

body:{"type":"offer","sdp":"v=0\r\no=- 6240597405130466022 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:JoeM\r\na=ice-pwd:dQyMcihMjpALfGr7TgMIlaNe\r\na=fingerprint:sha-256 D5:E7:A7:56:1B:19:52:2C:06:64:BD:9A:00:FC:99:8B:E8:91:CC:C6:E1:F2:F7:42:BC:71:8C:15:10:25:60:F3\r\na=setup:actpass\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 isac/16000\r\na=rtpmap:104 isac/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 cn/32000\r\na=rtpmap:105 cn/16000\r\na=rtpmap:13 cn/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 98 100 102 127 97 99 101 125\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:JoeM\r\na=ice-pwd:dQyMcihMjpALfGr7TgMIlaNe\r\na=fingerprint:sha-256 D5:E7:A7:56:1B:19:52:2C:06:64:BD:9A:00:FC:99:8B:E8:91:CC:C6:E1:F2:F7:42:BC:71:8C:15:10:25:60:F3\r\na=setup:actpass\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=recvonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtpmap:100 H264/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=fmtp:100 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:102 red/90000\r\na=rtpmap:127 ulpfec/90000\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:125 rtx/90000\r\na=fmtp:125 apt=102\r\n"}
body:{"candidate":"candidate:2370302700 1 udp 2113937151 192.168.0.169 56396 typ host generation 0 ufrag JoeM network-cost 50","sdpMid":"audio","sdpMLineIndex":0}
answer:1

body:{"candidate":"candidate:2370302700 1 udp 2113937151 192.168.0.169 56398 typ host generation 0 ufrag JoeM network-cost 50","sdpMid":"video","sdpMLineIndex":1}
answer:1

[008:083] [3840] (PeerConnectionManager.cpp:606): Cannot create capturer rtsp://192.168.0.31/axis-media/media.amp
[008:083] [3840] (PeerConnectionManager.cpp:655): Cannot find stream
body:{"candidate":"candidate:842163049 1 udp 1677729535 131.161.41.22 56398 typ srflx raddr 192.168.0.169 rport 56398 generation 0 ufrag JoeM network-cost 50","sdpMid":"video","sdpMLineIndex":1}
answer:1

body:{"candidate":"candidate:842163049 1 udp 1677729535 131.161.41.22 56396 typ srflx raddr 192.168.0.169 rport 56396 generation 0 ufrag JoeM network-cost 50","sdpMid":"audio","sdpMLineIndex":0}
answer:1

[image]

mpromonet, 2017-05-04:
Hi
We can see from the log that rtsp capturer is not available : 
> [008:083] [3840] (PeerConnectionManager.cpp:606): Cannot create capturer rtsp://192.168.0.31/axis-media/media.amp
In order to enable the RTSP supportm you need to install live555 before building webrtc-streamer.
Best Regards,
Michel

daltonjoi, 2017-05-08:
Hello,

I have the same  problem after install live555 , i updated for the last version now  and a will try run docker.

daltonjoi, 2017-05-08:
Hello,

I have the same problem after install live555 , i updated for the last version now and a will try run docker.
[trace.txt](https://github.com/mpromonet/webrtc-streamer/files/984973/trace.txt)

mpromonet, 2017-05-08:
Hi,
The problem is during compilation, you should see HAVE_LIVE555 in the compilation command.
You can look to the [Docker](https://github.com/mpromonet/webrtc-streamer/blob/master/Dockerfile) configuration of the [Travis](https://github.com/mpromonet/webrtc-streamer/blob/master/.travis.yml) configuration. I configure the automated build using dockerhub, but it is too long and timeout.
The problem comes probably that live555 is not detected by the build process.
Best Regards,
Michel.

daltonjoi, 2017-05-08:
Perfect , wonderful, with parameter PREFIX=/usr/local solve my problem.

Thank you very much !!!

closed 2017-05-08 by daltonjoi