mpromonet/webrtc-streamer #41: Need for HTTP API /getAudioDeviceList and /getVideoDeviceList

saket424, 2017-05-16:
Michel,
HTTP API /getDeviceList needs to be extended from only enumerating Video Devices to enumerating both Audio and Video devices

I am having an issue in that the webrtc-streamer associates the wrong audio device with the video device for a Magewell v4l2(with audio). It picks the PC microphone as the input instead of the Magewell audio

Should the deviceId passed in as a constraint when CreateVideoSource and CreateAudioSources?

This sample in javascript works well https://webrtc.github.io/samples/src/content/devices/input-output/ . We need a similar C++ sample I guess

saket424, 2017-05-16:
This question was posed earlier in https://groups.google.com/forum/#!topic/discuss-webrtc/LnXeIA_95lo

mpromonet, 2017-05-16:
Hi,
Maybe you can define the audio source you would like to select as a default audio source.
Otherwise, as you said, it needs some modification to associate audio track and video track
Regards,
Michel.

saket424, 2017-05-16:
Michel,
Can you give me a spot in the code, where I can set the audio source or preferably an example? I assume it is in PeerConnection.cpp ?

saket424, 2017-06-26:
Michel,
I confirm that the audiourl parameter works!!
Thanks
http://webrtc-server-name:8000/?url=VideoDeviceNameWithoutQuotes&audiourl=AudioDeviceNameWithoutQuotes

If the device name has spaces replaces space with %20

closed 2017-06-26 by saket424