mpromonet/webrtc-streamer #42: Provide binary for RPi Zero and 1

nistath, 2017-06-26:
The build provided by Travis CI currently does not work on a Raspberry Pi Zero W.
Would it be possible to add builds for the Pi Zero (armv6h) for future releases?

nistath, 2017-06-26:
Here is gdb output:
```
(gdb) run
Starting program: /home/pi/wrtcstrm/webrtc-streamer
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0x00053840 in rtc::LogMessage::LogMessage(char const*, int, rtc::LoggingSeverity, rtc::LogErrorContext, int, char const*) ()
```

nistath, 2017-06-26:
I think it might have worked! https://travis-ci.org/mpromonet/webrtc-streamer/jobs/247299384
Can we get the binary released so that I can test it?

mpromonet, 2017-06-27:
Hi,
It doesnot run on a raspberry pi B, compiler option or cross-compiler is not good.
Regards,
Michel.

nistath, 2017-07-05:
Can you also please test [this build your travis-ci made](https://travis-ci.org/mpromonet/webrtc-streamer/jobs/249195228) and provide me the binary somehow so that I can also have a go at it?
Thank you for your work.

mpromonet, 2017-07-17:
Hi nistath,
You can try https://github.com/mpromonet/webrtc-streamer/releases/download/v0.0.3/webrtc-streamer_v0.0.3_armv6.tgz, but it doesnot works on raspberry piB.
It exists with Illegal instruction, then probably it will not works on Pi Zero too.
Regards,
Michel.

mpromonet, 2017-08-10:
Hi nistash,
After some numers of tries,  I build for armv6, the package https://github.com/mpromonet/webrtc-streamer/releases/download/v0.0.4/webrtc-streamer_v0.0.4_rpi-armv6.tgz is running on my raspberry pi B.
I guess it could run on a pi zero. The compilation of WebRTC have less feature, but it works with the raspbicam.
Let me know if it works.
Best Regards,
Michel.

closed 2017-08-31 by mpromonet