mpromonet/webrtc-streamer #43: Webrtc -> Rtsp Streaming

GhostGumm, 2017-07-06:
Hi Mr Mpromonet, 

I'am Hakeem, i posted a question on Stackoverflow two days earlier about your Live555 application on WebRTC. 

My question was : is it possible to modify your application to stream directly from FFMPEG / live555 to a web browser ? 
I have a live555 + FFMPEG streamer which sends in the Rtsp media flux to a vlc program. But what i want to do is to stream that flux directly to a web browser, using WebRtc. But unfortunately i can't figure it out right now. 
 I asked if (you understand my question, i can go further in details) you think it is possible, and if you can give me hints on where to operate to make the transformation ? 

Any help would be much appreciated.

mpromonet, 2017-07-08:
Hi Hakeem,
The input that are streamed by thias application are created in  PeerConnectionManager::CreateVideoTrack.
Till know there is 2 kind of sources : 
- WebRtcVideoDeviceCapturer (belongs to WebRTC code) that are capture device /dev/videoXX
- RTSPVideoCapturer (belongs to this project) that accept RTSP stream (H264 orr JPEG) 

[RTSPVideoCapturer](src/rtspvideocapturer.cpp) use live555 for RTSP/RTP,  h264itstream to decode SPS/PPS (WebRTC manage less case), and WebRTC to uncompress H264 (using ffmpeg).
You can inspired from this code to write your own VideoCapturer based of live555 and ffmpeg.

Best Regards,
Michel.

GhostGumm, 2017-07-18:
Thank you very much for your hints Michel. (Sorry for answering late) 
I'm digging toward this direction.
I still have a little question, is it normal that i don't see any image at : 
https://webrtc-streamer.herokuapp.com/
Or is it that i have to toggle some options on my Browser ? 
Best Regards, 
Hakeem.

closed 2017-08-31 by mpromonet