mpromonet/webrtc-streamer #44: Cannot build

spascou, 2017-07-06:
Hello,

I cannot  build with current version of WebRTC native code; folder structure is as follows : 
`~/Projects/webrtc/webrtc-streamer/...`
`~/Projects/webrtc/src/out/Release/...`

Build command is `make WEBROOT=../webrtc WEBRTCBUILD=Release PREFIX=/usr/local`
Build log is here [https://pastebin.com/LS6jtFk3](https://pastebin.com/LS6jtFk3)
System is up to date Debian 9

It seems to me that the linking process fails, though I did not succeed to understand why; WebRTC was proprely built beforehand.

Could you help me please ?

Thanks !

mpromonet, 2017-07-08:
Hi,
This problems occurs updating in webrtc, it seems use_custom_libcxx default value changed.
In order to link with g++ it is needed to build WebRTC with `use_custom_libcxx=false`
I will fix this.
Regards,
Michel.

closed 2017-07-08 by mpromonet

spascou, 2017-07-10:
I've seen your fix; you might also want to update README.md to take into account the need of `use_custom_libcxx=false` for the _Build WebRTC with H264 support_ category.

Thanks for your hard work !

commit d23f96b0 (2017-07-08): fix #44: fix build with latest webrtc that needs use_custom_libcxx=false