mpromonet/webrtc-streamer #48: Can't compile webrtc-streamer

GhostGumm, 2017-07-24:
Hi Mr Promonet,

I may request your help because i have a very hard time compiling your project. 
Few weeks ago i managed to do so, i modified the : VideoCapturer, PeerConnection and Html server to fit my needs. 
It happend that i had a problem with my Ubuntu installation and i had to wipe it and start on a fresh new one. 
Since i'am trying to re compile your project but unfortunately i can't manage to do it. 
I mainly have a problem with the webrtc installation linkage. 
From what i understood, the Streamer Makefile go fetch for json, loggin, and some few other header in the /src/webrtc/base or /src/webrtc/*. 
I did everything according to your Readme and even on the webrtc web site. But it allways fails. 

I can give the logs if they can be of any help. I'am compiling under Ubuntu 16.04 LTS 

Best Regards.

Hakeem

Mohmoulay, 2017-07-24:
from my experience with your issue compiling it with Ubuntu is a nightmare i recommend you use a VM only for this with Kubuntu or any other flavor , and everything will run smoothly use docker containers too that's my suggestions

GhostGumm, 2017-07-24:
Thank you for your help Mr Mohmoulay. 
I ll try you recommendations right away.

govnoid, 2017-07-24:
HI, according to webrtc repository, they renamed src/webrtc/base to src/webrtc/rtc_base.
So you have to modify a couple of headers and .cpp files replacing /base/ with /rtc_base/.
Following files have to be modified:

        modified:   inc/PeerConnectionManager.h
        modified:   inc/rtspvideocapturer.h
        modified:   src/main.cpp
        modified:   src/rtspvideocapturer.cpp

mpromonet, 2017-07-25:
Hi govnoid,
The last build was ok https://travis-ci.org/mpromonet/webrtc-streamer but WebRTC developper often move files, I will check and update path.
Thanks & Regards,
Michel.

commit 7c5e1964 (2017-07-25): fix #48: fix build with webrtc@ec390b5
closed 2017-07-25 by mpromonet