mpromonet/webrtc-streamer #49: example of joining janus videoroom using NodeJS

antongisli, 2017-08-17:
Hi, you wrote in your readme that it should be possible to do this. 

The main usecase for this is that imagine you have say a raspberry pi with a remote rtsp camera, and you want it to collect the video and publish it to a janus hosted video room.

Have you actually gotten this to work using NodeJS? If so, would you be so kind as to post an example of how to do it?

mpromonet, 2017-08-19:
Hi antongisli,
I modify a little https://github.com/mpromonet/webrtc-streamer/tree/master/html/janusvideoroom.js in order to allow to use it in nodejs.

Then giving a send command using node you can do something like : 
```
function send(method,headers,data,onSuccess,onFailure,scope) {
	
	console.log("HTTP call "+ method);
	var request = require('request');
	var verb = 'GET';
	if (data) {
		verb = 'POST';
		data = JSON.stringify(data);
	}	
	request({
			method: verb,
			uri: method,
			body: data,
		},
		function (error, response, body) { 
			console.log("HTTP code:"+ response.statusCode);
			if ( (response.statusCode === 200) && onSuccess ) {
				onSuccess.call(scope,JSON.parse(body));
			}
			else if (onFailure) {
				onFailure.call(scope,response.statusCode);
			}
		}
	)
}

var JanusVideoRoom = require('./janusvideoroom.js'); 
var janus = new JanusVideoRoom("http://192.168.0.15:8088/janus", null, "http://192.168.0.15:8000", send)
janus.join(1234,"mmal service 16.1","video")
```
You just need to give : 

 * the URL of janus
 * the URL of webrtc-streamer
 * the source of the webrtc-streamer you would like to publish to janus.

You can also try some node module like https://www.npmjs.com/package/janus-videoroom-client. You need to forward the createOffer answer from webrtc-streamer to janus-gateway and the publish answer from janus-gateway to webrtc-streamer

Best Regards,
Michel.

closed 2017-09-01 by mpromonet

antongisli, 2017-09-03:
Thank you Michel! I am still struggling with transcoding some h264 from a camera into constrained baseline on a raspberry pi. Once I get that working I will give this tip of yours a go. Thank you again.

mpromonet, 2017-09-04:
Hi antongisli,
If you are using the v4l2 driver for raspicam, you can configure the contrained base profile using: 

    v4l2-ctl -c h264_profile=1
Best Regards,
Michel.

antongisli, 2017-09-04:
Hi Michel,

I am trying to use an RTSP camera. It's h264, main profile, level 4.1 or
something and there is no way to change the encoding settings on the
camera. so I need to transcode to "webrtc" compliant (baseline,
constrained) using an RPI. I'm not sure that the v4l2 driver can help me
for RTSP?

Ive been trying to use the gstreamer omx plugin but the pipeline always
crashes when I try to use "level=2.1" or whatever. Using the latest code
too. Do you know any other ways around this? I couldnt find in ffmpeg how
to set the level either.

Regards,
Anton