mpromonet/webrtc-streamer #51: stream /dev/video to hangouts or similar services

marcheschi, 2017-09-01:
Hi
First of all thank you for your effort, 
this is a very beautiful project in order to simplify webrtc communications.
I'm looking for a simple way to stream a video device to google hangouts or janus or jangouts video rooms .
Is there a way to use your tool to do that. 
I think at something like that:
./webrtc-streamer /dev/video1 https://hangouts.google.com/call/ktir5k7shbduhfubougepbx3euy

or to use piping with gstreamer or cvlc or ffmpeg.

Thank you
Paolo Marcheschi

mpromonet, 2017-09-01:
Hi Paolo,

It is possible to send the stream to a Janus Video Room, there is a simple sample code in the readme.

A live demo is available https://webrtc-streamer.herokuapp.com/janusvideoroom.html
Changing the Janus URL, you can publish a stream to a video room.

Best Regards,
Michel.

marcheschi, 2017-09-04:
Hi
Thank you,
I installed the webrtc distribution on my linux host but it does not seem to work in my internal network.
the page http://localhost:8000/janusvideoroom.html
works and I can connect to https://janus.conf.meetecho.com/videoroomtest.html

if I connect to my working janus https://janus.mydomain.my/janus/demos/videoroomtest.html

It does not.
I tried different urls:
 https://janus.mydomain.my/janus
 https://janus.mydomain.my/janus/demos
 https://janus.mydomain.my/

But with no luck.
How can I troubleshoot it.
Thank you
Paolo

mpromonet, 2017-09-18:
Hi Paolo,
The url should be the url of the janus REST service, not the url of the HTML served by apache, nginx or an other http server. 
You should try http://janus.mydomain.my:8088/janus or https://janus.mydomain.my:8089/janus.
Theses ports are configured in janus.transport.http.cfg
Best Regards,
Michel.

closed 2019-03-02 by mpromonet