mpromonet/webrtc-streamer #52: video lags

63l06ri5, 2017-09-04:
Hello Michel!

I have a problem with a video streaming: it lags - it's like it plays for two seconds then stops and after a little while play another two and so on... I read your sources and can't understand where there can be a problem or how can i tweak this. Can you point it out, please?

There is a snippet of console logs when video is played:

```[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(78) is large
[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(81) is large
[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(84) is large
[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(87) is large
[h264 @ 0x7fe2acaec8a0] no frame!
[201:193] [601] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[h264 @ 0x7fe2acaec8a0] no frame!
[203:189] [601] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[h264 @ 0x7fe2acaec8a0] no frame!
[205:198] [601] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(3) is large
[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(3) is large
[h264 @ 0x7fe2acaec8a0] no frame!
[207:198] [601] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[h264 @ 0x7fe2acaec8a0] no frame!
[209:199] [601] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(3) is large
[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(6) is large
[h264 @ 0x7fe2acaec8a0] Increasing reorder buffer to 1
[OpenH264] this = 0x0x7fe28c019820, Warning:Actual input framerate 15.000000 is different from framerate in setting 22.000000, suggest to use other rate control modes
[h264 @ 0x7fe2acaec8a0] no frame!
[211:188] [601] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[h264 @ 0x7fe2acaec8a0] no frame!
[213:199] [601] (h264_decoder_impl.cc:330): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[OpenH264] this = 0x0x7fe28c019820, Warning:[Rc] iDid = 0,iContinualSkipFrames(3) is large
```

Thanks, Ilya

mpromonet, 2017-09-04:
Hi llya,
You're right, on a local network it seems less lag.
I guess this is linked to the timestamping and/or the variability of decompressing time, I look in webrtc code and decompressor doesnot have thread, and still know I haven't add a thread between incoming frame and decompressor.
This is an open point.
Best Regards,
Michel.

REPRESSION, 2018-03-09:
Hi I am facing the same problem, I have done some testing with an local IP-Camera, seems that this lag happens at each IFrame (once per sec with 1 IFrame per sec / once every 4 sec with 1 IFrame every 4 sec).

My output is nearly the same as yours :

[h264 @ x3266180] no frame!
[085:128] [3033] (h264_decoder_impl.cc:331): avcodec_send_packet error: -1094995529
[h264 @ x3266180] no frame!
[089:122] [3033] (h264_decoder_impl.cc:331): avcodec_send_packet error: -1094995529
... etc

here we can see that with 1 IFrame every 4 sec, it happens every 4 secs, so for me it seems related.
the difference between your output and mine is the ffmpeg version (avcodec_decode_video2 is the old way to decode video, avcodec_send/receive_packet is the new one) and your IFrame rate (seems to be 2 secs for you).

Do you have any idea how to fix this lag / decoding error ?

REPRESSION, 2018-03-13:
It seems like this error comes from NAL_UNIT_TYPE_SEI frames, when skipped the error message disappears but the lag remains.

mpromonet, 2018-03-13:
Hi,
The SEI stuff was reported in https://github.com/mpromonet/webrtc-streamer/issues/89.
Since my latest answer, I add a thread to avoid to block the input data, however this is not enough, the encoder seems to drop frames, maybe timestamp of frame is incorrect, maybe it is needed to include a dejitter buffer..
Regards,
Michel.

REPRESSION, 2018-03-14:
When i drastically reduce the Camera's resolution output (to 240p for example) the lag is almost not noticeable, so it seems that those bigger IFrames are the cause of the lag, because they take more time to be decoded/encoded.
I am looking to adapt your code to send directly h264, and avoid transcoding, but this is not trivial.