mpromonet/webrtc-streamer #54: Demo links  shows no video just shows blank with header on the top  and also my compiled code also shows the same thing

prashanthag, 2017-09-06:


Jamie0, 2017-09-07:
I can replicate this, too. Latest Firefox and Chromium.

mpromonet, 2017-09-07:
Hi prashanthag & Jamie0,

Probably something changes in WebRTC, this occurs time to times.
Did you tried with the latest release https://github.com/mpromonet/webrtc-streamer/releases/tag/v0.0.5 ?

Best Regards,
Michel.

proxy-m, 2017-09-08:
https://github.com/mpromonet/webrtc-streamer/pull/56

prashanthag, 2017-09-08:
Hi mpro
I updated code v0.0.5(tag)  but still am not getting error 
I get this error   ----- AudioSendStream::Stop failed with error: -**1   

[166:608] [2702] (webrtcvoiceengine.cc:701): webrtc: StopSend() failed to stop recording (error=10030)
[166:608] [2702] (audio_send_stream.cc:255): **AudioSendStream::Stop failed with error: -**1
Received 188 new bytes of response data.
Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 4
Date: Fri, Sep 08 2017 18:43:42 GMT
Range: npt=0.000-
Session: 815ECE6D
RTP-Info: url=rtsp://192.168.1.198:5560/source2/track1;seq=62978;rtptime=3061731423

Received 188 new bytes of response data.
Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 4
Date: Fri, Sep 08 2017 18:43:42 GMT
Range: npt=0.000-
Session: 07AC639B
RTP-Info: url=rtsp://192.168.1.198:5560/source2/track1;seq=62979;rtptime=3061731453

[166:609] [2702] (webrtcvoiceengine.cc:701): webrtc: StopSend() failed to stop recording (error=10030)
[166:609] [2702] (audio_send_stream.cc:255): A**udioSendStream::Stop failed with error: -1
answer:{**
   "sdp" : "v=0\r\no=- 6735096114903117477 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video\r\na=msid-semantic: WMS rtsp://192.168.1.198:5560/source2\r\nm=audio 56728 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 110 112 113 126\r\nc=IN IP4 192.168.1.170\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=candidate:25531464 1 udp 2122260223 192.168.1.170 56728 typ host generation 0 network-id 1 network-cost 50\r\na=ice-ufrag:VJ3i\r\na=ice-pwd:acUM4FPHvBPmjiLzuQRSSEcp\r\na=ice-options:trickle\r\na=fingerprint:sha-256 CB:83:FA:E4:81:7A:41:44:83:6E:A5:E1:91:4B:CA:0F:2C:45:E9:E1:E9:2E:F2:8F:57:25:02:95:57:0F:F0:5B\r\na=setup:active\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=sendonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=rtcp-fb:111 transport-cc\r\na=fmtp:111 minptime=10;useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:104 ISAC/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 CN/32000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:110 telephone-event/48000\r\na=rtpmap:112 telephone-event/32000\r\na=rtpmap:113 telephone-event/16000\r\na=rtpmap:126 telephone-event/8000\r\na=ssrc:2962036377 cname:gOtb0IaLEGrpHmnZ\r\na=ssrc:2962036377 msid:rtsp://192.168.1.198:5560/source2 audio_label\r\na=ssrc:2962036377 mslabel:rtsp://192.168.1.198:5560/source2\r\na=ssrc:2962036377 label:audio_label\r\nm=video 9 UDP/TLS/RTP/SAVPF 96 98 100 102 127 97 99 101 125\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:VJ3i\r\na=ice-pwd:acUM4FPHvBPmjiLzuQRSSEcp\r\na=ice-options:trickle\r\na=fingerprint:sha-256 CB:83:FA:E4:81:7A:41:44:83:6E:A5:E1:91:4B:CA:0F:2C:45:E9:E1:E9:2E:F2:8F:57:25:02:95:57:0F:F0:5B\r\na=setup:active\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=extmap:5 http://www.ietf.org/id/draft-holmer-rmcat-transport-wide-cc-extensions-01\r\na=extmap:6 http://www.webrtc.org/experiments/rtp-hdrext/playout-delay\r\na=sendonly\r\na=rtcp-mux\r\na=rtcp-rsize\r\na=rtpmap:96 VP8/90000\r\na=rtcp-fb:96 ccm fir\r\na=rtcp-fb:96 nack\r\na=rtcp-fb:96 nack pli\r\na=rtcp-fb:96 goog-remb\r\na=rtcp-fb:96 transport-cc\r\na=rtpmap:98 VP9/90000\r\na=rtcp-fb:98 ccm fir\r\na=rtcp-fb:98 nack\r\na=rtcp-fb:98 nack pli\r\na=rtcp-fb:98 goog-remb\r\na=rtcp-fb:98 transport-cc\r\na=rtpmap:100 H264/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtcp-fb:100 transport-cc\r\na=fmtp:100 level-asymmetry-allowed=1;packetization-mode=1;profile-level-id=42e01f\r\na=rtpmap:102 red/90000\r\na=rtpmap:127 ulpfec/90000\r\na=rtpmap:97 rtx/90000\r\na=fmtp:97 apt=96\r\na=rtpmap:99 rtx/90000\r\na=fmtp:99 apt=98\r\na=rtpmap:101 rtx/90000\r\na=fmtp:101 apt=100\r\na=rtpmap:125 rtx/90000\r\na=fmtp:125 apt=102\r\na=ssrc-group:FID 1434768014 4143955276\r\na=ssrc:1434768014 cname:gOtb0IaLEGrpHmnZ\r\na=ssrc:1434768014 msid:rtsp://192.168.1.198:5560/source2 video_label\r\na=ssrc:1434768014 mslabel:rtsp://192.168.1.198:5560/source2\r\na=ssrc:1434768014 label:video_label\r\na=ssrc:4143955276 cname:gOtb0IaLEGrpHmnZ\r\na=ssrc:4143955276 msid:rtsp://192.168.1.198:5560/source2 video_label\r\na=ssrc:4143955276 mslabel:rtsp://192.168.1.198:5560/source2\r\na=ssrc:4143955276 label:video_label\r\n",
   "type" : "answer"
}

body:{"candidate":"candidate:3707591233 1 udp 2113937151 192.168.1.67 58698 typ host generation 0 ufrag /rkl network-cost 50","sdpMid":"video","sdpMLineIndex":1}
answer:1

body:{"candidate":"candidate:833594220 1 udp 2113932031 2001::9d38:953c:18b2:26d0:b8f9:f728 58699 typ host generation 0 ufrag /rkl network-cost 50","sdpMid":"video","sdpMLineIndex":1}
answer:1

[166:633] [2702] (voe_base_impl.cc:618): StartSend: Failed to initialize recording
[166:633] [2702] (webrtcvoiceengine.cc:701): webrtc: StartSend() failed to start recording (error=10028)
[166:633] [2702] (audio_send_stream.cc:244): AudioSendStream::Start failed with error: -1

prashanthag, 2017-09-08:
Thanks .. this works rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov ..  but not RTSP with 1080p60 ..does it have limitation for resolution and bandwidth

prashanthag, 2017-09-08:
it can display rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.mov.. with your new fix ...
it decodes successfully 1080p60 rtsp stream(local stream) as per log .. no error .. but it can not display this 1080p60 6mb video ? should I need to change some parameter to display this 1920x1080 resolution ?

Jamie0, 2017-09-08:
Works for me now :)

mpromonet, 2017-09-09:
hi
thanks to proxy-m for the fix.
@prashanthag if network loose i-frame nothing could be decoded. you can try to add to the url `?options=rtptransport%3dtcp`
then it will ask for tcp transport between rtsp source and webrtc-streamer
Best regards
Michel

closed 2017-09-12 by mpromonet