mpromonet/webrtc-streamer #57: Local video  -To-> WebRTC-To->RTSP  [question]

GhostGumm, 2017-09-13:
Hi Mr Promonet, 
I would like to ask a question about the rtsp streaming part. 
I read well your code but still i have some interrogations. 
My question is : If i want to implement my own video capture system : Local FFMPEG Capture on a local  video source and stream via Live555.
Where do i need to cut ? Is the parsing of the candidate and registration of the IP is enough ? Is it possible to have any clues please ?

Thank you for your work and time.

mpromonet, 2017-09-13:
Hi GhostGumm,
I have some difficulties to understand your question, this project stream WebRTC, not RTSP.
For capturing using ffmpeg and stream using live555, there is no need of WebRTC stuff, just implement the [live555 DeviceSource sample](https://github.com/xanview/live555/blob/master/liveMedia/DeviceSource.cpp) to your ffmpeg capture.
Best Regards,
Michel.

GhostGumm, 2017-09-14:
Hi Mr Promonet, 
Sorry for my explanations, i will put it more simply.
My goal is to find the right  IP and Port to stream from my Live555 setup. 
But sometimes i need WebRTC to help me to resolve NATs and find the real IP of my client who seeks the stream. 
I found WebRTC handy to do that, with the SDP negociation and ice server operations.
So if i wanted to only keep signaling and the WebRTC SDP exchange, where would i have to cut in the Streamer Project to not make the code unstable ? 
I also have hard time to find where to get the IP of my client and open Port once the negotiation is over.

Hoping my questions are clear.
 
Thank you very much for your time.

GhostGumm, 2017-09-14:
Hi Mr Promonet,

If i understand well the streamer can use WebRTC to resolve NATs, get the real IP of the client, open a port on the client using SDP exchange, then stream directly to the front end client but not using RTSP.
What is done here is that the video capturer capture a RTSP stream then give it to a front end client through WebRTC, not through RTSP.

Sorry if i don't understand well everything but i am trying to wrap around your work and webrtc. 

Thank you very much for your time.

closed 2019-05-01 by mpromonet