mpromonet/webrtc-streamer #58: demo on herokuapp

antopar, 2017-09-13:
Hi mpromonet,

demo page at https://webrtc-streamer.herokuapp.com/ doesn't work properly.
Firefox tells: "ICE failed, add a TURN server and see about:webrtc for more details"

I tryed in my local machine and it works only with local streamer (VLC), what is the problem?
Thankyou

mpromonet, 2017-09-13:
Hi antopar,

I don't think it is a STUN/TURN issue, it seems to works better with Chrome.
Till know I did now understand why ICE negotiation fails under firefox, maybe a iceservers arguments are not formatted correctly.
This need to be investigated....
Regards,
Michel.

antopar, 2017-09-14:
Hi mpromonet,
thank you for your availability.

In my local network all works fine. In a real network there are some issues.
Chrome does exactly the same on herokuapp, so I'm trying to deploy your streamer in a remote server.
Yesterday I tryed to deploy your streamer in a remote server and I cannot stream.

There is something to kwon in order to properly deploy?
Thank you

mpromonet, 2017-10-14:
Hi antopar,
In your log we see RTSPVideoCapturer::Decoded, I guess the wideo is decoded, as it complains about audio, maybe you can try to disable it setting audiourl to dummy.
Best Regards,
Michel.

closed 2018-01-01 by mpromonet