mpromonet/webrtc-streamer #60: webrtc part xlibs linkage errors

Sfinx, 2017-09-27:
The last git failed to link the webrtc-streamer:

```
libWebRTC__Release.a(audio_device_alsa_linux.o): In function `webrtc::AudioDeviceLinuxALSA::Terminate() [clone .part.11]':
audio_device_alsa_linux.cc:(.text._ZN6webrtc20AudioDeviceLinuxALSA9TerminateEv.part.11+0x9b): **undefined reference to `XCloseDisplay'**
libWebRTC__Release.a(audio_device_alsa_linux.o): In function `webrtc::AudioDeviceLinuxALSA::Init()':
audio_device_alsa_linux.cc:(.text._ZN6webrtc20AudioDeviceLinuxALSA4InitEv+0x83): **undefined reference to `XOpenDisplay'**
```

Any configure opts to webrts are missing ? Or the xlibs are really needed for webrtc-streamer ?

mpromonet, 2017-09-27:
Hi sfinx,
Xlib is not needed building webrtc with 'use_ozone=true'
You could find webrtc build options in .travis.yml file.
Best regards
Michel

Sfinx, 2017-09-27:
Adding the -lX11 still do not helps. I'm seeing the similar logs at the closed issue #9

```
g++ -o webrtc-streamer src/PeerConnectionManager.o src/rtspvideocapturer.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a live555helper/live555helper.a civetweb/libcivetweb.a h264bitstream/.libs/libh264bitstream.a -pthread --sysroot=/ -ldl -lrt  live555helper/live555helper.a -L //usr/local/lib -l:libliveMedia.a -l:libgroupsock.a -l:libUsageEnvironment.a -l:libBasicUsageEnvironment.a  -L civetweb -l civetweb h264bitstream/.libs/libh264bitstream.a -lX11 
src/PeerConnectionManager.o: In function `PeerConnectionManager::getMediaList()':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:114: undefined reference to `Json::Value::Value(std::string const&)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:124: undefined reference to `Json::Value::Value(std::string const&)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::getVideoDeviceList()':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:159: undefined reference to `Json::Value::Value(std::string const&)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::getAudioDeviceList()':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:187: undefined reference to `Json::Value::Value(std::string const&)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::getIceServers()':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:201: undefined reference to `Json::Value::Value(std::string const&)'
src/PeerConnectionManager.o:/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:209: more undefined references to `Json::Value::Value(std::string const&)' follow
src/PeerConnectionManager.o: In function `PeerConnectionManager::addIceCandidate(std::string const&, Json::Value const&)':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:230: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:231: undefined reference to `rtc::GetIntFromJsonObject(Json::Value const&, std::string const&, int*)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:232: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:238: undefined reference to `webrtc::CreateIceCandidate(std::string const&, int, std::string const&, webrtc::SdpParseError*)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::createOffer(std::string const&, std::string const&, std::string const&, std::string const&)':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:323: undefined reference to `Json::Value::Value(std::string const&)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:324: undefined reference to `Json::Value::Value(std::string const&)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::setAnswer(std::string const&, Json::Value const&)':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:343: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:344: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:350: undefined reference to `webrtc::CreateSessionDescription(std::string const&, std::string const&, webrtc::SdpParseError*)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::call(std::string const&, std::string const&, std::string const&, std::string const&, Json::Value const&)':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:380: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:381: undefined reference to `rtc::GetStringFromJsonObject(Json::Value const&, std::string const&, std::string*)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:418: undefined reference to `webrtc::CreateSessionDescription(std::string const&, std::string const&, webrtc::SdpParseError*)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:463: undefined reference to `Json::Value::Value(std::string const&)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:464: undefined reference to `Json::Value::Value(std::string const&)'
src/PeerConnectionManager.o: In function `PeerConnectionManager::getPeerConnectionList()':
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:591: undefined reference to `Json::Value::Value(std::string const&)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:604: undefined reference to `Json::Value::Value(std::string const&)'
/home/rus/Projects/ring/ast/webrtc-streamer/src/PeerConnectionManager.cpp:604: undefined reference to `Json::Value::operator[](std::string const&)'
[... cut ...]
/home/rus/Projects/ring/ast/webrtc-streamer/../webrtc/src/media/base/mediachannel.h:80: undefined reference to `webrtc::RtpExtension::ToString() const'
src/main.o: In function `std::string cricket::VectorToString<cricket::DataCodec>(std::vector<cricket::DataCodec, std::allocator<cricket::DataCodec> > const&)':
/home/rus/Projects/ring/ast/webrtc-streamer/../webrtc/src/media/base/mediachannel.h:80: undefined reference to `cricket::DataCodec::ToString() const'
collect2: error: ld returned 1 exit status
```

Sfinx, 2017-09-27:
> Xlib is not needed building webrtc with 'use_ozone=true'
> You could find webrtc build options in .travis.yml file.
> Best regards
> Michel

But I have not changed the travis.yml, checked - it has the use-ozone=true. My build info:

```
Ubuntu 16.04
webrtc config:

gn gen out/Release --args='is_debug=false use_custom_libcxx=false rtc_use_h264=true ffmpeg_branding="Chrome" rtc_include_tests=false is_clang=false treat_warnings_as_errors=false rtc_enable_protobuf=false'
```

mpromonet, 2017-09-27:
Hi,
Usually this occurs because jsoncpp was not build or with different std options.
Webrtc is also often changing path, recently the level webrtc was removed (issue #9 is then completly obsolete)
You can try the docker
Best regards
Michel

Sfinx, 2017-09-27:
Hi, ok, trying docker...

closed 2017-09-27 by Sfinx

ozankonur, 2017-10-06:
Same error... Also When I try to docker, it says;

Logger level:4
[000:000] [1] (audio_device_alsa_linux.cc:150): failed to load symbol table
[000:000] [1] (audio_device_impl.cc:402): Audio device initialization failed.
[000:000] [1] (voe_base_impl.cc:190): Init() failed to initialize the ADM

#
# Fatal error in ../../media/engine/webrtcvoiceengine.cc, line 283
# last system error: 0
# Check failed: 0 == voe_wrapper_->base()->Init(adm_.get(), apm(), decoder_factory_) (0 vs. -1)
#
#

what can I do ?

mpromonet, 2017-10-08:
Hi ozankonur,
This should be fixed installing libasound-dev in the docker see commit  6530f2b.
Regards,
Michel.

823639792, 2017-10-10:
same problem
```hzy@ubuntu:~/webrtc/webrtc-streamer$ make WEBRTCROOT=../webrtc-checkout WEBRTCBUILD=Release PREFIX=/usr/local
ALSA not found in //usr/local/include
g++ -o webrtc-streamer src/PeerConnectionManager.o src/rtspvideocapturer.o src/HttpServerRequestHandler.o src/main.o libWebRTC__Release.a live555helper/live555helper.a civetweb/libcivetweb.a h264bitstream/.libs/libh264bitstream.a -pthread --sysroot=/ -ldl -lrt  live555helper/live555helper.a -L //usr/local/lib -l:libliveMedia.a -l:libgroupsock.a -l:libUsageEnvironment.a -l:libBasicUsageEnvironment.a  -L civetweb -l civetweb h264bitstream/.libs/libh264bitstream.a
libWebRTC__Release.a(audio_device_alsa_linux.o): In function `webrtc::AudioDeviceLinuxALSA::Init()':
audio_device_alsa_linux.cc:(.text._ZN6webrtc20AudioDeviceLinuxALSA4InitEv+0x41): undefined reference to `XOpenDisplay'
libWebRTC__Release.a(audio_device_alsa_linux.o): In function `webrtc::AudioDeviceLinuxALSA::Terminate()':
audio_device_alsa_linux.cc:(.text._ZN6webrtc20AudioDeviceLinuxALSA9TerminateEv+0xa4): undefined reference to `XCloseDisplay'
libWebRTC__Release.a(audio_device_alsa_linux.o): In function `webrtc::AudioDeviceLinuxALSA::RecThreadProcess()':
audio_device_alsa_linux.cc:(.text._ZN6webrtc20AudioDeviceLinuxALSA16RecThreadProcessEv+0x59e): undefined reference to `XQueryKeymap'
libWebRTC__Release.a(audio_device_alsa_linux.o): In function `webrtc::AudioDeviceLinuxALSA::KeyPressed() const':
audio_device_alsa_linux.cc:(.text._ZNK6webrtc20AudioDeviceLinuxALSA10KeyPressedEv+0x2a): undefined reference to `XQueryKeymap'
libWebRTC__Release.a(audio_device_pulse_linux.o): In function `webrtc::AudioDeviceLinuxPulse::Init()':
audio_device_pulse_linux.cc:(.text._ZN6webrtc21AudioDeviceLinuxPulse4InitEv+0x2f): undefined reference to `XOpenDisplay'
libWebRTC__Release.a(audio_device_pulse_linux.o): In function `webrtc::AudioDeviceLinuxPulse::Terminate()':
audio_device_pulse_linux.cc:(.text._ZN6webrtc21AudioDeviceLinuxPulse9TerminateEv+0x8a): undefined reference to `XCloseDisplay'
libWebRTC__Release.a(audio_device_pulse_linux.o): In function `webrtc::AudioDeviceLinuxPulse::ProcessRecordedData(signed char*, unsigned int, unsigned int)':
audio_device_pulse_linux.cc:(.text._ZN6webrtc21AudioDeviceLinuxPulse19ProcessRecordedDataEPajj+0xa7): undefined reference to `XQueryKeymap'
libWebRTC__Release.a(audio_device_pulse_linux.o): In function `webrtc::AudioDeviceLinuxPulse::KeyPressed() const':
audio_device_pulse_linux.cc:(.text._ZNK6webrtc21AudioDeviceLinuxPulse10KeyPressedEv+0x2a): undefined reference to `XQueryKeymap'
collect2: error: ld returned 1 exit status
Makefile:98: recipe for target 'webrtc-streamer' failed
make: *** [webrtc-streamer] Error 1
```

ozankonur, 2017-10-10:
@823639792 

I solved that problem with changing the Makefile. I added lX11 to end of the LDFLAGS in ALSA part like;

# alsa-lib
ifneq ($(wildcard $(SYSROOT)$(PREFIX)/include/alsa/asoundlib.h),)
CFLAGS += -DHAVE_ALSA -I $(SYSROOT)$(PREFIX)/include
LDFLAGS+= -L  $(SYSROOT)$(PREFIX)/lib -lasound -lX11
else
$(info ALSA not found in $(SYSROOT)$(PREFIX)/include)
endif

823639792, 2017-10-10:
@ozankonur thanks,I find my code run into the
``` 
$(info ALSA not found in $(SYSROOT)$(PREFIX)/include)
```
branch, but I am not find alsa/asoundlib.h anywhere

mpromonet, 2017-10-10:
@823639792 
first you should build webRTC without needs of X11, then using `use_ozone=true rtc_include_pulse_audio=false`.
Next you should use the same sysroot as WebRTC build, if you build webrtc with use_sysroot=true (the deafult) you should add to the make command 
SYSROOT=<path to WebRTC>/src/build/linux/debian_jessie_amd64-sysroot. alsa/asound.lib.h is inside webrtc sysroot.
Best Regards,
Michel.

ozankonur, 2017-10-10:
@823639792 you can check under /usr/include and if you find alsa there, please copy it to under /usr/local/include.. I solved it like that

823639792, 2017-10-10:
@mpromonet  I find you update README.md just now, but still not OK.

823639792, 2017-10-10:
@ozankonur 
I update my peoject to newest.
now my error is.
```
hzy@ubuntu:~/webrtc/webrtc-streamer$ make WEBRTCROOT=../webrtc WEBRTCBUILD=Release PREFIX=/usr    SYSROOT=../webrtc/src/build/linux/debian_jessie_amd64-sysroot
g++ -o src/rtspvideocapturer.o -c src/rtspvideocapturer.cpp -Wall -pthread -g -std=c++11 -Iinc --sysroot=../webrtc/src/build/linux/debian_jessie_amd64-sysroot  -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -DHAVE_JPEG -I ../webrtc/src -I ../webrtc/src/third_party/jsoncpp/source/include -I ../webrtc/src/third_party/libyuv/include -DNDEBUG=1 -DHAVE_ALSA -I ../webrtc/src/build/linux/debian_jessie_amd64-sysroot/usr/include -DHAVE_LIVE555 -I live555helper/inc -I ../webrtc/src/build/linux/debian_jessie_amd64-sysroot/usr/include/liveMedia  -I ../webrtc/src/build/linux/debian_jessie_amd64-sysroot/usr/include/groupsock -I ../webrtc/src/build/linux/debian_jessie_amd64-sysroot/usr/include/UsageEnvironment -I ../webrtc/src/build/linux/debian_jessie_amd64-sysroot/usr/include/BasicUsageEnvironment/ -I civetweb/include -I h264bitstream -DVERSION="\"v0.0.5-37-g585433e-dirty webrtc@d65ae4a live555helper@7f49624 civetweb@v1.9.1-177-gd9b713b h264bitstream@0.2.0\""
In file included from src/rtspvideocapturer.cpp:21:0:
inc/rtspvideocapturer.h: In member function ‘virtual bool RTSPAudioSource::onData(const char*, unsigned char*, ssize_t, timeval)’:
inc/rtspvideocapturer.h:157:29: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      while (m_buffer.size() > segmentLength*m_channel) {
                             ^
src/rtspvideocapturer.cpp: In function ‘int decodeRTPTransport(const string&)’:
src/rtspvideocapturer.cpp:27:21: error: ‘RTPUDPUNICAST’ is not a member of ‘RTSPConnection’
  int rtptransport = RTSPConnection::RTPUDPUNICAST;
                     ^
src/rtspvideocapturer.cpp:29:18: error: ‘RTPOVERTCP’ is not a member of ‘RTSPConnection’
   rtptransport = RTSPConnection::RTPOVERTCP;
                  ^
src/rtspvideocapturer.cpp:31:18: error: ‘RTPOVERHTTP’ is not a member of ‘RTSPConnection’
   rtptransport = RTSPConnection::RTPOVERHTTP;
                  ^
src/rtspvideocapturer.cpp:33:18: error: ‘RTPUDPMULTICAST’ is not a member of ‘RTSPConnection’
   rtptransport = RTSPConnection::RTPUDPMULTICAST;
                  ^
src/rtspvideocapturer.cpp: In member function ‘virtual bool RTSPVideoCapturer::onData(const char*, unsigned char*, ssize_t, timeval)’:
src/rtspvideocapturer.cpp:119:37: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if ( (GetCaptureFormat()->width != width) || (GetCaptureFormat()->height != height) )  {
                                     ^
src/rtspvideocapturer.cpp:119:78: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if ( (GetCaptureFormat()->width != width) || (GetCaptureFormat()->height != height) )  {
                                                                              ^
src/rtspvideocapturer.cpp: In member function ‘virtual ssize_t RTSPVideoCapturer::onNewBuffer(unsigned char*, ssize_t)’:
src/rtspvideocapturer.cpp:193:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if (size > sizeof(marker))
            ^
Makefile:94: recipe for target 'src/rtspvideocapturer.o' failed
make: *** [src/rtspvideocapturer.o] Error 1

```

mpromonet, 2017-10-10:
@823639792
This means the submodule live555helper is not the correct align with webrtc-streamer, you should update it or use a fresh clone of the project.
Best Regards,
Michel.