mpromonet/webrtc-streamer #61: do not work with h264 video from stock gstreamer rtsp-server

Sfinx, 2017-09-27:
Started the local stream with gst-rtsp-server (video only), checked it is playing ok witj gst-play and ffplay. Started webrtc-streamer docker:

```
docker run -p 8000:8000 -it mpromonet/webrtc-streamer rtsp://10.1.1.208:8554/ring0
Logger level:4
[000:000] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL 
[000:005] [1] (audio_mixer_manager_alsa_linux.cc:196): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:006] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL 
[000:006] [1] (audio_mixer_manager_alsa_linux.cc:286): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:006] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL 
[000:006] [1] (audio_mixer_manager_alsa_linux.cc:196): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:006] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM 
[000:006] [1] (audio_device_alsa_linux.cc:961): unable to open playback device: No such file or directory (-2)
[000:006] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL 
[000:007] [1] (audio_mixer_manager_alsa_linux.cc:286): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:007] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM 
[000:007] [1] (audio_device_alsa_linux.cc:1108): unable to open record device: No such file or directory
[000:007] [1] (audio_device_generic.cc:38): BuiltInAECIsAvailable: Not supported on this platform
[000:007] [1] (audio_device_generic.cc:48): BuiltInAGCIsAvailable: Not supported on this platform
[000:007] [1] (audio_device_generic.cc:58): BuiltInNSIsAvailable: Not supported on this platform
[000:007] [1] (adm_helpers.cc:47): Unable to set recording channel to kChannelBoth.
[000:008] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL 
[000:008] [1] (audio_mixer_manager_alsa_linux.cc:286): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:008] [1] (adm_helpers.cc:56): Unable to access microphone.
[000:008] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL 
[000:008] [1] (audio_mixer_manager_alsa_linux.cc:286): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:009] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM 
[000:009] [1] (audio_device_alsa_linux.cc:1108): unable to open record device: No such file or directory
[000:009] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL 
[000:009] [1] (audio_mixer_manager_alsa_linux.cc:196): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:009] [1] (adm_helpers.cc:100): Unable to access speaker.
[000:010] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL 
[000:010] [1] (audio_mixer_manager_alsa_linux.cc:196): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:010] [1] (audio_device_alsa_linux.cc:1654): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2501:(snd_pcm_open_noupdate) Unknown PCM 
[000:010] [1] (audio_device_alsa_linux.cc:961): unable to open playback device: No such file or directory (-2)
HTTP Listen at 0.0.0.0:8000
answer:"v0.0.5-19-gc776a8a webrtc@47791cf-dirty live555helper@5d3072a civetweb@v1.9.1-177-gd9b713b h264bitstream@0.2.0"

answer:[
   {
      "video" : "rtsp://10.1.1.208:8554/ring0"
   }
]

answer:{
   "iceServers" : [
      {
         "url" : "stun:stun.l.google.com:19302"
      }
   ]
}

body:{"type":"offer","sdp":"v=0\r\no=- 3745170295405834109 2 IN IP4 127.0.0.1\r\ns=-\r\nt=0 0\r\na=group:BUNDLE audio video\r\na=msid-semantic: WMS\r\nm=audio 9 UDP/TLS/RTP/SAVPF 111 103 104 9 0 8 106 105 13 126\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:y+9zqAPdsUYGAD86\r\na=ice-pwd:ZjvTITpydbPff/QcPa/nXue/\r\na=fingerprint:sha-256 AF:C6:EC:59:73:67:1A:94:F1:68:69:71:BF:E5:75:FC:2B:61:3D:5A:06:31:DF:71:EE:9A:98:D9:9D:6F:60:AA\r\na=setup:actpass\r\na=mid:audio\r\na=extmap:1 urn:ietf:params:rtp-hdrext:ssrc-audio-level\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:111 opus/48000/2\r\na=fmtp:111 minptime=10; useinbandfec=1\r\na=rtpmap:103 ISAC/16000\r\na=rtpmap:104 ISAC/32000\r\na=rtpmap:9 G722/8000\r\na=rtpmap:0 PCMU/8000\r\na=rtpmap:8 PCMA/8000\r\na=rtpmap:106 CN/32000\r\na=rtpmap:105 CN/16000\r\na=rtpmap:13 CN/8000\r\na=rtpmap:126 telephone-event/8000\r\na=maxptime:60\r\nm=video 9 UDP/TLS/RTP/SAVPF 100 101 116 117 96\r\nc=IN IP4 0.0.0.0\r\na=rtcp:9 IN IP4 0.0.0.0\r\na=ice-ufrag:y+9zqAPdsUYGAD86\r\na=ice-pwd:ZjvTITpydbPff/QcPa/nXue/\r\na=fingerprint:sha-256 AF:C6:EC:59:73:67:1A:94:F1:68:69:71:BF:E5:75:FC:2B:61:3D:5A:06:31:DF:71:EE:9A:98:D9:9D:6F:60:AA\r\na=setup:actpass\r\na=mid:video\r\na=extmap:2 urn:ietf:params:rtp-hdrext:toffset\r\na=extmap:3 http://www.webrtc.org/experiments/rtp-hdrext/abs-send-time\r\na=extmap:4 urn:3gpp:video-orientation\r\na=recvonly\r\na=rtcp-mux\r\na=rtpmap:100 VP8/90000\r\na=rtcp-fb:100 ccm fir\r\na=rtcp-fb:100 nack\r\na=rtcp-fb:100 nack pli\r\na=rtcp-fb:100 goog-remb\r\na=rtpmap:101 VP9/90000\r\na=rtcp-fb:101 ccm fir\r\na=rtcp-fb:101 nack\r\na=rtcp-fb:101 nack pli\r\na=rtcp-fb:101 goog-remb\r\na=rtpmap:116 red/90000\r\na=rtpmap:117 ulpfec/90000\r\na=rtpmap:96 rtx/90000\r\na=fmtp:96 apt=100\r\n"}
body:{"candidate":"candidate:4167255511 1 udp 2122260223 10.1.1.100 38380 typ host generation 0","sdpMid":"audio","sdpMLineIndex":0}
body:{"candidate":"candidate:4167255511 2 udp 2122260222 10.1.1.100 35098 typ host generation 0","sdpMid":"audio","sdpMLineIndex":0}
body:{"candidate":"candidate:4167255511 1 udp 2122260223 10.1.1.100 33307 typ host generation 0","sdpMid":"video","sdpMLineIndex":1}
answer:true

[... 780 lines cut ...]
[h264 @ 0x7f4a44010be0] non-existing PPS 0 referenced
[h264 @ 0x7f4a44010be0] decode_slice_header error
[h264 @ 0x7f4a44010be0] no frame!
[019:969] [65] (h264_decoder_impl.cc:331): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[h264 @ 0x7f4a44010be0] no frame!
[026:360] [65] (h264_decoder_impl.cc:331): avcodec_decode_video2 error: -1094995529
NOTIFY failed
[030:238] [1] (PeerConnectionManager.cpp:494): Close PeerConnection
[030:238] [1] (PeerConnectionManager.cpp:508): Close PeerConnection no more used rtsp://10.1.1.208:8554/ring0
Close session: video/H264
=> RemoveSink
[030:250] [1] (webrtcvoiceengine.cc:701): webrtc: StopSend() failed to stop recording (error=10030)
[030:250] [1] (audio_send_stream.cc:255): AudioSendStream::Stop failed with error: -1
[030:264] [1] (webrtcvoiceengine.cc:701): webrtc: StopSend() failed to stop recording (error=10030)
[030:264] [1] (audio_send_stream.cc:255): AudioSendStream::Stop failed with error: -1
[030:264] [1] (webrtcvoiceengine.cc:701): webrtc: StopSend() failed to stop recording (error=10030)
[030:264] [1] (audio_send_stream.cc:255): AudioSendStream::Stop failed with error: -1
[030:264] [1] (webrtcvoiceengine.cc:701): webrtc: StopSend() failed to stop recording (error=10030)
[030:264] [1] (audio_send_stream.cc:255): AudioSendStream::Stop failed with error: -1
[030:265] [1] (webrtcvoiceengine.cc:701): webrtc: StopSend() failed to stop recording (error=10030)

````

So the http://10.1.1.208:8000 page shows the grey square

mpromonet, 2017-09-27:
Hi sfinx
It seems the h264 décoder fails decoding the stream, increasing verbosity should help to understand why (loosing idr frame, mistake decoding or processing sps/pps ....)
Best regards
Michel

Sfinx, 2017-09-27:
Hi,

You can download the -vvv log from:

https://mega.nz/#!GoF32RoJ!3GNTYdOqsvYj4AJUcREwalirwFA0Fcvvj5rYWx3IFlA

Sfinx, 2017-09-27:
Please note that ffplay/mplayer/mpv and gstreamer plays this URL ok

mpromonet, 2017-10-08:
Hi sfinx,
From the log you posted, decoding works after an IDR frame is received:

```
[023:338] [65] (rtspvideocapturer.cpp:101): RTSPVideoCapturer:onData size:27 ts:1506513947024^M
[023:339] [65] (rtspvideocapturer.cpp:109): NAL:7^M
[023:339] [65] (rtspvideocapturer.cpp:111): ===========================SPS^M
[023:339] [65] (rtspvideocapturer.cpp:118): =========set timing_info_present_flag:1 30 1^M
[023:339] [65] (rtspvideocapturer.cpp:101): RTSPVideoCapturer:onData size:8 ts:1506513947024^M
[023:339] [65] (rtspvideocapturer.cpp:109): NAL:8^M
[023:339] [65] (rtspvideocapturer.cpp:139): ===========================PPS^M
[023:339] [65] (rtspvideocapturer.cpp:101): RTSPVideoCapturer:onData size:20445 ts:1506513947024^M
[023:339] [65] (rtspvideocapturer.cpp:109): NAL:5^M
[023:339] [65] (rtspvideocapturer.cpp:144): ===========================IDR^M
[023:345] [65] (rtspvideocapturer.cpp:206): RTSPVideoCapturer::Decoded 307200 2628344448 2628344448 0 2628344^M
[023:345] [67] (video_stream_encoder.cc:765): Video frame parameters changed: dimensions=640x480, texture=0.^M
[023:345] [67] (video_codec_initializer.cc:134): No temporal layers and nack enabled -> resilience off^M
```
Then it seems working ?
Could you add in the issue what command line do you used to generate the RTSP service using gstreamer ?
Best Regards,
Michel.

Sfinx, 2017-10-09:
Hello,

No it is not working (see below). To check :

- install libgstrtspserver-1.0-dev libgstrtspserver-1.0
- clone the https://github.com/GStreamer/gst-rtsp-server (no need to build it)
- compile examples/test-mp4.c with the stock system includes and libraries
- launch it with some MP4 file (./test-mp4 some-video.mp4)
- check with ffplay that RTSP link is working **(from another box/IP)**

Next launch the docker image with this link. **But try access the HTTP URL from the another box/IP**.

I've found that accessing webrtc-streamer HTTP URL at the same box is working but do not work when accessing it from the another IP even in the same network

Sfinx, 2017-10-09:
I guess the root problem here - old/buggy webrtc library. But as was already reported - it has the linkage errors when building using stock webrtc-streamer build instructions.

Sfinx, 2017-10-09:
So it is not clear how the WebRTC ICE working with this docker image. The explicit port mapping has the single port (HTTP 8000) but there must exists other ports for WebRTC media streams delivery (?!)

mpromonet, 2017-10-09:
Hi sfinx,
Thanks for your test report.
Build use the latest webrtc, then if it is a bug, this is not an old/buggy webrtc.
As I told you, in your log we see H264 stream is decoded, so this is consistent with an ICE problem.
By default webrtc-streamer use google STUN server, but you can also use an embeded STUN, and also a external TURN server.
Regards,
Michel.

Sfinx, 2017-10-09:
Hi Michel,

As was already said - all the tests was made in local (the same ethernet switch) network so STUN is not the issue here. Have you tested your software with Chrome on any Android based phone ?

mpromonet, 2017-10-09:
Hi Sfinx,
Yes, I tested with some android phone with chrome, some are working others display a fix image. 
I know there is some fix to do with ICE negociation, maybe others things.
Regards,
Michel.

closed 2018-09-27 by mpromonet