mpromonet/webrtc-streamer #63: Not working on Safari 11

AdoPi, 2017-10-04:
Hi, thanks for your big work on this streamer!
The website does not work on Safari 11, maybe using adapter.js can solve the problem.
[https://github.com/webrtc/adapter](url)

GregBakker, 2017-10-04:
adapter may be part of the solution - but it's not 100% for Safari. Also had to use
```
	if (isSafari) {
            this.pc.addTransceiver('video')
	    this.pc.addTransceiver('audio')
	}
```
on the PeerConnection. Even so, this doesn't immediately work on IOS safari - might have to do with the safety of the origin of the element; haven't tried this via https yet.

GregBakker, 2017-10-04:
...but definitely a useful codebase, thx!

mpromonet, 2017-10-08:
Hi,
Maybe I will look what is different in safari webrtc interface, however probably the html/js in the project will stay as simple as it could. 
I would like to avoid dependencies with some extra layer like adapter.js, maybe I will change my mind...
You could develop an interface using adapter.js and webrtc-streamer API.
Best Regards,
Michel.

AdoPi, 2018-03-02:
I see that the adapter lib has been included on the project so I close this issue.
Thanks for your works :)

closed 2018-03-02 by AdoPi