mpromonet/webrtc-streamer #66: SSL is possible?

roberto497, 2017-10-11:
I need to use ssl, it's possible?

mpromonet, 2017-10-11:
Hi roberto497,
civetweb support it, you need to give -H8000,8001s to listen http on port 8000 and https on 8001.
It needs probably a little modification in order to give key/certificate to civetweb.
Best Regards,
Michel.

roberto497, 2017-10-11:
Not working for me.
I've changed the civetweb.conf file, includes the lines:
listening_ports 80, 443s
ssl_certificate ./ssl_cert.pem

I started webrtc-streamer with -H 8000,8001s, but it only works on http

mpromonet, 2017-10-14:
Hi roberto497,
There is no configuration file, your civetweb.conf is not read.
I added an option -c to give the ssl material.
Best Regards,
Michel.

commit 26720e9f (2017-10-14): fix #66: allow to set ssl-certificate option of civetweb
closed 2017-10-14 by mpromonet