mpromonet/webrtc-streamer #68: Build binary

alecapu80, 2017-10-15:
Hi,
I've seen that you use Travis for test build with 3 different build. 
Is possibile to access to the build binary without rebuild it?

mpromonet, 2017-10-15:
Hi alecapu80,
I don't know how to access to travis build, but the travis job push binaries to github releases.
You can get it from https://github.com/mpromonet/webrtc-streamer/releases
Best Regards,
Michel.

alecapu80, 2017-10-15:
WOW, thank you very much. It works very well with ONE RTSP stream on Raspberry Pi 3 B.
Unfortunately with more contemporary stream it crash, probably for the limited resources of the Raspberry. Very very nice job!! It's very useful! 
Tnks

closed 2017-10-28 by mpromonet

xunxin03, 2017-11-01:
How can I use the Build binary file? Where should I put it in? Which system is ok? ubuntu?

mpromonet, 2017-11-01:
Hi xunxin03,
You can extract file from tar.gz and run webrtc-streamer.
The build is for ubuntu trusty, it should also works on next release. I test it on xenial and it run well.
Best Regards,
Michel.