mpromonet/webrtc-streamer #71: What kind of commits by webrtc do you recommend for building version v0.0.6?

ghostzero2000, 2017-10-26:
Hello!!
Your project is very good. I used ready packages for armv7 in raspberry pi 2, it started well. I tried to build it myself. I builted it. The package what I builted did not start.
I think need use another commit by webrtc. Current commit by webrtc what i used is 0295a967c08b06f25b57334c815ae3e71aa3050b.

mpromonet, 2017-10-27:
Hi ghostzero2000,
You can know whta webrtc commit was used in a release running webrtc-streamer with `-V` arguments.
It should print something like webrtc@<commit>.
If what you built deosnot start, maybe  some library is missing, didi you tried to run `ldd ./webrtc-streamer` ?
Best Regards,
Michel.

ghostzero2000, 2017-10-28:
Ok. Thank you.  I'll switch to commit 8eb9c7d8388b449fb8c19b2e06615e4c5b304005 by webrtc project and will try to build it. When I finish, I'll write it off. Have a nice day.

mpromonet, 2017-10-29:
Hi ghostzero2000,
In order to workaround a not nice management of WebRTC initialization of sound capture, I add an option `-a` to disable audio in commit a363f42b31f0990d66a64d847e1308d827f1f704
Maybe this is what prevent you to start the application you built ?
Best Regards,
Michel.

ghostzero2000, 2017-10-30:
I am very grateful to you. Everything works perfectly on the commit 8eb9c7d8388b449fb8c19b2e06615e4c5b304005. Yes, I used ldd, but the error was "undefined reference" classes and methods by source webrtc. When I run without the -a option, it displays errors initializing the audio, but it fulfills the video transfer. Thank you again and have a nice day.

closed 2017-11-06 by mpromonet