mpromonet/webrtc-streamer #73: Crash when playing rtsp stream

9a4gl, 2017-10-27:
Camera has two streams, with low res stream no issue, with high res stream it won't start and webrtc-streamer crash, communication log and stack trace follows

Received a complete DESCRIBE response:
RTSP/1.0 200 OK
Cseq: 2 
Server: Rtsp Server 1920*1080*30*4096
Content-Type: application/sdp
Content-length: 518
Content-Base: rtsp://44.170.133.52/0

v=0
o=StreamingServer 3331435948 1116907222000 IN IP4 44.170.133.52
s=h264.mp4
c=IN IP4 0.0.0.0
t=0 0
a=control:*
m=video 0 RTP/AVP 96
a=control:trackID=0
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1;sprop-parameter-sets=Z2QAKK2EBUViuKxUdCAqKxXFYqOhAVFYrisVHQgKisVxWKjoQFRWK4rFR0ICorFcVio6ECSFITk8nyfk/k/J8nm5s00IEkKQnJ5Pk/J/J+T5PNzZprQDwBE/LKkAAAMAeAAAHCBgQAAW42AABm/yve+F4RCNQA==,aO48sA==
a=videoinfo:1920*1080*5*4096
m=audio 0 RTP/AVP 0
a=control:trackID=1
a=rtpmap:0 PCMU/8000
a=ptime:20

Sending request: SETUP rtsp://44.170.133.52/0/trackID=0 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2017.09.12
Transport: RTP/AVP;unicast;client_port=36468-36469

Received 676 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
Cseq: 2 
Server: Rtsp Server 1920*1080*30*4096
Content-Type: application/sdp
Content-length: 518
Content-Base: rtsp://44.170.133.52/0

v=0
o=StreamingServer 3331435948 1116907222000 IN IP4 44.170.133.52
s=h264.mp4
c=IN IP4 0.0.0.0
t=0 0
a=control:*
m=video 0 RTP/AVP 96
a=control:trackID=0
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1;sprop-parameter-sets=Z2QAKK2EBUViuKxUdCAqKxXFYqOhAVFYrisVHQgKisVxWKjoQFRWK4rFR0ICorFcVio6ECSFITk8nyfk/k/J8nm5s00IEkKQnJ5Pk/J/J+T5PNzZprQDwBE/LKkAAAMAeAAAHCBgQAAW42AABm/yve+F4RCNQA==,aO48sA==
a=videoinfo:1920*1080*5*4096
m=audio 0 RTP/AVP 0
a=control:trackID=1
a=rtpmap:0 PCMU/8000
a=ptime:20

Sending request: SETUP rtsp://44.170.133.52/0/trackID=0 RTSP/1.0
CSeq: 3
User-Agent: LIVE555 Streaming Media v2017.09.12
Transport: RTP/AVP;unicast;client_port=32906-32907

Received 176 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 3
Server: Rtsp Server 
Session: 63856115
Transport: RTP/AVP;unicast;client_port=36468-36469;source=44.170.133.52;server_port=32769-0;ssrc=00004E87

Start playing sink for "video/H264" subsession
markerSize:4
Sending request: SETUP rtsp://44.170.133.52/0/trackID=1 RTSP/1.0
CSeq: 4
User-Agent: LIVE555 Streaming Media v2017.09.12
Transport: RTP/AVP;unicast;client_port=57260-57261
Session: 63856115

Received 176 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 3
Server: Rtsp Server 
Session: 98481030
Transport: RTP/AVP;unicast;client_port=32906-32907;source=44.170.133.52;server_port=32769-0;ssrc=00004E87

=> onNewSession video/H264 sdp:m=video 0 RTP/AVP 96
a=control:trackID=0
a=rtpmap:96 H264/90000
a=fmtp:96 packetization-mode=1;sprop-parameter-sets=Z2QAKK2EBUViuKxUdCAqKxXFYqOhAVFYrisVHQgKisVxWKjoQFRWK4rFR0ICorFcVio6ECSFITk8nyfk/k/J8nm5s00IEkKQnJ5Pk/J/J+T5PNzZprQDwBE/LKkAAAMAeAAAHCBgQAAW42AABm/yve+F4RCNQA==,aO48sA==
a=videoinfo:1920*1080*5*4096

Received 176 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 4
Server: Rtsp Server 
Session: 63856115
Transport: RTP/AVP;unicast;client_port=57260-57261;source=44.170.133.52;server_port=32771-0;ssrc=00004E87

Sending request: PLAY rtsp://44.170.133.52/0 RTSP/1.0
CSeq: 5
User-Agent: LIVE555 Streaming Media v2017.09.12
Session: 63856115
Range: npt=0.000-

Received 81 new bytes of response data.
Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 5
Server: Rtsp Server 
Session: 63856115;timeout=120

terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoi
[017:653] [10237] (webrtcvoiceengine.cc:700): webrtc: StopSend() failed to stop recording (error=10030)
[017:653] [10237] (audio_send_stream.cc:220): AudioSendStream::Stop failed with error: -1
[New Thread 0x7ffff42dd700 (LWP 10314)]

Thread 73 "Thread 0x0x7fff" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff777fe700 (LWP 10312)]
0x00007ffff6878428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff6878428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff687a02a in __GI_abort () at abort.c:89
#2  0x00007ffff73d884d in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff73d66b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff73d6701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff73d6919 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff73ff20f in std::__throw_invalid_argument(char const*) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x000000000041c7f7 in __gnu_cxx::__stoa<long, int, char, int> (__convf=0x40bd20 <strtol@plt>, __name=0xec4421 "stoi", __str=0x7ffff00169d8 "k", __idx=0x0)
    at /usr/include/c++/5/ext/string_conversions.h:65
#8  0x00000000004181e1 in std::stoi (__str="k", __idx=0x0, __base=10) at /usr/include/c++/5/bits/basic_string.h:5260
#9  0x000000000041b620 in RTSPAudioSource::onNewSession (this=0x7fffcc016190, id=0x7ffff0016a50 "liveMedia4", media=0x7ffff0003230 "video", codec=0x7ffff0003400 "H264", 
    sdp=0x7ffff0003090 "m=video 0 RTP/AVP 96\r\na=control:trackID=0\r\na=rtpmap:96 H264/90000\r\na=fmtp:96 packetization-mode=1;sprop-parameter-sets=Z2QAKK2EBUViuKxUdCAqKxXFYqOhAVFYrisVHQgKisVxWKjoQFRWK4rFR0ICorFcVio6ECSFITk8nyfk/"...) at inc/rtspvideocapturer.h:117
#10 0x0000000000441632 in RTSPConnection::RTSPClientConnection::continueAfterSETUP (this=0x7ffff00039c0, resultCode=0, resultString=0x0) at src/rtspconnectionclient.cpp:244
#11 0x0000000000441cbd in RTSPConnection::RTSPClientConnection::continueAfterSETUP (rtspClient=0x7ffff00039c0, resultCode=0, resultString=0x0) at inc/rtspconnectionclient.h:111
#12 0x0000000000e950e1 in RTSPClient::handleResponseBytes (this=this@entry=0x7ffff00039c0, newBytesRead=<optimised out>) at RTSPClient.cpp:1880
#13 0x0000000000e95339 in RTSPClient::incomingDataHandler1 (this=0x7ffff00039c0) at RTSPClient.cpp:1565
#14 0x0000000000e9535d in RTSPClient::incomingDataHandler (instance=<optimised out>) at RTSPClient.cpp:1558
#15 0x0000000000ec2e5d in BasicTaskScheduler::SingleStep (this=0x7fffcc00a170, maxDelayTime=<optimised out>) at BasicTaskScheduler.cpp:153
#16 0x0000000000ec3dd7 in BasicTaskScheduler0::doEventLoop (this=0x7fffcc00a170, watchVariable=0x7fffcc016750 "") at BasicTaskScheduler0.cpp:80
#17 0x00000000004426e3 in Environment::mainloop (this=0x7fffcc016338) at src/environment.cpp:30
#18 0x000000000041b3ae in RTSPAudioSource::Run (this=0x7fffcc016190) at inc/rtspvideocapturer.h:95
#19 0x000000000046e814 in rtc::Thread::PreRun(void*) ()
#20 0x00007ffff6c146ba in start_thread (arg=0x7fff777fe700) at pthread_create.c:333
#21 0x00007ffff694a3dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

mpromonet, 2017-10-28:
Hi 9a4gl,
The parsing of SDP to get freq/channel for audio is too weak.
Best Regards,
Michel.

closed 2017-10-28 by mpromonet

9a4gl, 2017-10-30:
Problem fixed. Thanks a lot.

commit cf629dc0 (2017-10-28): fix #73: improve SDP parsing to extract freq ans channel