mpromonet/webrtc-streamer #75: Can I use you binary release webrtc-streamer_v0.0.6_x86_64.tgz?

xunxin03, 2017-11-01:
I am a newer in this field, I would like to ask how can I use your release? Where should I put them in? Then how I can to see the rtsp source from chrome???

xunxin03, 2017-11-01:
When I run docker, there is many errors:
root@bigdata3:~# docker run -p 8000:8000 -it mpromonet/webrtc-streamer rtsp://85.255.175.241/h264
Logger level:4
[000:000] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:008] [1] (audio_mixer_manager_alsa_linux.cc:196): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:008] [1] (voe_base_impl.cc:200): Init() failed to initialize the speaker
[000:009] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:009] [1] (audio_mixer_manager_alsa_linux.cc:286): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:009] [1] (voe_base_impl.cc:209): Init() failed to initialize the microphone
[000:009] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:009] [1] (audio_mixer_manager_alsa_linux.cc:196): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:009] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:009] [1] (audio_device_alsa_linux.cc:953): unable to open playback device: No such file or directory (-2)
[000:010] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:010] [1] (audio_mixer_manager_alsa_linux.cc:286): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:010] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:010] [1] (audio_device_alsa_linux.cc:1098): unable to open record device: No such file or directory
[000:011] [1] (audio_device_generic.cc:38): BuiltInAECIsAvailable: Not supported on this platform
[000:011] [1] (audio_device_generic.cc:48): BuiltInAGCIsAvailable: Not supported on this platform
[000:011] [1] (audio_device_generic.cc:58): BuiltInNSIsAvailable: Not supported on this platform
[000:011] [1] (adm_helpers.cc:47): Unable to set recording channel to kChannelBoth.
[000:012] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:012] [1] (audio_mixer_manager_alsa_linux.cc:286): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:012] [1] (adm_helpers.cc:56): Unable to access microphone.
[000:012] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:012] [1] (audio_mixer_manager_alsa_linux.cc:286): snd_mixer_attach(_inputMixerHandle, ) error: No such file or directory
[000:012] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:012] [1] (audio_device_alsa_linux.cc:1098): unable to open record device: No such file or directory
[000:013] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:013] [1] (audio_mixer_manager_alsa_linux.cc:196): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:013] [1] (adm_helpers.cc:100): Unable to access speaker.
[000:013] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib control.c:954:(snd_ctl_open_noupdate) Invalid CTL
[000:013] [1] (audio_mixer_manager_alsa_linux.cc:196): snd_mixer_attach(_outputMixerHandle, ) error: No such file or directory
[000:013] [1] (audio_device_alsa_linux.cc:1596): GetDevicesInfo - Could not find device name or numbers
ALSA lib pcm.c:2266:(snd_pcm_open_noupdate) Unknown PCM
[000:013] [1] (audio_device_alsa_linux.cc:953): unable to open playback device: No such file or directory (-2)
HTTP Listen at 0.0.0.0:8000

mpromonet, 2017-11-01:
Hi xunxin03,
With the docker you should be able to access to the application using http://127.0.0.1:8000 (the error about sound could be avoid disabling the sound capture initialization adding -a, anyway this doesnot prevent the application to run)
It is the same extracting tar.gz file and running ./webrtc-streamer.
In both case you can give some RTSP url from the command line or from the browser side  for instance like https://webrtc-streamer.herokuapp.com/webrtcstream.html?video=rtsp://217.17.220.110/axis-media/media.amp or building your own html page.
Best Regards,
Michel.

xunxin03, 2017-11-02:
I am so sorry, I want to trouble you again, I can not see the video from my Chrome,  is there any question about my explorer? I used this link rtsp://85.255.175.241/h264, I can see the video from the VLC app. But I can not see it from mt Chrome. Also, when I started the docker container, I can see the above logs: Is that mean the server is started success, I can not see the video because there are some questions in my Explorer???

Received 233 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 3
Cache-Control: must-revalidate
Date: Thu, Nov 02 2017 05:53:26 GMT
Transport: RTP/AVP;unicast;destination=1.85.218.153;source=85.255.175.241;client_port=51510-51511;server_port=6970-6971
Session: 4479

Start playing sink for "video/H264" subsession
markerSize:4
Sending request: SETUP rtsp://85.255.175.241/h264/track2 RTSP/1.0
CSeq: 4
User-Agent: LIVE555 Streaming Media v2017.09.12
Transport: RTP/AVP;unicast;client_port=7878-7879
Session: 4479

Received 231 new bytes of response data.
Received a complete SETUP response:
RTSP/1.0 200 OK
CSeq: 4
Cache-Control: must-revalidate
Date: Thu, Nov 02 2017 05:53:26 GMT
Transport: RTP/AVP;unicast;destination=1.85.218.153;source=85.255.175.241;client_port=7878-7879;server_port=6972-6973
Session: 4479

Sending request: PLAY rtsp://85.255.175.241/h264/ RTSP/1.0
CSeq: 5
User-Agent: LIVE555 Streaming Media v2017.09.12
Session: 4479
Range: npt=0.000-

Received 242 new bytes of response data.
Received a complete PLAY response:
RTSP/1.0 200 OK
CSeq: 5
Date: Thu, Nov 02 2017 05:53:27 GMT
Range: npt=0.000-
Session: 4479
RTP-Info: url=rtsp://85.255.175.241/h264/track1;seq=24975;rtptime=2433065478,url=rtsp://85.255.175.241/h264/track2;seq=2539;rtptime=904105342

Close session: video/H264
Created new TCP socket 25 for connection
Connecting to 85.255.175.241, port 554 on socket 25...
...remote connection opened
Sending request: DESCRIBE rtsp://85.255.175.241/h264 RTSP/1.0
CSeq: 2
User-Agent: LIVE555 Streaming Media v2017.09.12
Accept: application/sdp

Received 789 new bytes of response data.
Received a complete DESCRIBE response:
RTSP/1.0 200 OK
CSeq: 2
Date: Thu, Nov 02 2017 05:53:38 GMT
Content-Base: rtsp://85.255.175.241/h264/
Content-Type: application/sdp
Content-Length: 629

v=0
o=- 1508333589644487 1 IN IP4 0.0.0.0
s=Session streamed by "nessyMediaServer"
i=h264
t=0 0
a=tool:LIVE555 Streaming Media v2010.04.09_dyna_modi_2010.05.05
a=type:broadcast
a=control:*
a=range:npt=0-
[... 140 lines cut ...]
Date: Thu, Nov 02 2017 05:53:50 GMT
Range: npt=0.000-
Session: 4481
RTP-Info: url=rtsp://85.255.175.241/h264/track1;seq=26177;rtptime=2435182261,url=rtsp://85.255.175.241/h264/track2;seq=2727;rtptime=904293338

[1896:009] [1] (PeerConnectionManager.cpp:542): Close PeerConnection
[1896:010] [1] (PeerConnectionManager.cpp:556): Close PeerConnection no more used rtsp://85.255.175.241/h264
Close session: video/H264
=> RemoveSink
[1896:013] [1] (voe_base_impl.cc:450): StopSend() failed to stop recording
[1896:013] [1] (audio_send_stream.cc:256): AudioSendStream::Stop failed with error: -1
Close session: audio/PCMU
[1896:023] [1] (voe_base_impl.cc:450): StopSend() failed to stop recording
[1896:023] [1] (audio_send_stream.cc:256): AudioSendStream::Stop failed with error: -1
[1896:023] [1] (voe_base_impl.cc:450): StopSend() failed to stop recording
[1896:023] [1] (audio_send_stream.cc:256): AudioSendStream::Stop failed with error: -1
[1896:023] [1] (voe_base_impl.cc:450): StopSend() failed to stop recording
[1896:023] [1] (audio_send_stream.cc:256): AudioSendStream::Stop failed with error: -1
[1896:023] [1] (voe_base_impl.cc:450): StopSend() failed to stop recording
[1896:023] [1] (audio_send_stream.cc:256): AudioSendStream::Stop failed with error: -1
[1896:024] [1] (voe_base_impl.cc:450): StopSend() failed to stop recording

Thank you  very much!

[image]

xunxin03, 2017-11-02:
When I used the Explorer to show the video, I see the html5 video tag, In my Explorer, it does not work. Is this ok in your Chrome? I am puzzled in the webrtc-streamer. Does it just to capture the video in many ways like rtsp,  h264 and so on, then when we have captured the video, how can we to do next? Save the video? Or show the video with our server to customer from the Explorer Client? In the demos, you have applied some html demos, but from the demos, I can not see the videos.

xunxin03, 2017-11-02:
When I used your binary release there some errors, my system is  Ubuntu 14.04 LTS, is this system not supported the webrtc-streamer server? Or I should make other configures, such as I also need to setup the webcrt in my system. Are these dependencies not packaged in the binary release???

mpromonet, 2017-11-02:
Hi xunxin03,
If the RTSP negociation works, it is usualy a problem negociating ICE, you should look in the console log of the browser and in the about:webrtc page (the url depends on the browser you're using).
Instead of using google STUN server you can try using the embeded one, using the '-S' option.
Best Regards,
Michel.

xunxin03, 2017-11-03:
When I use your binary release in my Ubuntu 14.04 LTS, then I start the ./webrtc-streamer -S with a valid rtsp url, there are some errors like this:
root@bigdata3:/opt/soft/tomcat/apache-tomcat-8.0.41/webapps/webrtc-streamer# ./webrtc-streamer -S rtsp://85.255.175.241/h264
Logger level:4
[000:000] [13261] (audio_device_generic.cc:38): BuiltInAECIsAvailable: Not supported on this platform
[000:000] [13261] (audio_device_generic.cc:48): BuiltInAGCIsAvailable: Not supported on this platform
[000:000] [13261] (audio_device_generic.cc:58): BuiltInNSIsAvailable: Not supported on this platform
[000:000] [13261] (adm_helpers.cc:47): Unable to set recording channel to kChannelBoth.
ALSA lib conf.c:3942:(snd_config_update_r) Cannot access file /home/travis/build/mpromonet/webrtc-streamer/webrtc/src/build/linux/debian_jessie_amd64-sysroot/usr/share/alsa/alsa.conf
ALSA lib conf.c:3942:(snd_config_update_r) Cannot access file /home/travis/build/mpromonet/webrtc-streamer/webrtc/src/build/linux/debian_jessie_amd64-sysroot/usr/share/alsa/alsa.conf
HTTP Listen at 0.0.0.0:8000
STUN Listening at 0.0.0.0:3478

And In the chrome console, I can see some errors like this:
ajax.js:5 HTTP call http://192.168.15.203:8000/call?peerid=0.4337220997038864&url=rtsp%3A%2F%2F85.255.175.241%2Fh264
ajax.js:5 HTTP call http://192.168.15.203:8000/addIceCandidate?peerid=0.4337220997038864
ajax.js:42 POST http://192.168.15.203:8000/addIceCandidate?peerid=0.4337220997038864 404 (Not Found)
send @ ajax.js:42
WebRtcStreamer.onIceCandidate @ webrtcstreamer.js:156
pc.onicecandidate @ webrtcstreamer.js:103

closed 2018-01-14 by mpromonet