mpromonet/webrtc-streamer #76: IOS Chrome Error creating RTCPeerConnection

choychris, 2017-11-08:
Hi,
Thank you for making such a great project.
I try to open the demo on latest version of IOS Chrome application.
However, i get the error:

[image].

Also, I am actually very new to webrtc,  I would like to know if I want to use webrtc api in mobile application, how can I get the encoded video from webrtc-streamer?

Thanks in advance!

mpromonet, 2017-11-08:
Hi Chris,
I have no IOS Chrome, did sample WebRTC app like https://appr.tc/ works ?
If it works, you should look to the https://github.com/mpromonet/webrtc-streamer/blob/master/html/webrtcstreamer.js to fix creation of RTCPeerConnection, maybe some options are not supported. 
A possible way could be to use  https://github.com/webrtc/adapter. Till now I avoid to use to much layer, but it could helps to create a more robust html page
Regards,
Michel.

choychris, 2017-11-09:
Hi Michel,

Thank you very much for the reply.
After googling while, I find that webrtc is not supported in WKWebview nor UIWebview in IOS.
Chrome in IOS is built with WKWebview. 
https://stackoverflow.com/questions/45055329/does-webkit-in-ios-11-beta-support-webrtc

In fact, I am looking for a solution to stream from an Ip camera with very low latency in mobile application.
After digging for a few days,  I find webrtc and this project can be the answer.
Sorry for bothering, but I still have some questions in mind.

1. To my understanding, this project works as a server transcode the rtsp stream to V8 codec and  http server helps client to access the transcoded stream through webrtc. 
If I want to get the corresponding rtsp stream, i need to createOffer in client side and send `this.srvurl + "/call?peerid="+ peerid+"&url="+encodeURIComponent(videourl)` to my http server right?
Does the http server act as a signalling server as well ?

2. If I want to add another rtsp uri, I need to go into my server and exec ./webrtc-streamer rtsp://217.17.220.110/axis-media/media.amp ?

3. I am planning to try it remotely on AWS. Approximately, how much does CPU and memory consume if there are 4 rtsp concurrent streams?

I really appreciate any answer and help. Thanks again!

Regards,
Chris

mpromonet, 2017-11-12:
Hi Chris,
This is clearly an experimental project, and there is still work to improve it, and maybe more.
To try to answer to your question:

1. In fact in uncompress RTSP/RTP stream (H264 & JPEG) to I420 and give it to WebRTC, next decpending on negociation between Browser & WebRTC server it comrpess it in VP8, VP9 or H264. The /call API give send the WebRTC Offer and give the WebRTC Answer, it also need to call /addIceCandidate when WebRTC onIceCandidate is trigged and call /getIceCandidate to get the server candidate. By now the HTML/JS sample call /getIceCandidate after receiving the last candidate, I am not sure this is sufficiant nor compliant with the WebRTC standard. 

2. This is not needing you can use /call with an RTSP url that is not known from the server, the url given in the command line are just a way to give it to the HTML, maybe I will chance this. It could be given in some config.js as well, the server do not need to know the RTSP url, it is different for capture interface.

3. I have no idea, this clearly depends on streams bandwidth, H264 profiles, and GPU capacity (probably AWS doesnot have and will compress/uncompress using CPU). The demo deployed in heroku seems to works with 4 RTSP streams, but I did not spend time to monitor it.

Best Regards,
Michel.

choychris, 2017-11-13:
Hi Michel,

Thank you so much.
This is an awesome project!
I will keep experimenting this project and try to improve it if I can.

Cheers,
Chris

choychris, 2017-11-15:
Hi @mpromonet 

I am sorry for bothering you again. I put a lot of time these days to try to run this project in my machine. But I still cannot find out the reason of not playing out my rtsp.

I start with Docker, it seems good without any errors.
Then, I request for my rtsp url.
The server gives me these logging. Seems nothing wrong too.
(https://gist.github.com/choychris/e07be4ec82e28752b38a48386f39d330)
Also, looking at chrome loggings, peer connection state is completed, so I suppose peer connection is correct as well.

However, I cannot see the video showing out in my browser
http://localhost:8000/webrtcstream.html?rtsp://188773sc14.iask.in:554/live/sub

Please find the loggings in the gist 
(https://gist.github.com/choychris/e07be4ec82e28752b38a48386f39d330)

I am a newbie to webrtc and C++ coding env. Big appreciation for any help!

mpromonet, 2017-11-15:
Hi Chris

The reason your build have not rtsp support is probable because you forgot to run make live555 to instal live555.

If you like you can download binaries from the github release.

From your log i did not see obvious problem. You could try to increase the log level (using -vvv or the API /loglevel).
Another thing to check is whether openRTSP could receive your stream.

Best regards
Michel

choychris, 2017-11-18:
Thanks a lot @mpromonet !
I try to deploy the ubuntu(trusty) release in aws.
And it finally works! Bravo!

Though this is minor, what i can do to disable this error message?
[image]

And, my rtsp is already at H.264,  if I do not want do the H.264 to VP8/VP9 transcoding, how can I do so?

Anyway, do you take donation for this project? Cause I am willing to donate for expressing my appreciation ;) 

Cheers,
Chris

choychris, 2018-02-03:
Hi,

I discover that after a while we haven't use the server to stream any ipcam. 
The initial stream always fails to connect.
Is there a sleep mode or something in the server ?

mpromonet, 2018-02-03:
Hi Chris,
I think this is the behaviour of AWS, that hibernate the VM and wake-up when you make an HTTP request.
Maybe some configuration of AWS VM is possible ?
Best Regards,
Michel.

closed 2018-07-15 by mpromonet