mpromonet/webrtc-streamer #77: Type error on vanilla Ubuntu

johnmetta, 2017-11-10:
Building HEAD source on stock Ubuntu 16.04, after a seemingly successful build of webrtc, yields the following:

```
In file included from src/PeerConnectionManager.cpp:22:0:
inc/rtspvideocapturer.h:63:12: error: ‘InternalDecoderFactory’ in namespace ‘cricket’ does not name a type
   cricket::InternalDecoderFactory       m_factory;
            ^
In file included from src/PeerConnectionManager.cpp:22:0:
```

Seems like a header's not getting loaded, but I'd expect you to have seen that in the release, however, I get the same thing when building release 7.0.0. Any ideas?

johnmetta, 2017-11-10:
It looks like maybe your HEAD diverged from webrtc HEAD? Is there a git commit hash in webrtc that you have a known good build against, just to lock the versions?

johnmetta, 2017-11-11:
FYI, the interface changed a few days ago:

https://webrtc.googlesource.com/src/+/b2fc9b1b104240e68047901309deaee3e8b94bea

That was then reverted in 267d84ba and relanded in 7501b1c

I'll fork this and see if I can't make a pull request to update it.

closed 2017-11-11 by johnmetta

mpromonet, 2017-11-11:
Hi John,
I fix the build with the lastest WebRTC.
You can get the WebRTC GIT SHA1 of the latest deployed version using https://webrtc-streamer.herokuapp.com/version.
Best Regards,
Michel.

johnmetta, 2017-11-13:
Oh wonderful, I'll pull Master. Thank you.

commit 6551e17a (2017-11-11): fix #77: follow WebRTC update: change LOG->RTC_LOG and cricket::InternalDecoderFacotry to webrtc::InternalDecoderFacotry