mpromonet/webrtc-streamer #78: Failed to create the platform specific ADM implementation

johnmetta, 2017-11-13:
Hi Michael,

I got a core dump with a "Failed to create the platform specific ADM implementation" message (Ubuntu 16.04 in VirtualBox on a Mac.) Thinking it was something on my system, I tried your Docker image and got the same thing (Full log below). It looks from [this oldl bug report](https://bugs.chromium.org/p/webrtc/issues/detail?id=3416#c2) that Webrtc might want to "_really_ wants to initialize both voice & video engine even before it creates its first PC." But I don't know that's the case. 

I'm trying to build now on non VM Ubuntu using an Arm64 processor (Nvidia TX2, which is where I need to run this). Unfortunately, getting Docker running on TX2 is [non-trivial](https://devtalk.nvidia.com/default/topic/1000224/jetson-tx2/docker-on-the-tx2), so I can't test your docker instance there as well. Thought you'd want to see this as it is possibly broader than my system.

Cheers,
-J
```
sudo docker run -p 8000:8000 -it mpromonet/webrtc-streamer -vvv rtsp://184.72.239.149/vod/mp4:BigBuckBunny_175k.movLogger level:1
[000:000] [1] (audio_device_impl.cc:72): Create
[000:000] [1] (audio_device_buffer.cc:66): AudioDeviceBuffer::ctor
[000:000] [1] (audio_device_impl.cc:98): AudioDeviceModuleImpl
[000:000] [1] (audio_device_impl.cc:102): CheckPlatform
[000:000] [1] (audio_device_impl.cc:113): current platform is Linux
[000:000] [1] (audio_device_impl.cc:132): CreatePlatformSpecificObjects
[000:000] [1] (audio_device_impl.cc:996): PlatformAudioLayer
[000:000] [1] (audio_device_impl.cc:249): Failed to create the platform specific ADM implementation.
[000:000] [1] (audio_device_impl.cc:263): ~AudioDeviceModuleImpl
[000:000] [1] (audio_device_buffer.cc:80): AudioDeviceBuffer::~dtor
[000:000] [1] (audio_processing_impl.cc:406): Capture post processor activated: 0
[000:000] [1] (webrtcvoiceengine.cc:241): WebRtcVoiceEngine::WebRtcVoiceEngine
[000:000] [1] (webrtcvideoengine.cc:444): WebRtcVideoEngine::WebRtcVideoEngine()
[000:001] [1] (webrtcvoiceengine.cc:259): WebRtcVoiceEngine::Init
[000:001] [1] (webrtcvoiceengine.cc:272): Supported send codecs in order of preference:
[000:001] [1] (webrtcvoiceengine.cc:275): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:002] [1] (webrtcvoiceengine.cc:275): ISAC/16000/1 (103)
[000:002] [1] (webrtcvoiceengine.cc:275): ISAC/32000/1 (104)
[000:002] [1] (webrtcvoiceengine.cc:275): G722/8000/1 (9)
[000:002] [1] (webrtcvoiceengine.cc:275): ILBC/8000/1 (102)
[000:002] [1] (webrtcvoiceengine.cc:275): PCMU/8000/1 (0)
[000:002] [1] (webrtcvoiceengine.cc:275): PCMA/8000/1 (8)
[000:002] [1] (webrtcvoiceengine.cc:275): CN/32000/1 (106)
[000:002] [1] (webrtcvoiceengine.cc:275): CN/16000/1 (105)
[000:002] [1] (webrtcvoiceengine.cc:275): CN/8000/1 (13)
[000:002] [1] (webrtcvoiceengine.cc:275): telephone-event/48000/1 (110)
[000:002] [1] (webrtcvoiceengine.cc:275): telephone-event/32000/1 (112)
[000:002] [1] (webrtcvoiceengine.cc:275): telephone-event/16000/1 (113)
[000:002] [1] (webrtcvoiceengine.cc:275): telephone-event/8000/1 (126)
[000:002] [1] (webrtcvoiceengine.cc:278): Supported recv codecs in order of preference:
[000:002] [1] (webrtcvoiceengine.cc:281): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:002] [1] (webrtcvoiceengine.cc:281): ISAC/16000/1 (103)
[000:002] [1] (webrtcvoiceengine.cc:281): ISAC/32000/1 (104)
[000:002] [1] (webrtcvoiceengine.cc:281): G722/8000/1 (9)
[000:002] [1] (webrtcvoiceengine.cc:281): ILBC/8000/1 (102)
[000:002] [1] (webrtcvoiceengine.cc:281): PCMU/8000/1 (0)
[000:002] [1] (webrtcvoiceengine.cc:281): PCMA/8000/1 (8)
[000:002] [1] (webrtcvoiceengine.cc:281): CN/32000/1 (106)
[000:002] [1] (webrtcvoiceengine.cc:281): CN/16000/1 (105)
[000:002] [1] (webrtcvoiceengine.cc:281): CN/8000/1 (13)
[000:002] [1] (webrtcvoiceengine.cc:281): telephone-event/48000/1 (110)
[000:002] [1] (webrtcvoiceengine.cc:281): telephone-event/32000/1 (112)
[000:002] [1] (webrtcvoiceengine.cc:281): telephone-event/16000/1 (113)
[000:002] [1] (webrtcvoiceengine.cc:281): telephone-event/8000/1 (126)
[000:002] [1] (audio_device_impl.cc:72): Create
[000:002] [1] (audio_device_buffer.cc:66): AudioDeviceBuffer::ctor
[000:002] [1] (audio_device_impl.cc:98): AudioDeviceModuleImpl
[000:002] [1] (audio_device_impl.cc:102): CheckPlatform
[000:002] [1] (audio_device_impl.cc:113): current platform is Linux
[000:002] [1] (audio_device_impl.cc:132): CreatePlatformSpecificObjects
[000:002] [1] (audio_device_impl.cc:996): PlatformAudioLayer
[000:002] [1] (audio_device_impl.cc:249): Failed to create the platform specific ADM implementation.
[000:002] [1] (audio_device_impl.cc:263): ~AudioDeviceModuleImpl
[000:002] [1] (audio_device_buffer.cc:80): AudioDeviceBuffer::~dtor
[000:002] [1] (voe_base_impl.cc:166): Init() failed to create the ADM

#
# Fatal error in ../../media/engine/webrtcvoiceengine.cc, line 287
# last system error: 0
# Check failed: 0 == voe_wrapper_->base()->Init(adm_.get(), apm(), decoder_factory_) (0 vs. -1)
# 
#

==== C stack trace ===============================

./webrtc-streamer() [0x4ee3d2]
./webrtc-streamer() [0x4fc9f3]
./webrtc-streamer() [0x4fadba]
./webrtc-streamer() [0x80c521]
./webrtc-streamer() [0x4d7ee9]
./webrtc-streamer() [0x4d8333]
./webrtc-streamer() [0x80b237]
./webrtc-streamer() [0x54e925]
./webrtc-streamer() [0x5500ae]
./webrtc-streamer() [0x54e722]
./webrtc-streamer() [0x478098]
./webrtc-streamer() [0x477f10]
./webrtc-streamer() [0x478487]
./webrtc-streamer() [0x40d038]
./webrtc-streamer() [0x43d0dc]
16: __libc_start_main
./webrtc-streamer() [0x40bfe9]
```

mpromonet, 2017-11-13:
Hi,
This seems a problem initializing alsa.
You can try with option '-a' to disable audio capture.
Regards
Michel

mpromonet, 2017-11-19:
@johnmetta 
I closed this, now the default audio layer is ALSA that doesnot raise an exception.
Best Regards,
Michel.

closed 2017-11-19 by mpromonet