mpromonet/webrtc-streamer #79: Compilation failed if live555 is enabled

traw, 2017-11-14:
Compilation fails if live555 is enabled
Command: `make live555`
Trace:
``` c++
> make live555
LIVE555 not found in //usr/include
wget http://www.live555.com/liveMedia/public/live555-latest.tar.gz -O - | tar xzf -
--2017-11-14 11:50:06--  http://www.live555.com/liveMedia/public/live555-latest.tar.gz
Resolving www.live555.com (www.live555.com)... 4.79.217.242
Connecting to www.live555.com (www.live555.com)|4.79.217.242|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 629858 (615K) [application/x-gzip]
Saving to: ‘STDOUT’

-                                         100%[====================================================================================>] 615.10K  81.9KB/s    in 7.4s    

2017-11-14 11:50:14 (82.9 KB/s) - written to stdout [629858/629858]

cd live && ./genMakefiles linux-gdb
make -C live CPLUSPLUS_COMPILER="g++ -fno-rtti " C_COMPILER=gcc  LINK='g++ -o' PREFIX=//usr install
make[1]: Entering directory '/mnt/repos/cctv/webrtc/webrtc-streamer/live'
cd liveMedia ; make install
make[2]: Entering directory '/mnt/repos/cctv/webrtc/webrtc-streamer/live/liveMedia'
install -d //usr/include/liveMedia //usr/lib
install: cannot change permissions of ‘//usr/include/liveMedia’: No such file or directory
Makefile:418: recipe for target 'install1' failed
make[2]: *** [install1] Error 1
make[2]: Leaving directory '/mnt/repos/cctv/webrtc/webrtc-streamer/live/liveMedia'
Makefile:44: recipe for target 'install' failed
make[1]: *** [install] Error 2
make[1]: Leaving directory '/mnt/repos/cctv/webrtc/webrtc-streamer/live'
Makefile:113: recipe for target 'live555' failed
make: *** [live555] Error 2

```

mpromonet, 2017-11-14:
Hi traw,
This fails because you did not have permission to write in /usr.
You can specify a PREFIX to install it where you are allowed to write.

       make live555 PREFIX=/tmp
       make PREFIX=/tmp

Usually (as decribed in the README, in DockerFile and in .travis.iml) I use the WebRTC SYSROOT. when I built WebRTC with its SYSROOT in order to avoid to use 2 different STL version which cause some problems.

Best Regards,
Michel.

traw, 2017-11-14:
ok that's great. But I think Make file should install in somewhere in source tree this way all the header and lib created by live555 can easily be accessed, instead of root file system(/).

mpromonet, 2017-11-14:
Hi,
If you like you can do this using 'PREFIX=$(pwd)'
Regards
Michel

saket424, 2017-11-15:
I encountered a build error in the MJPEG block of code src/rtspvideocapturer.cpp line 171 and had to stub out that section of code for the rest of the code to build

g++ -o src/rtspvideocapturer.o -c src/rtspvideocapturer.cpp -Wall -pthread -g -std=c++11 -Iinc --sysroot=/  -DWEBRTC_POSIX -fno-rtti -D_GLIBCXX_USE_CXX11_ABI=0 -DHAVE_JPEG -I ../webrtc/src -I ../webrtc/src/third_party/jsoncpp/source/include -I ../webrtc/src/third_party/libyuv/include -DNDEBUG=1 -DHAVE_ALSA -I //tmp/include -DHAVE_LIVE555 -I live555helper/inc -I //tmp/include/liveMedia  -I //tmp/include/groupsock -I //tmp/include/UsageEnvironment -I //tmp/include/BasicUsageEnvironment/ -I civetweb/include -I h264bitstream -DVERSION="\"v0.0.7-5-g86f941e webrtc@6dfb5a8-dirty live555helper@5d3072a civetweb@v1.9.1-177-gd9b713b h264bitstream@0.2.0\""
In file included from src/rtspvideocapturer.cpp:21:0:
inc/rtspvideocapturer.h: In member function ‘virtual bool RTSPAudioSource::onData(const char*, unsigned char*, ssize_t, timeval)’:
inc/rtspvideocapturer.h:160:29: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      while (m_buffer.size() > segmentLength*m_channel) {
                             ^
src/rtspvideocapturer.cpp: In member function ‘virtual bool RTSPVideoCapturer::onData(const char*, unsigned char*, ssize_t, timeval)’:
src/rtspvideocapturer.cpp:119:37: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if ( (GetCaptureFormat()->width != width) || (GetCaptureFormat()->height != height) )  {
                                     ^
src/rtspvideocapturer.cpp:119:78: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if ( (GetCaptureFormat()->width != width) || (GetCaptureFormat()->height != height) )  {
                                                                              ^
src/rtspvideocapturer.cpp:171:52: error: ‘ConvertToI420’ was not declared in this scope
          webrtc::kVideoRotation_0, I420buffer.get());
                                                    ^
src/rtspvideocapturer.cpp:171:52: note: suggested alternative:
In file included from src/rtspvideocapturer.cpp:19:0:
../webrtc/src/third_party/libyuv/include/libyuv/convert.h:344:5: note:   ‘libyuv::ConvertToI420’
 int ConvertToI420(const uint8* src_frame,
     ^
src/rtspvideocapturer.cpp: In member function ‘virtual ssize_t RTSPVideoCapturer::onNewBuffer(unsigned char*, ssize_t)’:
src/rtspvideocapturer.cpp:193:12: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   if (size > sizeof(marker))
            ^
Makefile:94: recipe for target 'src/rtspvideocapturer.o' failed
make: *** [src/rtspvideocapturer.o] Error 1

kibot914, 2017-11-17:
[image]

I also got the same error

commit 1c1218e9 (2017-11-19): fix #79: use libyuv instead of webrtc_libyuv that have removed ConvertToI420
closed 2017-11-19 by mpromonet