mpromonet/webrtc-streamer #80: How to change encoder setting?

traw, 2017-11-14:
How can we change encoder setting whilst streaming local webcam?

mpromonet, 2017-11-14:
Hi traw,
You can use v4l2-ctl to change the webcam setting, however it is highly probable that your capture device will answer BUSY to this command, till the capture device is used by the WebRTC SDK.
Best Regards,
Michel.

traw, 2017-11-14:
ok, I was asking about audio/video codec like vp8, vp9, g711 etc.

mpromonet, 2017-11-14:
You can do this modifying the SDP you get.
There is lots of article about SDP munging https://bloggeek.me/webrtc-sdp-munging-prohibited/

traw, 2017-11-14:
does that mean I may need to modify `OnSuccess` in either `SetSessionDescriptionObserver` or `CreateSessionDescriptionObserver` class located in `PeerConnectionManager.h`.
Am I getting it right?

mpromonet, 2017-11-16:
Hi,
I guess you should rather modify the JavaScript to modify the sdp receive as call answer before giving it to setRemoteDescription.
Best regards
Michel

closed 2019-04-10 by mpromonet