mpromonet/webrtc-streamer #81: Support Proxy

Jamie0, 2017-11-14:
webrtc-native has proxy support. 

I would like to see support in webrtc-streamer for using a proxy to access, say, a TURN server. This would be useful behind corporate firewalls.

mpromonet, 2017-11-14:
Hi Jamie,
I am not sure to understand, however you can specify a TURN server using '-t' option.
If this doesnot fit your need, feel free to make modifications you need and submit pull requests.
Best Regards,
Michel.

Jamie0, 2017-11-21:
Here's a use case that hopefully details why I think this necessary:

- webrtc-streamer runs behind a corporate proxy A on network A
- Client runs behind a corporate proxy B on network B
- A TURN server runs on, say, network B, but has no outbound direct connection except via proxy
- Client attempts to connect to webrtc-streamer, using TURN server on network B

As the networks are disjoint, and non-LAN requests can only be made through the proxy, it is hence impossible for both peers to communicate.

The solution would be that webrtc-streamer can connect to a TURN server on network B through proxy A, and then to the client.

AneriFumtiwala, 2019-02-11:
@Jamie0 & @mpromonet 

Have you added proxy support to webrtc server? If yes, can you please guide me how can I configure it?

mpromonet, 2019-03-10:
Hi AneriFumtiwala,
I have not really understood the purpose of this request, it may be to access to a webrtc-streamer running on internet behind a proxy. 
You may look to the Jamie0 fork that made some modification in https://github.com/mpromonet/webrtc-streamer/compare/master...Jamie0:proxy-support in order to customize the port allocator.
This is too hard-coded to be merged.
Best Regards,
Michel.