mpromonet/webrtc-streamer #82: streaming video from file

traw, 2017-11-14:
I think streaming from file(webm, h264) is not implemented, so this might be a new feature request

saket424, 2017-11-15:
I did not know ffmpeg could write to v4l2 devices. Now I do

https://askubuntu.com/questions/881305/is-there-any-way-ffmpeg-send-video-to-dev-video0-on-ubuntu

traw, 2017-11-16:
That's good work around, but if we create new class `FFmpegVideoCapture`, it would be better. If we write what things need to be taken care any guide, achieving that I can give a try.

mpromonet, 2017-11-16:
Hi,

After getting the frame you should call onFrame on your videocapturer overide.

About recording, i think i will write to a vidéo v4l2 output device. This will allow either to record it, restream it, display it ..

Best regards
Michel

traw, 2017-11-18:
OK thats great, thanks

closed 2017-11-18 by mpromonet